freifunk-gluon/gluon #1194: gluon-status-page-api: improve cross origin policy

lemoer, 2017-07-20:
```Access-Control-Allow-Origin: *``` is maybe a bad idea for the status-page-api, since other websites could derive a geo-location from the status page by accessing the api via ajax.

jplitza, 2017-07-30:
Very true. I see two possible solutions without totally breaking the status page:

1. We do not send the header at all when the API is called via a "generic" (next_node) address. This makes it impossible to gather information without knowing the address of the node a user is connected to in advance. However, much more logic has to be implemented server-side (we currently simply ignore the Host header AFAIK).
2. We only send `Access-Control-Allow-Origin: $site.next_node.name$`. This will only display neighbour data if the status page is originally called via the next_node address (even if you then navigate away from your next node).

christf, 2017-08-04:
I like (2). It seems like a small change and it will address the problem.

NeoRaider, 2017-08-06:
While this will need some transition time, we should simply change the status page to actually link to other nodes' status pages instead of only changing the backend URL when another node is selected, thus avoiding the whole issue.

We could still provide a single `Access-Control-Allow-Origin: *` endpoint just returning an empty page that can be used to check reachablity (and select a neighbour node's address based on this, like it is already done for the backend URL at the moment).

commit fdf823a3 (2018-02-26): gluon-status-page-api: merge back into gluon-status-page

All Access-Control-Allow-Origin are removed to improve users' privacy. As
the status page API is thus not useful without the status page anymore,
merge them back into a single package.

The status-page-api respondd provider is removed as well.

Fixes #1194
closed 2018-02-26 by NeoRaider