freifunk-gluon/gluon #1195: UAPv2 Support

Eichi, 2017-07-21:
Hi,  
  
today I got a new UAP and as it came out it is a new revision.  
So here is a paste of dmesg:  
https://pastebin.com/KxgTSKqk  
  
If there are any other infos I can give, please ask me.  
  
I hope it helps to support the new UAPs.  
  
Thanks a lot!

Eichi, 2017-07-21:
And as requested the output of:  
`BZ.v3.7.19# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00060000 00010000 "u-boot"
mtd1: 00010000 00010000 "u-boot-env"
mtd2: 00740000 00010000 "kernel"
mtd3: 00040000 00010000 "cfg"
mtd4: 00010000 00010000 "EEPROM"
`

rotanid, 2018-01-19:
at first there has to be support in LEDE/OpenWrt before Gluon can do anything.
it looks like it won't be possible to support this device in LEDE/OpenWrt as it uses firmware signing and blocks unsigned firmware, see following links:
https://forum.lede-project.org/t/unifi-v2-signed-firmware/6928
https://forum.lede-project.org/t/are-ubiquiti-actively-blocking-installation-of-third-party-firmware/4301
https://forum.lede-project.org/t/are-ubiquiti-blocking-lede-installation/3877

closed 2018-01-19 by rotanid