freifunk-gluon/gluon #1198: batman-adv: bridge loop avoidance broken in Gluon

T-X, 2017-07-29:
I just tried connecting a network of just two nodes, both devices on the same desk, not just via wifi, but also connecting them via a cable between their LAN ports.

batman-adv has a bridge loop avoidance feature which should in theory make this possible. This batman-adv feature is enabled by default since not that long ago and by that it is enabled in Gluon, too.

However, in Gluon it does quite work yet. Two issues I was able to spot from having a brief look at it:

A) The ARP filtering is two restrictive. BLA sends its own "claim frames" and for that utilizes ARP. The result is that for instance my ssh connection broke every couple of seconds first. A few minutes later, there were a couple of hundreds of ARP packets circling for the custom 192.168.1.123/24 I had set on a host device connected to the LAN port of one of the two nodes. Removing the ebtables ARP drop rules in Gluon made connections stable and usable again. Instead of needing to remove the ARP drop rules completely to enable BLA, the ARP frames generated by BLA should be distinguishable by the magic MAC addresses it uses.

B) IGMP/MLD is currently only filtered into the mesh but not on the LAN side. This results in the querier of one node disabling the one of the other node. The other node further classifies the neighbor node's querier as a "shadowing one", resulting in disabling its own multicast TT entries and simply requesting all multicast packets instead. Currently should not make a difference, but to get the multicast optimizations and BLA play nicely later, this needs a fix, too.

christf, 2017-08-03:
@T-X for a) which mac-address should be allowed for all arp?

What should be done to address b)?

NeoRaider, 2017-08-06:
For b), I'd expect this to work as expected: there needs to be only one MLD querier per LAN segment, so it is correct behaviour that one of the nodes disables its querier. I think batman-adv shouldn't care which node does the querying (if it does: why?).

Will both nodes announce the multicast TT entries for the LAN segment in a BLA scenario (leading to duplication of multicast packets)?

A-Kasper, 2017-08-24:
my log is also full of messages, that the supernode receives packages with it's own adress.. seems like a bug. I don't know if gluon or batman related. if i should try to gather some information please tell me what you need...

CodeFetch, 2017-11-29:
@A-Kasper Seems like a duplicate of https://github.com/libremesh/lime-packages/issues/189 to me.

belzebub40k, 2017-12-11:
For **a)** [batman-adv utilizes gratuitous ARP packets](https://www.open-mesh.org/projects/batman-adv/wiki/Bridge-loop-avoidance-Protocol) with target MAC addresses `FF:43:05:00:XX:XX` to `FF:43:05:05:XX:XX`. Adding the following rule before the ARP DROP rules in the chain `MULTICAST` should be sufficient to allow the special ARP packets.

 `-p ARP --arp-gratuitous --arp-mac-dst FF:43:05:00:00:00/FF:FF:FF:F0:00:00 -j RETURN `

closed 2017-12-27 by NeoRaider

rubo77, 2019-01-03:
@T-X:
> This batman-adv feature is enabled by default since not that long ago and by that it is enabled in Gluon, too.

Are you sure? We just had a loop in our network in Kiel with gluon 2018.1.4 where a user connected two ffki-nodes with a cable from lan to lan port

commit f7995181 (2017-12-27): gluon-ebtables-filter-multicast: do not filter out Bridge Loop Avoidance

batman-adv uses gratuitous ARP packets with certain target addresses for
BLA.

Fixes #1198
commit c2c43901 (2018-01-30): gluon-ebtables-filter-multicast: do not filter out Bridge Loop Avoidance

batman-adv uses gratuitous ARP packets with certain target addresses for
BLA.

Fixes #1198