freifunk-gluon/gluon #1203: Missing WLAN channel on router status page

smoe, 2017-08-11:
Hello, I had thought it to be straight forward to add a patch myself, but, well, not so, I was defeated in an attempt to read anything from the uci into the respondd structure. Please help.

To increase air time we often have several devices at the same location that mesh on LAN but use different WLAN channels. Admittedly, at times one gets a bit confused and for machines without remote access in particular it would be very helpful to just see the channel mentioned on the status page.

As  a start the information could be with the nodeinfo block, I thought. Eventually, it would however be nice to see statistics (#clients, bandwidth) for each channel the device is communicating on.

Many thanks

Steffen

rotanid, 2017-08-11:
do you already have the module respondd-module-airtime in your site.mk?
with this, the data is in respondd and you only have to read it on the status page. this has to be made safe for people not using respondd-module-airtime (which is part of gluon since v2017.1)
we would welcome a pull request for this :)

NeoRaider, 2017-08-11:
respondd-module-airtime seems like overkill for this small piece of data though...

smoe, 2017-08-12:
Hi Rotanid, I'll give respondd-module-airtime a deeper look. Seems like something to learn for me. I'll see what I can do and eventually prepare the PR if not beaten to it.

@NeoRaider , in my perception it is fine to have a complicated status page since us admins are the only ones looking at them, anyway. The only dyscomfort comes with my very personal uncertainty about whom the respondd is responding to, i.e. fearing about a negative impact on bandwith when lots of respondd replies are synchronized.

Many thanks to you both!!!

rotanid, 2018-07-12:
let's track this in #644

closed 2018-07-12 by rotanid