freifunk-gluon/gluon #1208: RFC: use vlans on lan

A-Kasper, 2017-08-17:
I would like to discuss to make it possible to enable vlan tagging on the lan interface. 
I want to have this, because I could offer to enable an additional ssid in bigger environments, where freifunk just would be an add-on.
vlan tags for mesh and client network on the lan interface would make it possible to spread the network with stock devices. this could be very nice in some cases and now needs some "nicht updatefestes gefrickel"

kpanic23, 2017-08-17:
Why don't you just configure the network switch's port connected to the freifunk lan interface as untagged vlan?
We have multiple installations configured like that.

A-Kasper, 2017-08-17:
Because I want to have one vlan mesh (to connect gluon nodes) and one vlan client-network (to deploy it via installed wlan setup as aditional ssid)

rotanid, 2017-08-17:
in the future meshing via cable will be done via VXLAN (already in gluon master since june) which should make it easier. see also the roadmap: https://github.com/freifunk-gluon/gluon/wiki/Roadmap

AiyionPrime, 2022-06-03:
@A-Kasper have you tried the a recent gluon master build, yet?
I think what you wanted for the lan interface has been merged in #2372 or so.

You could configure your lan interface like this:

> ```
> root@router:~# cat /etc/config/gluon
> [...]
> 
> config interface 'iface_lan_vlan43'
> 	option name 'eth0.43'
> 	list role 'client'
> ```

_Originally posted by @lemoer in https://github.com/freifunk-gluon/gluon/issues/2393#issue-1131343356

If you agree, feel free to mark this issue as resolved :)

AiyionPrime, 2022-12-10:
It's been half a year, I'll close this.

@A-Kasper feel free to reopen this issue, if your requirements were not met yet,
otherwise thanks for your engagement.

closed 2022-12-10 by AiyionPrime