freifunk-gluon/gluon #1212: Autoupdater läuft als Zombie

MPW1412, 2017-08-20:
Hallo,

wir haben jetzt beim Update von 2016.2.1 auf 2016.2.6 an vielen Stellen das Problem gehabt, dass Knoten sich nicht aktualiaisert haben, weil der Autoupdater als Zombie-Prozess lief. Ist das Problem bekannt? Die Knoten funktionierten ganz normal, daher kommen die Betreiber auch nicht auf die Idee, die Dinger neu zu starten. 

Gibt es da Erfahrungen zu?

Grüße
Matthias

rotanid, 2017-08-20:
Das Problem ist bekannt und in aktuellen Versionen behoben.
Siehe dazu auch folgende issues:
#1160 
#1010 
#987

closed 2017-08-20 by rotanid

MPW1412, 2017-08-20:
Ah danke für die Info. Hatte nach Autoupdater und zombie gesucht, aber war wohl blind. Danke für's Raussuchen!

rotanid, 2017-08-20:
kein Problem. Meine Suche war:
https://github.com/freifunk-gluon/gluon/issues?utf8=%E2%9C%93&q=is%3Aissue%20autoupdater