freifunk-gluon/gluon #1213: VLAN not working in 11s config

mokkin, 2017-08-20:
The configured VLAN isn't working since switching to 11s.
```
batctl if
mesh0: active
primary0: active
```
site.conf:
https://github.com/freifunk-leipzig/freifunk-gluon-leipzig/commit/f2fccb34a691b9bc4825f6e9c50a72b64a782508

A-Kasper, 2017-08-21:
There is no VLAN support for 802.11s mode. So everything works fine.

VLAN support is already on the roadmap.

NeoRaider, 2017-08-21:
As @A-Kasper correctly noted, Gluon does not support VLAN over 11s; it is simply not implemented. The VLAN support code adds unnecessary complexity to the network configuration without significant gain, so we don't intend to support it (and the current VLAN over IBSS code will be removed when we drop IBSS support).

closed 2017-08-21 by NeoRaider