freifunk-gluon/gluon #1214: autoupdater (uclient) does not like http(s)

viisauksena, 2017-08-22:
i got this "bug" when i try to use autoupdater inside fresh build lede 2017.1.1

```Failed to allocate uclient context```

which seems to lay here 
https://bugs.lede-project.org/index.php?do=details&task_id=122

i havent test everything so far, but it seems that autoupdater adresses with http in front of the string fail - which in worst case means you have a dead end when all your adresses include the http string
if so, there may be a test of the autoupdater link strings to get rid of http stuff

NeoRaider, 2017-08-22:
The autoupdater URL need to start with "http://" (without, it is no valid URL), they should fail without. HTTPS is not supported.

viisauksena, 2017-08-22:
ah, vis versa, make sense

NeoRaider, 2017-08-22:
Before LEDE, busybox's wget implementation instead of uclient-fetch was used; it's possible that it erroneously accepted such URLs.

Generally, all features of a new firmware should be tested before a rollout (I'd go as far as saying: the autoupdater is the single component you really need to test, everything else can be fixed through updates :P ).

We should probably add a validator to ensure that all mirror URLs in site.conf start with "http://".

jplitza, 2017-08-22:
I'd like to note that I'm amazed the new autoupdater is being tested already, even though it's not merged yet! :smile: :+1:

NeoRaider, 2017-08-23:
@jplitza I assume this is about the old autoupdater, which now uses uclient as well, as uclient-fetch has replaced busybox's wget implementation in LEDE.

viisauksena, 2017-08-27:
this is by build of 2017.1.1
also i got a http error 400 when downloading the correct file 
(ping is working, wget also fails like autoupdater)
its basicly this ```wget "http://[fdf0:9bb:7814:a630::7]/support/support.manifest" -O -```
which works fine every else

NeoRaider, 2017-09-05:
@viisauksena What HTTP server are you using? I have an idea what is causing the error 400, but I'd like to reproduce it myself.

NeoRaider, 2017-09-06:
A validation for "http://" has been added, and I've fixed the issue that caused error 400 in uclient (fix is included in the latest LEDE update). The LEDE update will be backported to v2017.1.x after it has received some testing in the Gluon master.

closed 2017-09-06 by NeoRaider