freifunk-gluon/gluon #1218: D-Link DIR-635

HACKER-3000, 2017-08-29:
Yes, it's me again :)
Whats about to add some Routers:
1. Old *FritzBox* Router like:
  1a. FritzBox WLAN 3170
  1b. FritzBox Fon WLAN 7050
  1c. ect...
2. Old *Speedport* Router of the *Telekom* like:
  2a. Speedport W 501V
  2b. Speedport W 503V
  2c. Speedport W 701V
3. More D-Link Routers:
  3a. DIR-635
  3b. DI-524
4. Only one Netgear Router:
 4a DGND3800B

Yes it's a lot of Work :))

But Thanks

ambassador86, 2017-08-29:
Hello @HACKER-3000 what is this issue about? To support another router model we need at least stable LEDE support.  All of the models above have crappy hardware and are not supportable by lede or at least very unstable.

rubo77, 2017-08-29:
hi hacker, I appreciate your noble request, but unfortunately we cannot put so much effort in hardware, that is not potent enough to create a stable mesh network.

If you have the time, feel free to add support for those routers. There is a section that explains what has to be done at https://gluon.readthedocs.io/en/v2017.1.x/dev/hardware.html?highlight=hardware

But I think we should concentrate on some promising models to make them stable first, for example the ([C25](https://github.com/freifunk-gluon/gluon/issues?utf8=%E2%9C%93&q=is%3Aissue%20is%3Aopen%20c25))

HACKER-3000, 2017-08-29:
S***

HACKER-3000, 2017-08-30:
And LEDE is avauble for the DGND3800B!!

NeoRaider, 2017-08-30:
DGND3800B uses a Broadcom wifi chip, which is generally unsuitable for mesh networks.

At the moment the following WLAN chips/drivers are supported:

* ath9k (Atheros 11n)
* ath10k (Atheros 11ac, may be unstable)

The following work, but aren't supported officially yet (marked as BROKEN), as they are not well-tested or the drivers are not mature enough:

* mt76 (Mediatek 11n/ac, many chips are unstable)
* rt305x (some older Ralink chips, not well-tested)

We generally don't expect other hardware to be supportable any time soon, as high-quality open source WLAN drivers with some specific features (802.11s, VIFs) are required for meshing. Many WLAN vendors aren't supporting the open source community at all, so no support is possible.

closed 2017-08-30 by NeoRaider

HACKER-3000, 2017-09-24:
but you can replace the wifi Card of the DGND3800B

Adorfer, 2017-09-25:
@HACKER-3000  And doing so makes which kind of sense? 
There are more economic and technical more feasable models available. 
But if you like, feel free to test and do pull requests.