freifunk-gluon/gluon #1219: Upgrading from a Firmware to another with different Update Server leaves old Update branches behind

MTRNord, 2017-09-01:
When updating from gluon freifunk Nord to freifunk schleswig-flensburg and keeping config (known that this isn't the correct behaviour) the New Firmware Shows old branches in the config Mode. I know it is intendet to keep the selected When keeping configuration but all other bin existent should be cleaned up I think. Maybe it should if the side Code changes also reset the autoupdater branch to Not break the Router on the next update. (sorry if some german words are in the Text. Writting from Phone. Will fix them as soon as I am at a PC)

Adorfer, 2017-09-03:
this is a normal behavior, even if you migrate Nodes to another domain/community.
Same goes for vpn servers if you named them differently.
In such case you definitely have to operate with a dedicated migration-FW which blanks out the "non desired entries" of the old old domaindo

rotanid, 2017-09-03:
as @Adorfer pointed out, this is expected behaviour.

closed 2017-09-03 by rotanid