freifunk-gluon/gluon #1221: How To Calculate External Ipv6 To Acces Nodes

jerix777, 2017-09-09:
I Can Calculate the local ipv6 of a node from it's MAC address.
Please, What about it's external address as shown on status page ?

A-Kasper, 2017-09-10:
The external IP depends on your communities prefix. just replace the local
prefix with the prefix of your community. The used prefix in your network
segment may be a subnet of your communities network..you need to know the
community setup to know the prefix or the possible  prefixes for your
network

jerix777, 2017-09-10:
Okay ! Thanks @A-kasper 
The setup prefix works as far as we are directly connected to a node. But this prefix don't permit access to a node not directly connected.
Into the Status Page of a node, we can see all ip addresses to access this node. So the question << Is there a way to calculate the external ip6 of a node from his MAC ?>>

rotanid, 2017-09-10:
what is a "setup prefix" and what means "not directly connected" ?
perhaps you community has no public IPv6? which community is it?
Your question has already been answered: if there is one,  the public IPv6 is calculated exactly the same like the link-local one.

jerix777, 2017-09-10:
Rotanid:
- Here, "setup Prefix" is the ip6_prefix sets on site.conf and represents the community ipv6.

Adorfer, 2017-09-10:
So i understand this question more like: "How does SLAC work in ipv6". 
_"if have an interfaces mac adress, how is the SLAC adress calculated"_

But before perhaps: 
How does a node determine it's different MAC addresses for the individual interfaces. 
Since there is BIG differences between the targets, depending on the spacing the manufacturer used to give out base MACs on consecutive (or other means) serial numbers, in order to prevent collisions when customer use devices from the same batch in a local network. 

somethimes the spacing is a relaxed 4bit... so enough for 16 interfaces, but sometimes only 2 bits, enough for typical small wifi router, but not enough for a gluon node with 6+ physical and virtual interfaces.

jerix777, 2017-09-10:
External ip6 as a format like this fd10::xxxx...

rotanid, 2017-09-10:
sorry, fd10:: is no external adress but an ULA (unique local address)
nevertheless, the generation follows the same pattern as for public or link-local ones.

Adorfer, 2017-09-11:
O.k. now we go into nitpicking about "what the question references as an external adress is not an external addres, but private adress space. So we will not explain until we receive a formal appology for the wrong wording."

rotanid, 2017-09-11:
@Adorfer your comment is way less helpful than mine. such comments might be OK on the LEDE forum, but not here.
he already wrote that he knows how to calculate the link-local (fe80::) address, so what's wrong with telling to do exactly the same for the fd10: prefix?

jerix777, 2017-09-11:
Thanks @Rotanid
ULA is exactly the correct wording.
But I don't really understand how to calculate it.
Please, give an example

Gammelknup, 2017-09-11:
@jerix777 

[image]

rotanid, 2017-09-11:
maybe this tool under "MAC -- EUI-64 Converter" helps? the mac address is needed to generate a ULA or link-local IPv6 address.
http://silmor.de/ipaddrcalc.html#ip6
you would then combine the result of this with your "fd10..." prefix.
in total, there will be 8 parts separated by colons

closed 2017-09-11 by rotanid