freifunk-gluon/gluon #1222: Add auhentification and clients details to status-page

jerix777, 2017-09-10:
In administrator vue of a local network, it may be helpfull to have the list of connected clients, their name, ip and MAC, linkstate to the node.
This option require to protect the node status page with a authentication proccess.

rotanid, 2017-09-10:
first, this is a privacy issue and second, adding your suggestion to add an authentication process to solve this will possibly result in an unwanted increase in image size (and complexity)

NeoRaider, 2017-09-10:
Indeed, there is not way to add any secure authentication to the status page without including SSL libraries in Gluon, which is infeasible. You can use SSH to get a list of connected MAC addresses (`batctl tl`, `iw dev client0 station dump`).

In addition, a Gluon node doesn't know anything about the clients' names or IP addresses; this information is only known to the DHCP server.

closed 2017-09-10 by NeoRaider

jerix777, 2017-09-10:
Okay ... I give up. thank for your suggestions