freifunk-gluon/gluon #1223: TL-WR1043NDv4: wrong MAC on devices that had newer stock-firmware

rotanid, 2017-09-11:
if you get a TP-Link TL-WR1043ND with the stock firmware version 170401/2017-04-01 or you update an older one to this version - then the flashed LEDE or gluon doesn't find the MAC address anymore and uses something like "0000100000".

@NeoRaider created a patch ( https://paste.debian.net/plain/985485 ) that fixes the issue.
i tested this and it works as intended.

this ticket is intended to keep track of the status of integration into all the upstream & local branches as well as serve as an informational piece for people experiencing the same problem.

Adorfer, 2017-09-11:
same issue on 2016.2.7
[image]

Adorfer, 2017-09-11:
btw: As a quickfix for affected nodes, you may assign a new ID manually: 

```
export NEWMAC=00:60:2f$(dd bs=1 count=3 if=/dev/random 2>/dev/null |hexdump -v -e '/1 ":%02x"'); echo $NEWMAC>/lib/gluon/core/sysconfig/primary_mac; export NEWKEY=$(fastd --generate-key|grep Secret|cut -d" " -f 2)
uci set fastd.mesh_vpn.secret=$NEWKEY; uci set network.client.macaddr=$NEWMAC; uci set network.bat0.macaddr=$NEWMAC; uci commit; echo NEWMAC  $NEWMAC; echo NEWFASTDKEY $NEWKEY
reboot;

```

rotanid, 2017-09-11:
yup, has nothing to do with the gluon version, i already tested gluon v2016.2.x, gluon v2017.1.x, gluon master and lede-17.01 and lede-master -- all of these branches need the fix.

Adorfer, 2017-09-11:
Do we have "hope" to have a patch in the v2016.2.x-release?
If not (since it's perhaps EOLed), then i would make a quickfix-script to run after installation in order to randomize the MAC.

NeoRaider, 2017-09-11:
The backport was easy enough, so I pushed it to v2016.2.x, too.

commit c0b4fb54 (2017-09-11): modules: update LEDE

Fixes #1223
commit 3576c8e6 (2017-09-11): modules: update LEDE

Fixes #1223
closed 2017-09-11 by NeoRaider