freifunk-gluon/gluon #1225: autoupdater should check disk size before isntalling sysupgrade

RalfJung, 2017-09-17:
The [release notes for Gluon 2017](https://gluon.readthedocs.io/en/v2017.1.x/releases/v2017.1.html) say that x86 devices need to have a disk of at least 273MB to upgrade to the 2017 version. What the release notes don't say is that the updater will brick devices that have smaller disks. I think this is a fatal bug -- the autoupdater really should check that the disk is large enough before installing the upgrade.

rotanid, 2017-09-17:
@RalfJung the autoupdater of the running system doesn't know about which disk size the to-be-installed version needs - so this can't be done (at least not easily)

H4ndl3, 2017-09-17:
@rotanid it doesn't know, but it could still check if the partition size is at least 273 MB big.

rotanid, 2017-09-17:
well, that would have to be integrated into v2016.2.x then, as the old system is doing most of the update work as far as i know.
@NeoRaider will have to take over here :D

RalfJung, 2017-09-18:
Making the provisions now to figure out the size requirements before writing would at least help prevent this issue the next time the disk size requirements are increased.

NeoRaider, 2017-09-19:
Checking the disk size before writing the image would be a good idea, this should be fixed in LEDE (I'll look into it). The kernel won't grow to more than 16MB in the forseeable future though, so it is unlikely that the size requirements will be increased again any time soon.

If we have such a check in place, it would also be nice to extend the autoupdater to remember that a specific image was rejected, so it won't try again every hour (shutting down many non-essential services each time). I have an idea for that too, but this should wait after the new C-based autoupdater is merged.

Adorfer, 2017-09-23:
it would even nice if the autoupdater would identify "roomy environments" (like x64 most likely are) and not shut down services "to save ram/cpu" during the critical updating phasis. 
Same goes e.g. for devices with 64MB+ RAM under most circumstances (i assume)

mweinelt, 2019-12-04:
As this is an upstream issue I don't think it's reasonable to track this in our milestones, therefore removing the milestones.