freifunk-gluon/gluon #1230: uradvd replies to RAs send by any client on the network

RalfJung, 2017-10-01:
## Steps to reproduce

In a large(ish) mesh network, have a node (e.g. in a VM) with only a single client connected to it, and observe ICMPv6 traffic at that client.

## Expected behavior

When there are no router solicitations, RAs should be sent only every [200-600s](https://github.com/freifunk-gluon/packages/blob/706bef62456be42a5bd00792366ec5fdf34326c2/net/uradvd/src/uradvd.c#L70).

## Actual behavior

An RA is sent every 3 second. This matches [`MIN_DELAY_BETWEEN_RAS`](https://github.com/freifunk-gluon/packages/blob/706bef62456be42a5bd00792366ec5fdf34326c2/net/uradvd/src/uradvd.c#L75).

I assume what happens is that somewhere in the network, there's always a client sending a solicitation, and hence the node replies. However, the reply of course only gets sent to the clients in the local wifi cell (or maybe the local mesh, not sure). Ideally, uradvd would ignore solicitations from clients that are a VPN link away, or these solicitations would be filtered so as to never even reach uradvd.

NeoRaider, 2017-10-01:
This should already be fixed in Gluon 2017.1.x: uradvd runs on the local-node interface, which is unreachable from the mesh (by ebtables filtering). In older Gluon versions, uradvd runs on br-client, and there is no easy way to achieve the desired filtering.

Can you confirm that your are using an older Gluon version?

RalfJung, 2017-10-02:
I am using Gluon 2017.1.2.

closed 2017-10-03 by NeoRaider

rotanid, 2017-10-03:
i guess this will also be backported at some point in the future?

RalfJung, 2017-10-03:
@rotanid it already is in the 2017.1 branch: <https://github.com/freifunk-gluon/gluon/commit/9cecb1e836d607e1e53e3a3bf0f44035202a8b25>

rotanid, 2017-10-03:
it wasnt when i commented, it was added 2 minutes after my comment.

RalfJung, 2017-10-03:
Fix confirmed (I tried the current 2017.1 branch). Thanks!

commit 9324d18f (2017-10-03): gluon-mesh-batman-adv: filter out all packages between bat0 and local-port

Filtering by MAC address won't filter out multicast packages like router
solicitations, causing uradvd to send out router advertisements with
maximum frequency (every 3 seconds) in active meshes, even when no local
client is actually interested in the advertisements.

Fixes #1230
commit 9cecb1e8 (2017-10-03): gluon-mesh-batman-adv: filter out all packages between bat0 and local-port

Filtering by MAC address won't filter out multicast packages like router
solicitations, causing uradvd to send out router advertisements with
maximum frequency (every 3 seconds) in active meshes, even when no local
client is actually interested in the advertisements.

Fixes #1230