freifunk-gluon/gluon #1232: x86-generic in KVM: Kernel Oops when running "autoupdater -f"

RalfJung, 2017-10-02:
When I boot the x86-generic image in KVM, I get a kernel oops when I run `autoupdater -f`:
```
# autoupdater -f
Downloading 'http://mgmt.ffsaar/firmware/experimental/sysupgrade/experimental.manifest'
Connecting to fd4e:f2d7:88d2:ffff::250:80
Writing to stdout
[ 1486.102503] ------------[ cut here ]------------
[ 1486.105269] WARNING: CPU: 0 PID: 0 at net/core/skbuff.c:4196 skb_try_coalesce+0x322/0x340()
[ 1486.109757] Modules linked in: iptable_nat nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE ebtable_nat ebtable_filter ebtable_broute xt_time xt_tcpudp xt_state xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_addrtype xt_TCPMSS xt_REDIRECT xt_LOG xt_CT via_velocity via_rhine tulip sky2 sis900 r8169 pcnet32 nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack ne2k_pci iptable_mangle iptable_filter ip_tables forcedeth ebtables ebt_vlan ebt_stp ebt_snat ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_ip6 ebt_ip ebt_dnat ebt_arpreply ebt_arp ebt_among ebt_802_3 e1000e e100 crc_ccitt br_netfilter 8390 8139too 8139cp sch_teql sch_sfq sch_red sch_prio sch_pie sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_ipt act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_htb sch_hfsc i2c_dev batman_adv libcrc32c cfg80211 compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables 3c59x igb e1000 dummy l2tp_ip6 l2tp_ip l2tp_eth l2tp_netlink l2tp_core udp_tunnel ip6_udp_tunnel veth natsemi act_police cls_basic sch_tbf sch_ingress button_hotplug tg3 ptp pps_core mii libphy
[ 1486.153705] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.4.79 #0
[ 1486.156112] Hardware name: QEMU Standard PC (i440FX + PIIX, 1996), BIOS 1.10.2-1 04/01/2014
[ 1486.158005]  00000000 c12103e9 00000000 c190a3bc c1086f09 00000009 c70d8540 00000001
[ 1486.160047]  c70d83c0 000004b8 c1086fad 00000009 00000000 c15d4222 c70d83c0 00000400
[ 1486.160047]  c15d3a32 000004b8 00000000 c70d8540 c70d83c0 c6ead800 c780dd9b c1621c1a
[ 1486.160047] Call Trace:
[ 1486.160047]  [<c12103e9>] ? dump_stack+0x55/0x7c
[ 1486.160047]  [<c1086f09>] ? warn_slowpath_common+0x69/0xa0
[ 1486.160047]  [<c1086fad>] ? warn_slowpath_null+0xd/0x10
[ 1486.160047]  [<c15d4222>] ? skb_try_coalesce+0x322/0x340
[ 1486.160047]  [<c15d3a32>] ? __skb_checksum+0xd2/0x1e0
[ 1486.160047]  [<c1621c1a>] ? tcp_try_coalesce+0x3a/0x80
[ 1486.160047]  [<c1621ca8>] ? tcp_queue_rcv+0x48/0x120
[ 1486.160047]  [<c1626e1e>] ? tcp_rcv_established+0x33e/0x610
[ 1486.160047]  [<c168add8>] ? tcp_v6_do_rcv+0x288/0x3a0
[ 1486.160047]  [<c168b7a7>] ? tcp_v6_rcv+0x8b7/0x960
[ 1486.160047]  [<c160a9a1>] ? nf_iterate+0x51/0x60
[ 1486.160047]  [<c166947a>] ? ip6_input_finish+0xea/0x330
[ 1486.160047]  [<c160aa00>] ? nf_hook_slow+0x50/0xa0
[ 1486.160047]  [<c1669a6e>] ? ip6_input+0x7e/0x90
[ 1486.160047]  [<c1669390>] ? ip6_rcv_finish+0xa0/0xa0
[ 1486.160047]  [<c1669319>] ? ip6_rcv_finish+0x29/0xa0
[ 1486.160047]  [<c166990b>] ? ipv6_rcv+0x24b/0x330
[ 1486.160047]  [<c16692f0>] ? ip6_make_skb+0x1a0/0x1a0
[ 1486.160047]  [<c15e06ed>] ? __netif_receive_skb_core+0x5ad/0x800
[ 1486.160047]  [<c15e20af>] ? netif_receive_skb_internal+0x1f/0x80
[ 1486.160047]  [<c16a169e>] ? br_pass_frame_up+0xae/0xd0
[ 1486.160047]  [<c16a15b0>] ? br_handle_local_finish+0x50/0x50
[ 1486.160047]  [<c16a1796>] ? br_handle_frame_finish+0xd6/0x500
[ 1486.160047]  [<c160aa03>] ? nf_hook_slow+0x53/0xa0
[ 1486.160047]  [<c16a1dde>] ? br_handle_frame+0x21e/0x390
[ 1486.160047]  [<c16a16c0>] ? br_pass_frame_up+0xd0/0xd0
[ 1486.160047]  [<c16a1bc0>] ? br_handle_frame_finish+0x500/0x500
[ 1486.160047]  [<c15e0400>] ? __netif_receive_skb_core+0x2c0/0x800
[ 1486.160047]  [<c15e31f9>] ? process_backlog+0x89/0x110
[ 1486.160047]  [<c15e308f>] ? net_rx_action+0x15f/0x240
[ 1486.160047]  [<c1089bad>] ? __do_softirq+0xad/0x190
[ 1486.160047]  [<c1089b00>] ? cpu_callback+0x130/0x130
[ 1486.160047]  [<c1047fcc>] ? do_softirq_own_stack+0x1c/0x30
[ 1486.160047]  <IRQ>  [<c1089da5>] ? irq_exit+0x65/0x70
[ 1486.160047]  [<c10479f7>] ? do_IRQ+0x37/0xb0
[ 1486.160047]  [<c16b71b0>] ? common_interrupt+0x30/0x38
[ 1486.160047]  [<c107b0f2>] ? native_safe_halt+0x2/0x10
[ 1486.160047]  [<c104daf5>] ? default_idle+0x5/0x10
[ 1486.160047]  [<c104df66>] ? arch_cpu_idle+0x6/0x10
[ 1486.160047]  [<c10b528a>] ? cpu_startup_entry+0x16a/0x250
[ 1486.160047]  [<c199da88>] ? start_kernel+0x38c/0x393
[ 1486.196948] ---[ end trace 7da91eb56edd56ab ]---
```
The autoupdater continues to work fine, i.e., if there is an update available, it is downloaded and installed correctly.

This issue is pretty much reproducible, but it only happens for the first `autoupdater -f` after a reboot. Our firmware is based on v2017.1.2.

NeoRaider, 2017-10-02:
Thanks for the report.

Unfortunately, the stack trace is not very informative, as it doesn't show where the frame causing the warning is coming from. I'm assuming the frames are received over batman-adv; is there a tunneldigger VPN under batman-adv, or is Mesh-on-LAN/WAN used? What driver does the underlying Ethernet interface use?

RalfJung, 2017-10-02:
> is there a tunneldigger VPN under batman-adv, or is Mesh-on-LAN/WAN used?

This is using batman-adv over tunneldigger. You can find the original VM image at <http://mgmt.saar.freifunk.net/firmware/1.5.0~rc2/factory/gluon-ffsaar-1.5.0~rc2-x86-generic.img.gz>; maybe this is reproducible on other hosts. I suppose I could also upload the HDD image of the VM causing the issue here, if you think that'd be helpful.

> What driver does the underlying Ethernet interface use?

in virt-manager, this is a "virtio" network card. Is that what you are referring to?

NeoRaider, 2018-02-15:
Could you check if this issue is still reproducible with the Gluon "next" branch based on the OpenWrt master?

NeoRaider, 2018-02-26:
If the error has disappeared with v2017.1.5, it is likely that it is fixed in recent kernel versions; closing until further notice.

The manifest in master and next contains an image size field now; the new autoupdater can not handle v2017.1.x manifests.

closed 2018-02-26 by NeoRaider