freifunk-gluon/gluon #1236: Document site.conf:autoupdater.branch vs. GLUON_BRANCH

RalfJung, 2017-10-12:
The GLUON_BRANCH configuration variable sets the default autoupdater branch:
> GLUON_BRANCH
    Sets the default branch of the autoupdater. If unset, the autoupdater is disabled by default. For the make manifest command, GLUON_BRANCH defines the branch to generate a manifest for.

However, there also is a setting in the site.conf that supposedly does the same:

>     -- Default branch. Don't forget to set GLUON_BRANCH when building!
>     branch = 'stable',

So does one of the two variables override the other? Or is it the case that they have to both be set to the same value?  Or is the one in site.conf entirely ignored? The documentation does not explain this at all. So far, it seems to me that the one in site.conf has no effect.

rotanid, 2017-10-12:
the site.conf isn't used during build by the build process, it is the configuration for the systems that are set up using the built images.
therefore, you have to set GLUON_BRANCH for the build process and branch in the site.conf.
you could possibly set those to different values, maybe to change the default branch (e.g. currently experimental and rename to nightly)

NeoRaider, 2017-10-13:
Hmm yes, the documentation could be clearer on this point. The rules are:

* If set, GLUON_BRANCH overrides site.autoupdater.branch
* The autoupdater is enabled by default if and only if GLUON_BRANCH is set during build

RalfJung, 2017-10-13:
@NeoRaider so when would it ever be useful to set `site.autoupdater.branch` then? Only in situations where a branch is supposed to be set but the updater is disabled, it seems. Which then makes me wonder why to even bother setting a branch.

NeoRaider, 2017-10-13:
Setting `site.autoupdater.branch` is necessary, as the setup scripts need to set *some* branch (as the UCI field can't be left empty), even when GLUON_BRANCH is unset (for example during development on Gluon).

For Gluon setups that prefer to provide the autoupdater as an optional extra rather than the default, it can still be useful to explicitly set a default branch in site.conf.

CodeFetch, 2017-11-20:
@NeoRaider Does that mean that you cannot e.g. set site.autoupdater.branch to stable while setting GLUON_BRANCH to experimental to get images which can be used to update nodes from experimental to stable firmware (setting uci branch to stable on sysupgrade)? Because that's a feature I'd really like to have...

belzebub40k, 2017-11-20:
You can set different a GLUON_BRANCH for `make` and `make manifest`. For example I build our beta Firmware with _GLUON_BRANCH=stable_ but generate a manifest with _GLUON_BRANCH=beta_. This way I get images that default to the autoupdater branch stable but have a valid manifest to be deployed on the beta branch. Plus once they are proven to be stable I can simple regenerate the the manifest with BRANCH=stable and I have a stable release.

CodeFetch, 2017-11-22:
@belzebub40k I looked at the code and as far as I understand it, your suggestion should not work as I would have expected it.

If you take e.g. a beta image and generate a manifest with GLUON_BRANCH=stable the resulting image can be used to update nodes which have set autoupdater branch to beta, but it won't change the autoupdater setting on the node. That means that the next time the node looks for an update, it will still look in the beta branch and not stable.

What I want is to be able to change the autoupdater branch (e.g. migrating from experimental to stable firmware).

Look at https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-autoupdater/luasrc/lib/gluon/upgrade/500-autoupdater

```
if not uci:get('autoupdater', 'settings') then
	local enabled = false
	local branch = site.autoupdater.branch()

	local f = io.open('/lib/gluon/autoupdater/default_branch')
	if f then
		enabled = true
		branch = f:read('*line')
		f:close()
	end

	uci:section('autoupdater', 'autoupdater', 'settings', {
		enabled = enabled,
		branch = branch,
	})
end

```
The file 'default_branch' contains the GLUON_BRANCH value (set in the makefile for gluon-updater using `echo '$(call qstrip,$(CONFIG_GLUON_BRANCH))' > $(1)/lib/gluon/autoupdater/default_branch`).

Thus the upgrade script behaves like NeoRaider explained:
If there are uci autoupdater settings already, keep them as they are. 
Else if GLUON_BRANCH was set, enable the autoupdater and set branch to GLUON_BRANCH value.
Else disable the autoupdater and set branch to site value. 

My suggestion for allowing migration to another branch is following:
Add an optional bool site.conf parameter 'override_branch' in check_site.lua:
need_boolean('autoupdater.override_branch', false)

And change in 500-autoupdater
`if not uci:get('autoupdater', 'settings') then` 
to 
```
if site.autoupdater.override_branch or not uci:get('autoupdater', 'settings') then
  uci:delete('autoupdater', 'settings')
```
.

There may be better ways to achieve this, but we really need such a feature as there are ugly workarounds around already: https://forum.freifunk.net/t/firmware-modul-das-den-autoupdater-branch-auf-stable-setzt/13465.

NeoRaider, 2017-11-22:
@CodeFetch We usually don't provide any means to overwrite settings that may have been made by the user. You can provide an image with a default branch A and distribute it in the autoupdater branch B, but we never change the autoupdater branch automatically.

This policy is a result of UCI not giving us any way to determine if a setting was made by the user, or scripts of an older firmware version. While I hope we'll be able to replace UCI with a more powerful configuration system eventually, other development has higher priority at the moment.

belzebub40k, 2017-11-22:
@CodeFetch You are right. I've got you question wrong. If you want to change the set branch on existing nodes you can add a custom package that changes the UCI settings like we did to move nodes from our previous beta branch to the new testing.

https://github.com/freifunk-mwu/packages-ffmwu/tree/master/ffmwu-beta-to-testing

CodeFetch, 2017-11-22:
@NeoRaider 
> You can provide an image with a default branch A and distribute it in the autoupdater branch B

This does not cover the use-case I'm talking about. 
Sorry, but I don't understand the problem. It makes no difference to me whether a user or a script set this value. 
The use-case is: Users installed experimental firmware because there was a lack of support for their device in the stable branch. Now I have X devices hanging in my experimental branch which never had the intention to test experimental firmware, but only to get their nodes up and running as soon as possible. 

We need a way to migrate these nodes to stable when their device support becomes stable.
What would you do? As far as I can see you can only use dirty hacks or send a circular mail to all node owners with experimental firmware (which is no option to me and would fail if someone hasn't declared his/her address anyway).
If you mean that the choice of branch needs to be up to the user this will still be the case if override_branch isn't used... To me this is a feature and does not break anything.

In our community we plan to have beta, experimental and stable firmware. Experimental is for newly supported devices and we'd like to be able to update them to stable when device support becomes stable. Whereas beta firmware is for people who really wan't to do beta-testing and we won't ever need to use the override_branch setting there.

belzebub40k, 2017-11-22:
Freifunk Darmstadt build a package to cover what you are asking for but I can't see that they are using it. Maybe @mweinelt can shed some light on this.

https://github.com/freifunk-darmstadt/ffda-packages/tree/master/ffda-update-stabilizer

CodeFetch, 2017-11-22:
@belzebub40k That's a nice package. Still I think this should be a standard gluon-autoupdater-feature and we want to be able to build nightly images using a build bot and not manually append new model names to a package/change package feed commit id each time.

belzebub40k, 2017-11-22:
I agree having this as a standard Gluon package would be nice.

rotanid, 2017-11-22:
maybe you like this package more: https://github.com/tecff/gluon-packages/tree/master/tecff-autoupdater-to-stable
also solves the issue, we use it.

rubo77, 2018-10-24:
We could change the behaviour of Gluon in the way that, If the setting of a default branch in site.conf is missing (because you never plan to build a firmware without using GLUON_BRANCH during build) there could be **taken the** alphabetical **first** branch name from the autoupdater mirror list in the site.conf.

this way we could leace the option in site.conf out and prevent confusion.

NeoRaider, 2020-07-04:
How about the following changes:

- The site.conf branch settings becomes optional, defaulting to the first branch
- GLUON_AUTOUPDATER_BRANCH overrides the site.conf branch, but does not enable the autoupdater by default
- GLUON_AUTOUPDATER_ENABLE enables the autoupdater by default
- GLUON_BRANCH sets GLUON_AUTOUPDATER_BRANCH and GLUON_AUTOUPDATER_ENABLE for backwards compatiblity (possibly with a deprecation message)

RalfJung, 2020-07-04:
That makes a lot of sense. :)

PR #2071 (2020-08-15): Replace GLUON_BRANCH with GLUON_AUTOUPDATER_BRANCH and GLUON_AUTOUPDATER_ENABLED
The old setting is still supported, but will emit a deprecation warning.

Also, make the branch setting in `site.conf` optional.

Closes: #1236
closed 2020-08-15 by mweinelt