freifunk-gluon/gluon #1240: New dot release to patch KRACK

J0WI, 2017-10-17:
Please release a new version to patch the [KRACK](https://www.krackattacks.com/) vulnerability.
While free/open networks are not affected, the "private wlan" functionality of Gluon is.

Patches have already been merged in the LEDE project and they will soon ship a new stable release.

rotanid, 2017-10-17:
they didn't release yet because they aren't finished patching from what i read on their IRC.
so i guess gluon is waiting for them being finished

NeoRaider, 2017-10-18:
Please note that most of the KRACK attack variants must be mitigated on the client side. https://w1.fi/security/2017-1/wpa-packet-number-reuse-with-replayed-messages.txt lists only two issues that affect the AP side - one of them requiring 802.11r to be enabled (which is not the case in Gluon's private WLAN config), and the other depending on a supplicant implementation with a specific broken behaviour (the existence of such implementations is speculative).

Hence, the impact of KRACK on the supported Gluon feature set is very low. We will still make a new release very soon, to fix the issues for users with unusual deployments that use nodes as WLAN clients.

rubo77, 2017-10-18:
Couldn't we offer an extra special private WiFi that only forward TLS connections? 

Https and starttls only!

H4ndl3, 2017-10-18:
rubo77, I don't see how that would be beneficial or even necessary. But you can implement that using iptables if you want.

0x4A6F, 2017-10-19:
We might consider adding an option to implement this [hostapd commit](https://w1.fi/cgit/hostap/commit/?id=6f234c1e2ee1ede29f2412b7012b3345ed8e52d3).

J0WI, 2017-10-20:
LEDE 17.01.4 has been released: https://lede-project.org/releases/17.01/changelog-17.01.4#security_fixes

NeoRaider, 2017-10-23:
The Gluon v2017.1.x branch now contains the KRACK fixes.

The workaround mentioned by @0x4A6F can be configured using UCI after updating (see https://git.lede-project.org/?p=source.git;a=commitdiff;h=d501786ff25684208d22b7c93ce60c194327c771 ); I don't think we need to provide a graphical knob for the workaround (and enabling it is probably a bad idea, as it will make the WPA handshake less robust in lossy environments, and having a mesh network running on the same channel will definitely cause some interference...).

I'm still waiting for the kmod-jool build fix to be acked upstream (reported on Gluon ML), and plan to start preparing the next Gluon release after that.

rubo77, 2017-10-23:
What do you mean with a graphical knob?

NeoRaider, 2017-10-23:
@rubo77 A switch in the web GUI.

J0WI, 2017-11-11:
> I'm still waiting for the kmod-jool build fix to be acked upstream (reported on Gluon ML), and plan to start preparing the next Gluon release after that.

Is there an update on this? Do you consider a fix-KRACK only release if other patches are blocked upstream?

rotanid, 2017-11-11:
@J0WI that's not an issue anymore, he already prepared the release notes: #1251

closed 2017-11-15 by NeoRaider