freifunk-gluon/gluon #1242: Devices that don't support IBSS shouldn't be marked as BROKEN

mweinelt, 2017-10-20:
Currently four targets (ralink/mediatek) are marked as broken with the remark:
> BROKEN: No AP+IBSS support

This is not necessarily anything that will ever be resolved.

Since 802.11s is working quite well and seems to be the successor of IBSS those devices are hardly broken, but only useful if the site already uses 802.11s mesh networking. In turn these could be marked conditional with a flag like `MESH_11s` instead of `BROKEN`.

rubo77, 2017-10-21:
You could also add a tag 2.4GHZ_ONLY i.e. for the C25

rotanid, 2017-10-21:
@rubo77 please, separate issue, no derailing ;)

NeoRaider, 2017-10-21:
I think a nice solution would be to change the *GLUON_ATH10K_MESH* variable to a generic *GLUON_WLAN_MESH*. It would default to 11s, skip building images for models that don't support the chosen mode (unless *BROKEN* is set), and select ath10k firmwares.

FreifunkUFO, 2018-01-04:
yes, blame ath10k to NOT support ad-hoc.. 
we need more compatibility < https://github.com/libremesh/chef/issues/24

rotanid, 2018-01-07:
FYI: @2tata created a patch #1298 to fix this issue. maybe take a look at it to check if it works for you, too.

PR #1298 (2018-01-10): build: introduce GLUON_WLAN_MESH build variable
List additional targets if GLUON_WLAN_MESH=11s (default is 11s)

Closes #1242
closed 2018-01-10 by rotanid