freifunk-gluon/gluon #1243: High load on some devices after 2017.1.x update

mweinelt, 2017-10-22:
The devices were fine before, but with 2017.1.x high loads appeared. They seem to originate somewhere in the kernelspace.

Only some models are affected, but then all devices of that model experience this issue:

- TP-Link WR842ND v2
- TP-Link WR1043ND v1
- Ubiquiti Nanostation Loco M2

Probably more, but since our Grafana is currently down it's cumbersome to find more.

NeoRaider, 2017-10-22:
I've seen the issue on a WR841ND (I think it was v9 or v10). Possibly, all models with 32MB are affected?

mweinelt, 2017-10-22:
Some 841 V9 are affected, not all though.
The 841 V10 looks unaffected.

rotanid, 2017-10-22:
we also noticed that devices with more mesh neighbours are more likely to be affected - what a surprise!

Tarnatos, 2017-10-23:
I can confirm that dir Freifunk Nord

mweinelt, 2017-10-23:
Devices on the FFDA network with load > 2.0

```json
{
  "TP-Link TL-MR3420 v1": [
    8.63
  ],
  "TP-Link TL-WR841N/ND v11": [
    7.85,
    12.47,
    3.44,
    3.9
  ],
  "TP-Link TL-WR940N v4": [
    4.48
  ],
  "Ubiquiti AirRouter": [
    3.09
  ],
  "TP-Link TL-WR710N v1": [
    6.2
  ],
  "TP-Link TL-WR842N/ND v2": [
    2.28,
    12.43,
    8.98,
    15.45,
    8.98,
    3.47,
    7.22,
    10.07,
    12.14,
    9.2,
    7.25,
    8.48,
    2.91,
    6.22,
    8.91,
    3.14,
    2.13,
    9.9,
    10.92
  ],
  "Ubiquiti PicoStation M2": [
    9.97
  ],
  "Linksys WRT160NL": [
    4.69
  ],
  "TP-Link TL-WR710N v2.1": [
    4.19,
    11.26
  ],
  "TP-Link TL-WR1043N/ND v1": [
    6.98,
    3.57,
    6.25,
    2.23,
    5.62
  ],
  "TP-Link TL-WR841N/ND v9": [
    4.98,
    4.55,
    4.08,
    2.67,
    4.89,
    2.57,
    9.66,
    7.81
  ],
  "TP-Link TL-WA850RE v1": [
    7.71
  ],
  "TP-Link TL-WR841N/ND v10": [
    2.87,
    5.32,
    5.59
  ],
  "TP-Link TL-WA901N/ND v3": [
    7.96
  ],
  "Ubiquiti NanoStation loco M2": [
    4.94
  ],
  "TP-Link TL-WR842N/ND v1": [
    2.49
  ]
}
```

blocktrron, 2017-10-23:
`
`

blocktrron, 2017-10-23:
Affected nodes on the FFDA Network grouped by SoC:

## AR9341
TP-Link TL-WR842N/ND v2 19
TP-Link TL-WA801N/ND v2 1
TP-Link TL-WA850RE v1 1

## QCA9533
TP-Link TL-WR841N/ND v9 7
TP-Link TL-WR841N/ND v10 5
TP-Link TL-WR841N/ND v11 2

## AR9132
TP-Link TL-WR1043N/ND v1 4

## AR9331
TP-Link TL-WR710N v2.1 2
TP-Link TL-WR710N v1 1

## AR7240
Ubiquiti NanoStation loco M2 1
Ubiquiti PicoStation M2 1

## AR7241
TP-Link TL-MR3420 v1 1
Ubiquiti AirRouter 1

## AR9130
Linksys WRT160NL 1

A-Kasper, 2017-10-25:
Hi!

I don't got any nodes with a higher load than 0.8 in our network. 
we've had these problems when using batman 2017.x first time.. but then a bug with multicast optimization in batman-adv was found. After disabling this in firmware AND at all gateways the load was going down.

Could you please provide these information:
1. How many Nodes are on old firmware in your network? (maybe someone can add the first gluon Version without MO to name the affected versions....)
2. Which Batmand-adv is running on your Gateways?
3. Is Multicast Optimisation activated on gateway level?

I think this could be related. Maybe you could try to update your batman-adv gateways and disable multicast optimizations with "batctl mm 0" (don't forget your mapserver ;))

If the behaviour is related, you could try to eliminate the fulltable orgy by disabling all vpn tunnels for a minute to make the mismarked packages disappear

mweinelt, 2017-10-25:
1. ~700 new (v2017.1.3), ~30 old (v2017.1.2 and earlier)
2. batman-adv v2017.3
3. yes

Multicast optimizations are still disabled on v2017.1.x, see https://github.com/freifunk-gluon/gluon/commit/c6a3afa1301d0edda4a7eb69a8fc64af220d71d8.

A-Kasper, 2017-10-25:
I think the MO Bug is still not addressed. As already mentioned our load is in normal range (but we have a smaller network). If  you are able to, I would like to ask you to check if the load goes down if you disable MO at all gateways via batctl mm 0 i would like to make sure it's not still this mo thing..... 

Our Network went stable after ALL Sources of mismarked packages were eliminated...  Gateways had the biggest effect. I'm not sure, but if my memories are correct it was you who had a look at our network dump... if so: can you see these full table requests?

mweinelt, 2017-10-25:
This issue was fixed in https://git.open-mesh.org/batman-adv.git/commit/382d020fe3fa528b1f65f8107df8fc023eb8cacb, no?

A-Kasper, 2017-10-25:
Am 25.10.2017 14:53 schrieb "hexa-" <notifications@github.com>:

This issue was fixed in https://git.open-mesh.org/batman-adv.git/commit/
382d020fe3fa528b1f65f8107df8fc023eb8cacb, no?

My Bauchgefühl still says that a part of the problem is still not fixed.

Wirh mo disabled everwhere we don`t got the load problems. thats why i have
asked you for trying to disable it.

The other idea is,  that the mismarked packages from some nodes are enough
to corrubtvthe table of fixed batman versions.

I can't say if this is the point. i just want to make sure that ithis is
not related.

The fix in batman-adv is something like untested. the problem appears just
in big networks,  so the feedback goes to zero. They consider it is
fixed... but i don't think that someone dumped the traffic.

I can sure tell you that broken nodes affect the load of the fixed nodes.

mweinelt, 2017-10-27:
Disabled `mm` on our gateways.

Looking at this node for example at this time of day (3:00)
https://meshviewer.darmstadt.freifunk.net/#/de/map/30b5c2c2ead4

- Model 841 v9
- Load average 4,83
- RAM 74,7% used
   ```json
  "memory": {
    "total": 27808,
    "free": 5236,
    "buffers": 992,
    "cached": 1696
  },
    ```
- Clients 0
- Traffic neglible

5,2 MB of "free" RAM does not seem to be enough if we suspect the issue arises due to high memory fragmentation.

A-Kasper, 2017-10-27:
thank you for testing.
i'll have a look at free mem tomorrow. I'm interested in why we don't got
this problem anymore.....

rotanid, 2017-10-27:
@A-Kasper 
1) please try to not comment if you have nothing new to contribute. this makes the whole issue harder and harder to read. you could have waited with your comment until you had a look at your mem.
2) how many nodes does you network have? what about large groups of wireless-meshing-nodes? maybe your network is simply too small to have this issue.

CodeFetch, 2017-10-31:
@mweinelt Do you have any custom ash-scripts running on your nodes?
I have had an ash-script running and my node got slower and slower until logging in via ssh took about a minute. The same happened when a monitoring script logged into the router via SSH and just executed some commands. That's why I think it could be a problem with ash/Busybox. With only lua scripts running on the router executed by micrond everything works as expected.

mweinelt, 2017-10-31:
@CodeFetch No, we're not running anything custom.

rotanid, 2017-11-06:
perhaps this issue is actually the same as #753 - only that it appears earlier than before

mweinelt, 2017-11-06:
Looking at the issue by SoC alone does not seem to yield a clear result.

```json
[
  {
    "family": "AR9531",
    "loadavg": 0.17,
    "devices": {
      "TP-Link TL-WR842N/ND v3": 0.17
    }
  },
  {
    "family": "QCA9558",
    "loadavg": 0.18,
    "devices": {
      "TP-Link TL-WR1043N/ND v3": 0.16,
      "TP-Link Archer C7 v2": 0.19,
      "TP-Link TL-WR1043N/ND v2": 0.2,
      "TP-Link Archer C5 v1": 0.21
    }
  },
  {
    "family": "QCA9563",
    "loadavg": 0.19,
    "devices": {
      "TP-Link TL-WR1043N/ND v4": 0.19
    }
  },
  {
    "family": "AR7240",
    "loadavg": 0.67,
    "devices": {
      "Ubiquiti NanoStation loco M2": 0.59,
      "Ubiquiti NanoStation M2": 1.61,
      "TP-Link TL-WA801N/ND v1": 1.25,
      "TP-Link TL-WR740N/ND v4": 1.18,
      "TP-Link TL-WR740N/ND v1": 0.3,
      "TP-Link TL-WA901N/ND v1": 0.04,
      "TP-Link TL-WA830RE v1": 0.17,
      "TP-Link TL-WR741N/ND v1": 0.41,
      "TP-Link TL-WR841N/ND v5": 0.06
    }
  },
  {
    "family": "QCA9533",
    "loadavg": 1.05,
    "devices": {
      "TP-Link TL-WR841N/ND v11": 0.79,
      "TP-Link TL-WR841N/ND v10": 1.18,
      "TP-Link TL-WR841N/ND v9": 1.1
    }
  },
  {
    "family": "AR2316A",
    "loadavg": 1.49,
    "devices": {
      "Ubiquiti PicoStation M2": 1.49
    }
  },
  {
    "family": "AR9344",
    "loadavg": 0.21,
    "devices": {
      "TP-Link CPE210 v1.1": 0.21,
      "TP-Link TL-WDR4300 v1": 0.2,
      "TP-Link CPE210 v1.0": 0.24,
      "TP-Link TL-WDR3600 v1": 0.17
    }
  },
  {
    "family": "AR9341",
    "loadavg": 2.5,
    "devices": {
      "TP-Link TL-WR842N/ND v2": 3.5,
      "TP-Link TL-WA850RE v1": 1.42,
      "TP-Link TL-WR841N/ND v8": 1.25,
      "TP-Link TL-WA801N/ND v2": 0.18,
      "TP-Link TL-WR941N/ND v5": 0.07,
      "TP-Link TL-WA901N/ND v3": 2.26,
      "TP-Link TL-WA860RE v1": 0.15
    }
  },
  {
    "family": "AR9331",
    "loadavg": 2.76,
    "devices": {
      "TP-Link TL-WR710N v2.1": 6.76,
      "TP-Link TL-MR3020 v1": 1.15,
      "TP-Link TL-WR710N v1": 3.62,
      "TP-Link TL-WR741N/ND v4": 0.21,
      "TP-Link TL-WR710N v2": 0.12,
      "TP-Link TL-WA701N/ND v2": 0.61
    }
  },
  {
    "family": "AR9132",
    "loadavg": 1.95,
    "devices": {
      "TP-Link TL-WR1043N/ND v1": 2.51,
      "TP-Link TL-WR941N/ND v2": 0.18,
      "TP-Link TL-WA901N/ND v2": 0.11
    }
  },
  {
    "family": "AR7241",
    "loadavg": 2.56,
    "devices": {
      "TP-Link TL-MR3420 v1": 2.78,
      "TP-Link TL-WR842N/ND v1": 2.12,
      "Ubiquiti AirRouter": 2.77
    }
  },
  {
    "family": "TP9343",
    "loadavg": 1.06,
    "devices": {
      "TP-Link TL-WR941N/ND v6": 0.44,
      "TP-Link TL-WR940N v4": 2.07,
      "TP-Link TL-WA901N/ND v4": 0.07
    }
  },
  {
    "family": "MT7621AT",
    "loadavg": 0.08,
    "devices": {
      "D-Link DIR-860L B1": 0.08
    }
  },
  {
    "family": "AR7161",
    "loadavg": 0.26,
    "devices": {
      "Buffalo WZR-HP-AG300H/WZR-600DHP": 0.26
    }
  },
  {
    "family": "AR1311",
    "loadavg": 0.21,
    "devices": {
      "D-Link DIR-505 rev. A2": 0.21
    }
  },
  {
    "family": "AR9350",
    "loadavg": 0.15,
    "devices": {
      "TP-Link CPE510 v1.0": 0.15,
      "TP-Link CPE510 v1.1": 0.14
    }
  },
  {
    "family": "AR9130",
    "loadavg": 7.59,
    "devices": {
      "Linksys WRT160NL": 7.59
    }
  },
  {
    "family": "AR9342",
    "loadavg": 0.02,
    "devices": {
      "Ubiquiti Loco M XW": 0.02
    }
  }
]
```

edeso, 2017-11-10:
hey All,

on an Ubiquiti NanoStation loco M2 the issue seems to go away when the mesh wlan is deactivated. at least it looks that way since 2 days.

this is kbu freifunk, where the wireless mesh config looks like this

```
config wifi-iface 'ibss_radio0'
        option ifname 'ibss0'
        option network 'ibss_radio0'
        option device 'radio0'
        option bssid '02:d2:22:01:fc:22'
        option disabled '1'
        option mcast_rate '12000'
        option mode 'adhoc'
        option macaddr '42:84:e7:d9:c1:32'
        option ssid '02:d2:22:01:fc:22'
```
..ede

mweinelt, 2017-11-10:
Does this NSM2 Loco have a VPN connection or is it otherwise connected to the Mesh after disabling the WiFi-Mesh?

How big is the batadv L2 domain? Originators? Transtable (global) size?

T-X, 2017-11-13:
At least the three devices in the original post are all 32MB RAM _and_ 8MB flash devices. Is this ticket a duplicate of #1197 maybe? Or could we somehow separate these two tickets more clearly?

Just a crazy idea... As decently fast microSD cards seem to have gotten quite cheap: I'd be curious whether attaching some flash storage to the USB port of a router and configuring one partition for swap and one for /tmp/ would make a difference. For instance [this](https://www.reichelt.de/SD-Karten-Micro-/SDSDQUAN-008GG4A/3/index.html?ACTION=3&GROUPID=4802&ARTICLE=148605&SEARCH=SDSDQUAN-008GG4A&START=0&OFFSET=16&) plus [this](https://www.reichelt.de/Kartenleser-und-Adapter/DELOCK-91648/3/index.html?ACTION=3&GROUPID=5262&ARTICLE=78457&SEARCH=DELOCK%2B91648&START=0&OFFSET=16&) would cost less than 10€. Maybe there's even a decently fast, usable USB flash stick for less than 5€. Not suggesting this as a fix, but curious whether that'd change anything.

Also, if some people with devices constantly having high loads could recompile with CONFIG_KERNEL_SLABINFO=y and could dump /proc/slabinfo, that could be helpful (I asked for this in #1197, too).

mweinelt, 2017-11-13:
We tested an image for the 1043v1 without additional USB modules, that are currently being loaded unconditionally - and the device seemed to behave fine again. 

@blocktrron can maybe tell us more about what he saw during tests.

blocktrron, 2017-11-13:
At Freifunk Darmstadt, we were able to observe that 8MB/32MB devices rebooted frequently, most likely due to the integrated USB support.

We were also able to recreate the problems (High load/crashing) on an OpenWRT based Gluon by writing to tmpfs. Crashing/High load only occured, when RAM was filled before the batman transglobal table was initialized.

mweinelt, 2017-11-14:
Fyi: We're building images from master with slabinfo enabled and without additional USB modules tonight. If we can trigger the issue with those, we'll post slabinfo, else we'll retry again with the USB modules installed,

mweinelt, 2017-11-14:
From our 1043v1 rebooting in circles:

OOM Reboot
```
[   90.450092] hotplug-call invoked oom-killer: gfp_mask=0x2420848, order=0, oom_score_adj=0
[   90.458394] CPU: 0 PID: 2327 Comm: hotplug-call Not tainted 4.4.93 #0
[   90.464869] Stack : 803e96e4 00000000 00000001 80440000 807d5764 80434e63 803ca228 00000917
	  804a378c 00001b20 00000040 00000000 00000000 800a787c 00000006 00000000
	  00000000 00000000 803cdd4c 8172199c 804a6542 800a57f8 02420848 00000000
	  00000001 801f9300 00000000 00000000 00000000 00000000 00000000 00000000
	  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
	  ...
[   90.500943] Call Trace:
[   90.503422] [<80071f1c>] show_stack+0x54/0x88
[   90.507826] [<800d498c>] dump_header.isra.4+0x48/0x130
[   90.513003] [<800d515c>] check_panic_on_oom+0x48/0x84
[   90.518102] [<800d5288>] out_of_memory+0xf0/0x324
[   90.522847] [<800d8da0>] __alloc_pages_nodemask+0x6b8/0x724
[   90.528488] [<800d1b44>] pagecache_get_page+0x154/0x278
[   90.533765] [<80136e94>] __getblk_slow+0x15c/0x374
[   90.538617] [<8015e518>] squashfs_read_data+0x1c8/0x6e8
[   90.543888] [<80162728>] squashfs_readpage_block+0x32c/0x4d8
[   90.549602] [<801603a4>] squashfs_readpage+0x5bc/0x6d0
[   90.554780] [<800dc53c>] __do_page_cache_readahead+0x1f8/0x264
[   90.560673] [<800d393c>] filemap_fault+0x1ac/0x458
[   90.565526] [<800eeb4c>] __do_fault+0x3c/0xa8
[   90.569925] [<800f1d84>] handle_mm_fault+0x478/0xb14
[   90.574934] [<80076be8>] __do_page_fault+0x134/0x470
[   90.579944] [<80060820>] ret_from_exception+0x0/0x10
[   90.584933] 
[   90.586446] Mem-Info:
[   90.588769] active_anon:820 inactive_anon:9 isolated_anon:0
[   90.588769]  active_file:136 inactive_file:154 isolated_file:0
[   90.588769]  unevictable:0 dirty:0 writeback:0 unstable:0
[   90.588769]  slab_reclaimable:211 slab_unreclaimable:3104
[   90.588769]  mapped:59 shmem:29 pagetables:104 bounce:0
[   90.588769]  free:293 free_pcp:0 free_cma:0
[   90.620556] Normal free:1172kB min:1024kB low:1280kB high:1536kB active_anon:3280kB inactive_anon:36kB active_file:544kB inactive_file:616kB unevictable:0kB isolated(anon):0kB isolated(file):0kB present:32768kB managed:27776kB mlocked:0kB dirty:0kB writeback:0kB mapped:236kB shmem:116kB slab_reclaimable:844kB slab_unreclaimable:12416kB kernel_stack:472kB pagetables:416kB unstable:0kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB writeback_tmp:0kB pages_scanned:6972 all_unreclaimable? yes
[   90.664376] lowmem_reserve[]: 0 0
[   90.667738] Normal: 49*4kB (UME) 80*8kB (UME) 13*16kB (UME) 4*32kB (M) 0*64kB 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 1172kB
[   90.680287] 319 total pagecache pages
[   90.683970] 0 pages in swap cache
[   90.687314] Swap cache stats: add 0, delete 0, find 0/0
[   90.692565] Free swap  = 0kB
[   90.695472] Total swap = 0kB
[   90.698373] 8192 pages RAM
[   90.701093] 0 pages HighMem/MovableOnly
[   90.704947] 1248 pages reserved
[   90.708117] [ pid ]   uid  tgid total_vm      rss nr_ptes nr_pmds swapents oom_score_adj name
[   90.716721] [  515]     0   515      297       49       3       0        0             0 ubusd
[   90.725384] [  516]     0   516      296       40       4       0        0             0 ash
[   90.733889] [  807]     0   807      306       68       4       0        0             0 logd
[   90.742477] [  814]     0   814      429      189       4       0        0             0 haveged
[   90.751328] [ 1055]     0  1055      447       82       4       0        0             0 netifd
[   90.760092] [ 1102]     0  1102      264       40       3       0        0             0 dropbear
[   90.769029] [ 1119]     0  1119      225       42       3       0        0             0 uradvd
[   90.777794] [ 1330]     0  1330      296       39       4       0        0             0 udhcpc
[   90.786557] [ 1332]     0  1332      254       44       3       0        0             0 odhcp6c
[   90.795396] [ 1343]     0  1343      254       49       3       0        0             0 odhcp6c
[   90.804250] [ 1487]     0  1487      225       44       4       0        0             0 micrond
[   90.813101] [ 1521]     0  1521      224       39       3       0        0             0 sse-multiplexd
[   90.822562] [ 1685]     0  1685      320       50       3       0        0             0 uhttpd
[   90.831325] [ 1794]     0  1794      383       76       3       0        0             0 hostapd
[   90.840177] [ 1809]   453  1809      353      137       4       0        0             0 dnsmasq
[   90.849033] [ 1830]     0  1830      280       52       4       0        0             0 dnsmasq
[   90.857887] [ 2116]     0  2116      320       63       4       0        0             0 fastd
[   90.866563] [ 2213]     0  2213      517       71       3       0        0             0 respondd
[   90.875502] [ 2223]     0  2223      306       50       3       0        0             0 hotplug-call
[   90.884777] [ 2295]     0  2295      296       40       3       0        0             0 ntpd
[   90.893369] [ 2326]     0  2326      327       75       4       0        0             0 dhcpv6.script
[... 105 lines cut ...]
buffer_head         2472   2484    112   36    1 : tunables    0    0    0 : slabdata     69     69      0
nsproxy                0      0     72   56    1 : tunables    0    0    0 : slabdata      0      0      0
vm_area_struct       446    540    136   30    1 : tunables    0    0    0 : slabdata     18     18      0
[... cut ...]
kmalloc-512          447    456    640   12    2 : tunables    0    0    0 : slabdata     38     38      0
kmalloc-256          359    370    384   10    1 : tunables    0    0    0 : slabdata     37     37      0
kmalloc-128         8162   8192    256   16    1 : tunables    0    0    0 : slabdata    512    512      0
kmem_cache_node      113    128    128   32    1 : tunables    0    0    0 : slabdata      4      4      0
kmem_cache           113    128    256   16    1 : tunables    0    0    0 : slabdata      8      8      0
```

mweinelt, 2017-11-14:
Behaviour does not improve when:
 - removing `fq_codel` and using `pfifo_fast`
 - disabling Airtime Fairness

Adorfer, 2017-11-15:
testing with a 1043v1 sounds a bit like "trying to fix 2 issues with one shot", since (at least everybody seems to know for sure) that 1043v1 is instable by design, even in CC and BB.  (eventhough it's just a hanging wifi, not a high load or a reboot.)

edeso, 2017-11-15:
On 10.11.2017 16:03, hexa- wrote:
> Does this NSM2 Loco have a VPN connection or is it otherwise connected to the Mesh after disabling the WiFi-Mesh?

fastd is disabled. batman meshes on wan. there is an x86 offloader doing the fastd vpn.
  
> 
> How big is the batadv L2 domain? Originators? Transtable (global) size?

not in my area of expertise. do you have some commands you want me to run?

i am talking about this node, which is w/o symptoms since then
  https://map.kbu.freifunk.net/#!v:m;n:6872513e82a7

..ede

mweinelt, 2017-11-15:
@edeso So neither VPN nor WiFi-Mesh, that in itself should reduce memory usage quite a bit. However that's not the most prominent use case of these devices, but we can try and confirm that this stabilizes the device.

@Adorfer You are acknowledging that it the WiFi hangs on said device, however this issue is entire related to memory usage, load and oom reboots. The device currently reboots every minute, so I don't think WiFi hangs play into this issue, and I can easily take the WiFi out of the equation and still have it crash.

edeso, 2017-11-15:
On 15.11.2017 13:40, hexa- wrote:
> @edeso <https://github.com/edeso> So neither VPN nor WiFi-Mesh, that in itself should reduce memory usage quite a bit. However that's not the most prominent use case of these devices, but we can try and confirm that this stabilizes the device.

well, the device _didn't_ run vpn before as well but had huge blocking load issues. all i did now was disabling mesh, just to try it out (disabling non essentials, see if something changes [1]). it's not that important for this node as it primarily serves a train station where the likelyhood of finding a mesh node is very low by nature.

also, top did report free ram, but the load was something like

'''
Mem: 19780K used, 8028K free, 92K shrd, 1288K buff, 2748K cached
CPU:   3% usr  92% sys   0% nic   0% idle   0% io   0% irq   3% sirq
Load average: 4.34 4.68 5.06 4/52 4074
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 2698     2 root     RW       0   0%  90% [kworker/u2:2]
    3     2 root     RW       0   0%   2% [ksoftirqd/0]
 4033     2 root     RW       0   0%   2% [kworker/u2:4]
'''

..ede

[1] also tried, but w/o any success load wise
- disabling the logd
- minimizing hostapd logging
- having the node restart the network periodically via cron
- having the node restart wifi periodically via cron

mweinelt, 2017-11-18:
Collected slabinfo and sysrq-m outputs on a TL-WR1043ND-v1

- 2016.2.7: https://git.darmstadt.ccc.de/snippets/11
- master (20171114): https://git.darmstadt.ccc.de/snippets/12

RalfJung, 2017-11-19:
> Unaffected:
> TP-Link WR841ND v10

I have seen occasional high load on my WR841N/ND v10 with Gluon 2017.1.x. Are you sure it is not affected?

mweinelt, 2017-11-19:
@RalfJung maybe one in ten.

edeso, 2017-11-22:
UPDATE on the node w/ disabled mesh wlan. today it showed the symptoms again and only a reboot made it go away. so obviously it only mitigates but does not solve the this issue.

..ede

2tata, 2017-12-20:
In the Nordwest Freifunk network we can detect this issue as well:
```
Mem: 19164K used, 8628K free, 116K shrd, 788K buff, 1316K cached
CPU:   1% usr  90% sys   0% nic   0% idle   0% io   0% irq   8% sirq
Load average: 5.13 4.66 3.03 4/60 8719
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 7412     2 root     RW       0   0%  21% [kworker/0:0]
 3063     1 root     R     2148   8%  10% /usr/bin/respondd -d /usr/lib/respond
 8704  8285 root     R     1184   4%   5% top
 8680  8677 root     S     1932   7%   5% {hoodselector} /usr/bin/lua /usr/sbin
 2415     1 root     S     1276   5%   4% /usr/bin/fastd --config - --daemon --
 2871     1 root     S     1532   6%   3% /usr/sbin/hostapd -s -P /var/run/wifi
 2573  1062 root     S     1016   4%   3% odhcp6c -s /lib/netifd/dhcpv6.script
    3     2 root     SW       0   0%   2% [ksoftirqd/0]
  106     2 root     SW       0   0%   2% [kswapd0]
```
 * Device: TP-Link TL-WR841N/ND v11
 * Gluon-Version: v2017.1.x
 * Commit ID: 2ae74fe737a45e589f85ace098fa8466dfb7927e
 * Download: https://firmware.ffnw.de/20171216

mweinelt, 2017-12-21:
fyi:

```
2017-12-12 00:11:47 blocktrron T_X: echo fq_memory_limit 200 > /sys/kernel/debug/ieee80211/phy0/aqm resultiert darin, dass der Router stabil läuft
```

2tata, 2017-12-21:
On 12/21/17 02:01, Martin Weinelt wrote:
> fyi:
> 
> ```
> 2017-12-12 00:11:47 blocktrron T_X: echo fq_memory_limit 200 > /sys/kernel/debug/ieee80211/phy0/aqm resultiert darin, dass der Router stabil läuft
> ```

Ay, Danker dir.
Wir konnten das lösen, indem wir das batman-adv auf den gateways auf höher 2017.0.1 upgegaraded. 

vg
Tarek

mweinelt, 2017-12-21:
We are currently on batman-adv v2017.3, shortly v2017.4. I don't think it will fix this issue for us, seeing as the workaround in https://github.com/freifunk-gluon/gluon/issues/1243#issuecomment-353227388 reduces the memory of some 802.11 queuing mechanism.

A-Kasper, 2017-12-21:
I don't know what batman-adv you where using before, but it could be that batman-adv v2017.3 fixed the problem. 
There was a bug with multicast optimization in batman-adv before. This bug leaded to many small packages. your workaround helps to handle more small packages. but maybe your router wouldn't need this workaround in "normal" conditions without the batman failure. 
we had got these problems until we upgraded all batman-adv on all gateways and disabled multicast optimizations with batctl mm 0
I didn't had time yet to test if batctl mm 1 works now. I disabled it after the problems persisted. Later I found out, that I missed to upgrade the mapserver, so the mapserver still produced misflagged packets.

see: https://patchwork.open-mesh.org/patch/17072/

can you confirm this issue in a network with => batman-adv 2017.3 nodes only or with mm disabled also on gateways and mapserver? maybe you fix to work with existing error instead of removing the error

mweinelt, 2017-12-21:
That issue was already fixed in 2017.0.1, which we rolled out some time back in may. We do follow batman-adv quite closely.

rotanid, 2017-12-27:
we just noticed the debug-oom branch wasn't mentioned here before:
https://github.com/freifunk-gluon/gluon/tree/debug-oom
did this branch alone not help? or did i just miss comments regarding this?
or did it help even more together with the manual "200" limit?

edeso, 2017-12-27:
`echo fq_memory_limit 200 > /sys/kernel/debug/ieee80211/phy0/aqm` worked for me on an ubiquiti loco m2 so far for 3 days. testing higher limits now.

..ede

rotanid, 2017-12-28:
@edeso in our meeting the question also was, whether the autoupdater is working again with the lower limit of 200 - or if that still leads to a crash/load/OOM issue.

edeso, 2017-12-28:
@rotanid sorry can't help you with that. the gluon on these nodes does not have a valid autoupdate config. any way to simulate one? i would be willing to try.

..ede

rotanid, 2017-12-28:
yes, you could write a file (consisting of zeroes) of slowly increasing size to /tmp/ and see if you get to a size big enough that it is the same size as an sysupgrade file would be

edeso, 2017-12-28:
like a dd of say 4MB from /dev/zeroes to /tmp/test ? if the device reboots bad, if not success?

..ede

rotanid, 2017-12-28:
@edeso worst devices are the ones with 32mb ram and 8mb flash, like WR842v2, its sysupgrade image has 5046276 byte (similar Loco M2 and some others)
yes, if it doesn't reboot it's good. but the autoupdater already stops some services to save memory, so less than this byte count is not always bad.

edeso, 2017-12-28:
@rotanid ok. will try as soon as i have time and ipv6 again. probably in 2-3 days.. ede

edeso, 2017-12-28:
@rotanid found a shell w/ ipv6 and tried it (sitting in a db train right now). codel limit is currently 512 on this node. tried 1-5MB and that worked. 8MB rebooted the box. after a fresh start 8MB works even w/o the memory limit set, box get's a bit laggy though, so it seems to matter how long the box ran before the mounted ram is used.

```
:~# cat /sys/kernel/debug/ieee80211/phy0/aqm
access name value
R fq_flows_cnt 4096
R fq_backlog 0
R fq_overlimit 0
R fq_overmemory 40295713
R fq_collisions 67
R fq_memory_usage 0
RW fq_memory_limit 512
RW fq_limit 8192
RW fq_quantum 300

:~# dd bs=1024 count=$((5*1024)) if=/dev/zero of=/tmp/test
5120+0 records in
5120+0 records out

:~# ls -lah /tmp/test
-rw-r--r--    1 root     root        5.0M Dec 28 02:29 /tmp/test

:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 2.3M      2.3M         0 100% /rom
tmpfs                    13.6M      5.1M      8.5M  38% /tmp
/dev/mtdblock5            3.8M    372.0K      3.4M  10% /overlay
overlayfs:/overlay        3.8M    372.0K      3.4M  10% /
tmpfs                   512.0K         0    512.0K   0% /dev
```

rotanid, 2017-12-28:
looks promising to me, but @T-X and @NeoRaider can interprete this much better :D

T-X, 2018-01-30:
Documenting regarding what had been found out back then with the tests done by @mweinelt and @blocktrron back then:

[This](https://paste.linuxlounge.net/#/GUMP_cQOKwgqQhf5_gBfPU1Vv0A!7bT80KaCMPcUC_zIBbw6h5s3ewQ27IFuYW5_H42HWC0) paste produced with the [slabinfo-on-OOM patch backported](https://github.com/freifunk-gluon/gluon/commit/48c834a1ccb86941cbad15084f853862fe10c6df) seems to indicate that:

* The issue is not caused by userspace
* The issue is not caused by kernelspace kmalloc()/kmem_cache_alloc() allocations. This seems to outrule:
  * batman-adv or the Linux bridge, for instance, as they are using these and not vmalloc()

MPW1412, 2018-02-08:
This issue appeared to Freifunk Münsterland in v2017.1.4 and v2017.1.5 aswell. 

Example node: http://backend-aegidiistrasse.knoten.ffmsl.de/

So far we tried updating the batman version on the two gateways. Both are running Batman 2017.4 now.

Further up it was speculated, that some broken batman versions between v2017.0 and v2017.3 could cause this issue. We don't have any of those nodes in [this domain](https://karte.freifunk-muensterland.de/map16/). There are only v2016.2.7 Gluons with Batman 2016.2 and a very few v2017.1.5 test nodes.

There are no packages included in [our configuration](https://github.com/FreiFunkMuenster/site-ffms/tree/Dom%C3%A4ne-16), which could explain this load besides the ssid changer, which used to run perfectly.

Is there anything I can do to help solving this issue? The affected nodes are barely usable. The download speed goes down to a few kilobytes per second.

Bye,
Matthias

edeso, 2018-02-09:
@MPW1412 at least there is a workaround above https://github.com/freifunk-gluon/gluon/issues/1243#issuecomment-353227388

just add it to /etc/rc.local so it get's executed on every boot.

..ede

blocktrron, 2018-02-15:
Results in the network of Freifunk Darmstadt are documented here: https://md.darmstadt.ccc.de/ffda-gluon-debugwishlist

NeoRaider, 2018-04-11:
#1363 should significantly reduce RAM usage, please test!

TomSiener, 2018-04-25:
Even with a build of the actual master branch, the high cpu load problem is still reproducable on an WR841v9 (maybe the load is little bit lower bit still near 1 withoud much traffic)

mweinelt, 2018-05-01:
As of the firmware image with kmods-builtin a problem device (842NDv2) has started reaching uptimes of several days again.

[image]

At least the last three reboots are due to updates from the master branch.

Also I have a 1043NDv1 locally that has reached an uptime of 3 hours. Note that it used to crash in seconds earlier this year.

```sh
root@64xxx-90f652f45cc4:/# uptime
 22:28:28 up  3:05,  load average: 0.19, 0.16, 0.15
root@64xxx-90f652f45cc4:/# free
             total       used       free     shared    buffers     cached
Mem:         27512      20804       6708        120       1304       2480
-/+ buffers/cache:      17020      10492
Swap:            0          0          0
root@64xxx-90f652f45cc4:/# echo m > /proc/sysrq-trigger 
[11151.278546] sysrq: SysRq : Show Memory
[11151.282392] Mem-Info:
[11151.294013] active_anon:587 inactive_anon:10 isolated_anon:0
[11151.294013]  active_file:747 inactive_file:169 isolated_file:0
[11151.294013]  unevictable:0 dirty:0 writeback:0 unstable:0
[11151.294013]  slab_reclaimable:236 slab_unreclaimable:1899
[11151.294013]  mapped:371 shmem:30 pagetables:79 bounce:0
[11151.294013]  free:1708 free_pcp:0 free_cma:0
[11151.326043] Normal free:6832kB min:1024kB low:1280kB high:1536kB active_anon:2348kB inactive_anon:40kB active_file:2988kB inactive_file:676kB unevictable:0kB isolated(anon):0kB isolated(file):0kB present:32768kB managed:27512kB mlocked:0kB dirty:0kB writeback:0kB mapped:1484kB shmem:120kB slab_reclaimable:944kB slab_unreclaimable:7596kB kernel_stack:400kB pagetables:316kB unstable:0kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB writeback_tmp:0kB pages_scanned:0 all_unreclaimable? no
[11151.369605] lowmem_reserve[]: 0 0
[11151.372960] Normal: 198*4kB (UME) 213*8kB (UME) 93*16kB (UME) 37*32kB (UME) 16*64kB (ME) 5*128kB (UM) 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 6832kB
[11151.386939] 946 total pagecache pages
[11151.390623] 0 pages in swap cache
[11151.393963] Swap cache stats: add 0, delete 0, find 0/0
[11151.399231] Free swap  = 0kB
[11151.402129] Total swap = 0kB
[11151.405024] 8192 pages RAM
[11151.407758] 0 pages HighMem/MovableOnly
[11151.411616] 1314 pages reserved
```

I certainly can push the 1043v1 into an OOM, but I think that is out of scope for this issue. We are likely going to remove `opkg` on 8/32 ar71xx-generic devices to limit the possiblities of people shooting themselves in their foot.

rotanid, 2018-05-06:
so far i also see improvement on the lowmem devices, even in v2017.1.x branch.
firmware builders should make sure to not include many additional packages which could overload these lowmem devices again.
we in our community handle lowmem devices differently as we don't include any feature-only package for lowmem devices - examples: USB, airtime, custom packages like time-based stuff, etc. - we even removed opkg from some devices, so people can't shoot themselves in their feet accidently by installing stuff.
some of these ideas are borrowed from @mweinelt / Freifunk Darmstadt.

it would be nice to get reports about the progress in the Gluon master branch by @MPW1412 who had serious issues with some nodes

H4ndl3, 2018-05-11:
@rotanid I flashed the node that we used to analyze the load bug (a TL-WR841N v8) with a master based image 4 days ago, so far it didn't reappear.

[65 more replies]

closed 2020-03-03 by mweinelt