freifunk-gluon/gluon #1248: Gluon v2017.1.3 can not be flashed on most recent TP-Link C1200

SmithChart, 2017-11-01:
I have build a gluon v2017.1.3 with the Freifunk Braunschweig site-config ( https://github.com/ffbs/site-ffbs ) an tried to flash it on TP-Link C1200 (Archer C5). 
Hardware is v1.0. 
Stock firmware is "1.0.1 Build 20160918 rel.37066 (EU)". (Stock firmware download link: http://www.tp-link.de/download/Archer-C1200_V1.html#Firmware )

Stock firmware "Website" rejects the image after uploading.
If I serve it via TFTP in recovery (as filename "ArcherC1200v1_eu_tp_recovery.bin") the bootloader downloads the whole binary. After rebooting the new firmware has not been flashed.

IRC log regarding this issue so far is:

 [17-11-01 21:05:04] <chrissi^> ich habe hier gerade einen archer AC1200 V1 auf dem tisch und würde den gern mit einem gluon flashen. die firmware verweigert sich aber unserem image. per tftp bekomme ich das gleiche ergebnis. habt ihr einen zeiger für mich, was ich falsch mache?
[17-11-01 21:05:04] <chrissi^> ist es akutell überhaupt möglich einen aktuellen archer c5 mit nicht tp-link firmware zu betreiben?
[17-11-01 21:09:26] <neoraider> chrissi^, auf welcher Gluon-Version basiert das Image?
[17-11-01 21:09:45] <chrissi^> neoraider: v2017.1.3
[17-11-01 21:11:06] <neoraider> Ist das vielleicht ein C5 v2?
[17-11-01 21:11:36] <chrissi^> neoraider: firmware und aufkleber behaupten es ist ein v1.
[17-11-01 21:12:19] <neoraider> Hmm, die TP-Link-Webseite unterscheidet zwischen v1 und v1.20
[17-11-01 21:12:54] <neoraider> Hmm nein, v1 taucht gar nicht richtig auf, es gibt nur Firmware für v1.20
[17-11-01 21:12:54] <chrissi^> firmware darauf ist "1.0.1 Build 20160918 rel.37066 (EU)"
[17-11-01 21:13:24] <Brother_Lal> (EU) - region lock issue?
[17-11-01 21:13:28] <neoraider> Jo, vermutlich
[17-11-01 21:13:36] <neoraider> Ich finde nur nichts dazu auf der Webseite
[17-11-01 21:14:01] <neoraider> Scheinbar hast du eine neue Firmware drauf, als es zum Download gibt
[17-11-01 21:14:07] <neoraider> Ansonsten wäre das leicht zu beheben
[17-11-01 21:14:39] <chrissi^> neoraider: japp. das war uns auch aufgefallen. die box hat ein freifunk-interessierter eben aus der stadt mitgebracht.
[17-11-01 21:17:03] <neoraider> chrissi^, mach mal ein Ticket. Die werden bestimmt die gleichen Region Codes wie auf dem C7 verwenden, dann bekommt man das leicht hin
[17-11-01 21:19:26] <chrissi^> neoraider: ist gut. ich mache dir ein ticket. bis nächste woche habe ich das gerät hier noch im zugriff. falls du in der zeit dazu kommst kann ich gern was bauen und testen

NeoRaider, 2017-11-01:
Had a short look at it, and C1200 and C5 are different models (even though they're both adversized as "AC1200"). It will take some more work to port LEDE to the new hardware.

kpanic23, 2017-11-02:
Actually, I have a C1200. It is based on a Broadcom SoC, therefore I guess porting Gluon to run on it could be quite cumbersome...

NeoRaider, 2017-11-03:
Okay, in that case we can close this, as Broadcom hardware is not suitable for running mesh networks...

closed 2017-11-03 by NeoRaider