freifunk-gluon/gluon #1249: config mode remote access page misses passphrase entry

edeso, 2017-11-03:
hi,

on gluon v2017.1-125-g207337b i am missing the ssh passphrase entry form on the remote access page in config mode.

probably because of this change to 'remote.lua'
  
"
 -local config = (site.config_mode or {}).remote_login or {}
 -if not config.show_password_form then
 +local config = site.config_mode.remote_login
 +if not config.show_password_form(false) then
  	-- password login is disabled in site.conf
"
in commit
https://github.com/freifunk-gluon/gluon/commit/951f62ac8550889527c4421176b9d967a5179aad#diff-17a449f5149b634605ff9c5ab4cd6cd0

..ede

mweinelt, 2017-11-04:
What you are reporting sounds like it's working as intended.

By default the password form is disabled to prevent novice users from entering possibly weak passwords and encourage everyone to use public key authentication.

The community may decide that they still want the form to be shown. Therefore they have to explicitly enable this in their `site.conf`: https://gluon.readthedocs.io/en/latest/package/gluon-web-admin.html

rotanid, 2017-11-04:
what @mweinelt wrote is correct and this will be part of the next major gluon release (note that this change is not part of the v2017.1.x branch)

closed 2017-11-04 by rotanid

edeso, 2017-11-05:
thx!

wasn't aware of the change as intended. the commit message might have had a hint to the documented switch. ..ede