freifunk-gluon/gluon #1253: soft-brick on sysupgrade with UAP AC Pro swith v2016.2.7 and later

rotanid, 2017-11-08:
there may be a bug or at least problem in some cases with updates of the Ubiquiti UAP AC Pro devices.
i have four of these devices running Gluon, all have been bought in the 2nd half of 2016.

two of them soft-bricked reproducibly if they are updated from Gluon v2016.2.7 to a later build (e.g. v2017.1.3) or also from v2017.1.3 to any later commit of v2017.1.x.
the updates of earlier version worked well.

two other devices don't show this problem.
the difference between those groups of devices is, that the ones showing the problem are mounted at the ceiling of a fashion boutique and get their power from HP PoE switches via rather long cables.
the working ones use PoE, too - but with cables <10m.

my guess is, that the changes to sysupgrade starting with v2016.2.7 someshow resulted in increased power usage that results in problems together with long cables?
sounds crazy, other ideas are very welcome!

Adorfer, 2017-11-08:
just for curiosity: What is the "result" of the "softbrick"? What is the state the unit is to be found after the failed upgrade?
defective image -> kernel panic when booting LEDE image, in other words: bootloader o.k, TFTPserver-rescue still possible?
More drastic? Or just "loss of config data, after powercycle without old parameters in setup-mode"?

rotanid, 2017-11-08:
thanks for asking, i added that information above. the image isn't defective, it can be flashed after reverting to stock via TFTP.

mweinelt, 2017-11-08:
Does the sysupgrade work on plain LEDE?

It would probably be helpful if you provided a serial log of the upgrade process and the boot after that.

rotanid, 2017-11-08:
you already wrote that on IRC and you know the answer. how should i pack a serial console at the ceiling of a store? and they wouldnt be happy if i crack their devices open anyways.
as i already said, the sysupgrade works on other devices of the same type with a much shorter cable. at lease i don't see any other difference between the similar devices than the cable length.

NeoRaider, 2017-11-26:
Maybe you could try to replicate the situation with a long cable on a device you can get a console log from?

rotanid, 2017-11-27:
thanks, this is already on my todo list. have to find time for that...

rotanid, 2018-01-09:
this issue is probably the same as #1301

closed 2018-01-09 by NeoRaider

rotanid, 2018-01-23:
i tested it today in depth and it indeed seems to be #1301