freifunk-gluon/gluon #1255: `make` re-builds things even when nothing changed

RalfJung, 2017-11-10:
## Steps to reproduce:
* `make GLUON_TARGET=x86-generic -j4`
* `make GLUON_TARGET=x86-generic V=s`

## Expected behavior

Hardly anything should happen on the second `make`, because the build is already up-to-date. I am running this on an SSD, and I'd expect a NOP build to take less than 10s, or maybe a minute.

## Actual behavior

This commands takes around 10 minutes to complete, and judging from the output it recompiles plenty of things.

RalfJung, 2017-11-10:
Here's the beginning of the 2nd build log:
```
rm -f /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/localversion
/home/r/src/freifunk/gluon/lede/scripts/kconfig.pl  + + /home/r/src/freifunk/gluon/lede/target/linux/generic/config-4.4 /home/r/src/freifunk/gluon/lede/target/linux/x86/config-4.4 /home/r/src/freifunk/gluon/lede/target/linux/x86/generic/config-default > /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /home/r/src/freifunk/gluon/lede/.config >> /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/.config.target
/home/r/src/freifunk/gluon/lede/scripts/package-metadata.pl kconfig /home/r/src/freifunk/gluon/lede/tmp/.packageinfo /home/r/src/freifunk/gluon/lede/.config 4.4 > /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/.config.override
/home/r/src/freifunk/gluon/lede/scripts/kconfig.pl 'm+' '+' /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/.config.target /dev/null /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/.config.override > /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/.config.set
mv /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/.config.set /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/.config.old
grep -v INITRAMFS /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/.config.old > /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/.config.set
rm -rf /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/modules
cmp -s /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/.config.set /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/.config.prev || { cp /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/.config.set /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/.config; cp /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/.config.set /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/.config.prev; }
export MAKEFLAGS= ; [ -d /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/user_headers ] || make -C /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89 HOSTCFLAGS="-O2 -I/home/r/src/freifunk/gluon/lede/staging_dir/host/include -I/home/r/src/freifunk/gluon/lede/staging_dir/host/usr/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="i486-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Mon Jul 10 16:16:23 2017" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/r/src/freifunk/gluon/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  CC="i486-openwrt-linux-musl-gcc" INSTALL_HDR_PATH=/home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/user_headers headers_install
grep '=[ym]' /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/.config.set | LC_ALL=C sort | mkhash md5 > /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/.vermagic
touch /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/.configured
rm -f /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/vmlinux /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89/System.map
make -C /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/linux-x86_generic/linux-4.4.89 HOSTCFLAGS="-O2 -I/home/r/src/freifunk/gluon/lede/staging_dir/host/include -I/home/r/src/freifunk/gluon/lede/staging_dir/host/usr/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="i486-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Mon Jul 10 [... cut ...]
Packaged contents of /home/r/src/freifunk/gluon/lede/build_dir/target-i386_pentium4_musl-1.1.16/toolchain/ipkg-i386_pentium4/librt into /home/r/src/freifunk/gluon/lede/bin/targets/x86/generic/packages/librt_1.1.16-1_i386_pentium4.ipk
rm -rf /home/r/src/freifunk/gluon/lede/staging_dir/target-i386_pentium4_musl-1.1.16/root-x86/tmp-librt
mkdir -p /home/r/src/freifunk/gluon/lede/staging_dir/target-i386_pentium4_musl-1.1.16/root-x86/stamp /home/r/src/freifunk/gluon/lede/staging_dir/target-i386_pentium4_musl-1.1.16/root-x86/tmp-librt
install -d -m0755 /home/r/src/freifunk/gluon/lede/staging_dir/target-i386_pentium4_musl-1.1.16/root-x86/tmp-librt/lib
```

NeoRaider, 2017-11-11:
Not much is actually rebuilt, but as usual for complex multi-level Make executions, it will still need to look at a large number of files to even *know* if anything has changed.

Anyways, we're building a whole Linux distribution here, a lot of parts of the build are simply running scripts unconditionally. Some details about what is done on every build:
* Most package builds are skipped; the kernel `make` is called, but doesn't do much if nothing has changed
* Target-specific packages are rebuilt (to ensure the package repository built by Gluon doesn't have any outdated package files, the packages are deleted at the beginning of the build)
* Images are rebuilt every time (this can take some time on targets with many models, but is fast on x86). There is not much to optimize here: even the smallest change to the actual code inside the images will require rebuilding them. So unless you really call `make` again deliberately without changing anything, this step is always necessary.

To speed up builds for quick testing, it is helpful to build only a single model. This has a significant effect even on targets that only have a single "model" like x86, as passing the DEVICES variable also skips building the opkg repository, and thus, deleting package files before building.  An incremental build using
```
make GLUON_TARGET=x86-generic DEVICES=x86-generic -j5
```
takes about 1:40 minutes on my notebook.

Apart from these tricks, this is a WONTFIX. There have been various optimizations to the LEDE build system over the last 2 years, significantly decreasing build times and greatly improving parallelizability, but there aren't many low-hanging fruit left. Further optimization would require rewriting much of the LEDE build process, possibily using a tool better suited for building a whole Linux distribution than Make... which is something OpenEmbedded did a while ago, but AFAICT nothing OpenWrt/LEDE would attempt any time soon.

closed 2017-11-11 by NeoRaider

RalfJung, 2017-11-11:
I see. Thanks for the detailed explanation; I will try `DEVICES`.