freifunk-gluon/gluon #1260: support for TP-Link TL-WR1043N v5

rotanid, 2017-11-12:
there's a new revision of the WR1043 router.

the device TL-WR1043N v5 is very similar to the TL-WR1043ND v4.
the differences are:
- no detachable antennas
- no USB port

there's already a LEDE Pull Request by @Brother-Lal which has undergone a first review, should get accepted soon.
https://github.com/lede-project/source/pull/1506

a backport shouldn't be too hard.

as this device is very common in Freifunk networks (the main use case for Gluon) and there's a lack of affordable devices that are supported by Gluon and available to buy at the moment, i added this issue to the 2017.2 milestone.

mweinelt, 2017-11-15:
https://git.lede-project.org/?p=lede/pepe2k/staging.git;a=commit;h=673793d753717dc49e5a6f9b1bba52658cae63fc

rubo77, 2017-12-12:
Hat das schon Mal jemand getestet?

https://github.com/blocktrron/gluon/commits/new-devices

rotanid, 2017-12-13:
@rubo77 partly, the C58/C59/C60 part as a PR is there: #1275
i tested the C59 and it seems to work fine.
i don't have a TL-WR1043N v5 at the moment so i can't test this one at the moment, maybe @Brother-Lal can

Brother-Lal, 2017-12-13:
Testrouter ist geflasht und läuft.

rotanid, 2017-12-13:
@blocktrron so it has been tested: https://map.tecff.de/#!v:m;n:50c7bfbc41a4
PR welcome :D

Brother-Lal, 2017-12-13:
Ok, i had some time to let it run and test some features.

Looks like everything works, LEDs, WAN, even the Wifi button.
Wifi mesh works with 11s.
Flashed through the factory webinterface without a hitch.
Thank you @blocktron for the backport!

mablae, 2017-12-14:
Moin Leute, 

Der Paketbote war eben da und hat wie erwartet 6 stk v5 gebraucht. 

Ich werde auch testen mit FFHB, der Compiler glüht.

blocktrron, 2017-12-14:
I just received my 1043v5, will test it today and create the pull request.

blocktrron, 2017-12-14:
I can confirm the 1043 v5 working well: https://meshviewer.darmstadt.freifunk.net/#/de/map/704f57455740

pflueg, 2017-12-14:
Hi all! As a side note - i just applied the changes of the PR to v2017.1.4. After a quick check in a small setup i didn't experience any problems (yet ;) ).

mablae, 2017-12-14:
@pflueg I tried the same but the resulting image bricks the device. Any hint? Did you cherry-pick or merge?

I am just back from the tftp session ;-)

pflueg, 2017-12-15:
@mablae : I guess you figured it out (since you deleted your question), but just to make sure here my steps:
1. I downloaded the raw patch file from the PR and added it to the "patches/lede" folder
2. I added the new target to "targets/ar71xx-generic"
3. Just to make sure (my gluon folder is pretty old) i did a "make dirclean" (beware - rebuilding the toolchain takes some time >2h), but mostly "make clean GLUON_TARGET=ar71xx-generic" will do it, too
4. "make update && make -j8 GLUON_TARGET=ar71xx-generic"

To be honest i flashed the new image directly with tftp. But I just took a brand new 1043n v5 out of the box this morning and flashed it with the factory image over the tp-link web interface. Same result - looks fine.

Cleaning the build folders helps alot - sometimes i get weird results as well, when i don't do this from time to time ;)
Hope you get it running!

mablae, 2017-12-15:
@pflueg I wonder how that should work since @blocktrron pointed out 
there are some more patches needed: 

```
<blocktrron> https://github.com/freifunk-gluon/gluon/blob/master/patches/lede/0041-firmware-utils-tplink-safeloader-support-strings-as-soft_version.patch
<blocktrron> https://github.com/freifunk-gluon/gluon/blob/master/patches/lede/0040-firmware-utils-tplink-safeloader-add-TP-Link-Archer-C25-v1.patch
```

I am currently building @blocktrron 's whole branch, without any modifications made. 
Still no luck. Maybe it's incompatible with the site config. I am lost-

mablae, 2017-12-15:
Now I got this: 
```
Configuration failed:
 * unable to enable package 'ip6tables'
 * unable to enable package 'hostapd-mini'
 * unable to enable package 'respondd-module-airtime'
 * unable to enable package 'iputils-ping6'
 * unable to enable package 'iwinfo'
 * unable to enable package 'iptables'
 * unable to enable package 'haveged'
Makefile:102: recipe for target 'config' failed
``` 

Any idea?

pflueg, 2017-12-15:
Just thinking - is it possible that you are accidentally using the "site.mk" from the "2017.1.x" branch for building in the "master/2017.2.x" branch or vice versa?
That would result in valid images, but will soft-brick the devices. The structure of the file changed from the ["2017.1.x" branch](https://github.com/freifunk-gluon/gluon/blob/v2017.1.x/docs/site-example/site.mk) to the ["master/2017.2.x" branch](https://github.com/freifunk-gluon/gluon/blob/master/docs/site-example/site.mk)

mablae, 2017-12-15:
@pflueg @blocktrron Finally got some working image: 
https://map.bremen.freifunk.net/#/de/map/704f57af1630

Thanks for the pointers. I need to look into the build chain more deeply an maybe should talk to some one who understand this stuff in person.

closed 2017-12-19 by rotanid

MP3Chris2712, 2018-01-11:
Thank you for work, it does not work MESH on LAN with normal LAN cable only with crossover cable. anyone an advice?

Best greats 
Christian

blocktrron, 2018-01-11:
@MP3Chris2712 with which other type of router did you try MoL? Two 1043v5 or another device?

MP3Chris2712, 2018-01-11:
the problem is with the hook old - mesh in the lan mesh option. are there different Batman versions in master branch ?

Byggvir, 2018-01-12:
Am Donnerstag, 11. Januar 2018, 17:22:55 CET schrieb MP3Chris2712:
> the problem is with the hook old - mesh in the lan mesh option. are there
> different Batman versions in master branch ?

The problem is the change in the Gluon master to vlan.

As a work-around you can add the old br-mesh_lan with

# batctl if add br-mesh_lan

See:  https://forum.freifunk.net/t/tp-link-tl-wr1043n-v5/15700/56

Best regards 

Thomas

-- 
Thomas Arend
Zingsheimstraße 31
53359 Rheinbach

NeoRaider, 2018-01-12:
Just set legacy to 1 for the mesh_lan section (there is even a checkbox in the web interface for that). Don't try to add interfaces behind the back of netifd.

Byggvir, 2018-01-12:
A real man doesn't need documentation! Never saw 

 network.mesh_wan.legacy='0'
 network.mesh_lan.legacy='1'

before.

NeoRaider, 2018-01-12:
@Byggvir I agree that our documentation for upcoming features could be better (e.g. in the form of preliminary release notes containing all changes that will be part of the next major release). For now, the best way to keep up-to-date with Gluon development (including such new configuration options) is the Gluon mailing list.

rotanid, 2018-01-12:
@Byggvir this option is only relevant for upgrades, therefore it will be mentioned in the release notes once there is a new major release. this issue was mentioned on the gluon mailing list on 2017-06-27 and reading the mailing list is somewhat mandatory for developers of firmwares based on the Gluon framework.

Byggvir, 2018-01-12:
Am Freitag, 12. Januar 2018, 22:44:29 CET schrieb Andreas Ziegler:
> there's already a backport pull request to v2017.1.x - this is waiting for
> test reviews so please use this one and you 1) havent got the mesh-lan
> issue and 2) you help us by commenting afterwards if everything works with
> the backport: #1295

How do I get the backport?

Best regards

Thomas 
-- 
Thomas Arend
Zingsheimstraße 31
53359 Rheinbach

rotanid, 2018-01-12:
link again:
https://github.com/freifunk-gluon/gluon/pull/1295
which links to
https://github.com/CodeFetch/gluon/tree/v2017.1.x

Byggvir, 2018-01-13:
Have build a packport image. Works, but ...

Router reports as model tp-link-tl-wr1043n-v5 in the manifest it is listed as  tp-link-tl-wr1043n-**nd-**v5.

Will test the autoupdater tomorrow.

rotanid, 2018-01-14:
thanks for testing, let's discuss this over in the pull request.

PR #1279 (2017-12-19): ar71xx: add support for TP-Link TL-WR1043N v5
This adds support for TL-WR1043N v5.

closes #1260