freifunk-gluon/gluon #1262: 'make update' fails if there is a file site/modules

edeso, 2017-11-16:
```
 make update
--- Updating module 'lede' ---
Reinitialized existing Git repository in /raider/foo/gluon/gluon2017.1.4test/lede/.git/
--- Updating module 'packages/openwrt' ---
Reinitialized existing Git repository in /raider/foo/gluon/gluon2017.1.4test/packages/openwrt/.git/
--- Updating module 'packages/gluon' ---
Reinitialized existing Git repository in /raider/foo/gluon/gluon2017.1.4test/packages/gluon/.git/
--- Updating module 'packages/routing' ---
Reinitialized existing Git repository in /raider/foo/gluon/gluon2017.1.4test/packages/routing/.git/
--- Updating module 'packages/luci' ---
Reinitialized existing Git repository in /raider/foo/gluon/gluon2017.1.4test/packages/luci/.git/
--- Patching module 'lede' ---
Applying: build: move STAGING_DIR_HOSTPKG and BUILD_DIR_HOST back to a common directory for all targets
Applying: procd: add support for alternative rc.d directories
...
Reset branch 'patched'
--- Patching module 'packages/openwrt' ---
Reset branch 'patched'
--- Patching module 'packages/gluon' ---
Reset branch 'patched'
--- Patching module 'packages/routing' ---
Applying: alfred: adjust intervals
Reset branch 'patched'
--- Patching module 'packages/luci' ---
Reset branch 'patched'
Makefile:34: recipe for target 'update' failed
make: *** [update] Error 1
```

doesn't matter if the file is empty or does contain only commented lines ..ede

NeoRaider, 2017-11-17:
What distribution are you using, and in particular, which version of bash is installed?

edeso, 2017-11-17:
On 17.11.2017 12:10, Matthias Schiffer wrote:
> What distribution are you using, and in particular, which version of bash is installed?

:$ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 8.9 (jessie)
Release:        8.9
Codename:       jessie

:$ bash --version
GNU bash, Version 4.3.30(1)-release (x86_64-pc-linux-gnu)

..ede

rubo77, 2017-11-18:
If you add at least one real Repo to the file, does it work then?

closed 2017-11-23 by NeoRaider

edeso, 2017-11-24:
@NeoRaider , thx works fine now.. ede

commit 87c741b4 (2017-11-23): scripts: run feeds.sh with bash like the other scripts handling modules

Fixes #1262
commit 36bbe88f (2017-11-23): scripts: run feeds.sh with bash like the other scripts handling modules

Fixes #1262