freifunk-gluon/gluon #1263: mesh_wan is not disabled properly

lrnzo, 2017-11-16:
Hello, i am not sure if this still a thing, but i could reproduce it on 2017.1.3 and i did not find any issue regarding this so far. So here it is:

normally one would think that the following lines in the `/etc/config/network` lead to a non-meshing br-wan:
> config interface 'mesh_wan'
	option ifname 'br-wan'
	option transitive '1'
	option fixed_mtu '1'
	option proto 'gluon_mesh'
	option auto '0'

but in fact `batctl if` gives me `br-wan: active` on many gluon-nodes.  it is being actived automatically during boot. `logread` says:
>Thu Nov 16 19:22:02 2017 daemon.notice netifd: Interface 'mesh_wan' is setting up now
Thu Nov 16 19:22:02 2017 kern.info kernel: [  107.934431] batman_adv: bat0: Adding interface: br-wan
Thu Nov 16 19:22:02 2017 kern.info kernel: [  107.934718] batman_adv: bat0: The MTU of interface br-wan is too small (1500) to handle the transport of batman-adv packets. Packets going over this interface will be fragmented on layer2 which could impact the performance. Setting the MTU to 1532 would solve the problem.
Thu Nov 16 19:22:02 2017 kern.info kernel: [  107.935802] batman_adv: bat0: Interface activated: br-wan

NeoRaider, 2017-11-25:
I can't reproduce this issue. Please provice the output of `uci show network.mesh_wan`.

Does your firmware include any non-standard scripts that might call `ifup mesh_wan`?

2tata, 2017-12-09:
Yes, the hoodselektor calls `ifup mesh_wan` `ifdown mesh_wan` `ifdown mesh_lan` `ifup mesh_lan`.

NeoRaider, 2017-12-09:
Well, it should not do this if the interfaces are disabled in `/etc/config/network`.

Closing as this is not a bug in Gluon.

closed 2017-12-09 by NeoRaider