freifunk-gluon/gluon #1265: Support DNS cache using "discovered" DNS servers

RalfJung, 2017-11-19:
Ideally, we'd like the node's DNS server (when used as a cache by clients) to use exactly those resolvers that it discovered via IPv6 RAs (our RAs contain DNS server information). This is to make sure that if one of our gateways is down, it doesn't even try to use that for DNS resolution.

Right now, what we do is set the `cacheentries` in the `dns` section, but we do *not* set any `server`. The consequence of that seems to be that it indeed provides a DNS cache, and just uses the servers it discovered. However, the [documentation](https://gluon.readthedocs.io/en/v2017.1.x/features/dns-cache.html) does not explain what happens when `cacheentries` is present but `server` is not, so I am worried that in some future update, the behavior may change and suddenly, the WAN resolver is used. Would it be possible to support and document configurations with `cacheentries`, but without `server`?

rotanid, 2017-11-19:
i don't really understand your issue / problem...
the clients always use the discovered DNS server - and this has nothing to to with the "server" option of this feature.
you could set a not-server-specific IP in this configuration which you only use for DNS and therefore you can use this IP on any one of your gateways - ideally it would be an ULA address like our fdef:ffc0:4ffa::11
we actually have two of those IPs in our configuration so it works automatically
https://github.com/tecff/site-ffa/blob/master/site.conf#L61

NeoRaider, 2017-11-19:
If the DNS cache is used, the gateways should not announce any RDNSS records of their own, so the clients use the nodes as DNS servers and the DNS cache is effective. This means that there are no DNS servers to discover, making DNS fail completely when no servers are set in site.conf.

There is no easy solution with the current network setup, and with the switch to Babel, RAs from gateways will become obsolete entirely (so DNS servers must always be configured statically in site.conf). While we could build a new mesh-wide service announcement system that might also include DNS servers, this is out of scope for the near future of Gluon development.

rotanid, 2017-11-19:
@NeoRaider that information should be part of the docs, even i didn't know that the "global" RA shouldn't contain RDNSS anymore after enabling dns-cache.
i assumed, the central DHCP and RA should announce the next-hop IPs as DNS

NeoRaider, 2017-11-19:
@rotanid You are right, the central DHCP should announce the next-node IP address. For RA, it doesn't matter if the gateways/supernodes announces the next-hop address or nothing at all, but it should not announce other addresses.

rotanid, 2017-11-19:
ok, but the documentation doesn't mention neither DHCP nor RA - so this issue is kind of valid.
i'll reopen, hopefully someone (including myself) finds time to update the docs.

RalfJung, 2017-11-19:
> If the DNS cache is used, the gateways should not announce any RDNSS records of their own, so the clients use the nodes as DNS servers and the DNS cache is effective. This means that there are no DNS servers to discover, making DNS fail completely when no servers are set in site.conf.

That's a good point. I did not think far enough. Thanks!

PR #1268 (2017-12-26): explain that not setting dns.servers makes no sense
Fixes #1265
closed 2017-12-26 by rotanid