freifunk-gluon/gluon #1273: Random reboots at TP-Link Archer C25 with v2017.1.4

norcon, 2017-12-05:
Hi all,

i build last release (v2017.1.4 today 05.12.2017) to use it on C25. The device randomly reboots.

Here some log entrys:

```
[    0.000000] Linux version 4.4.93 (jenkins@dev) (gcc version 5.4.0 (LEDE GCC 5.4.0 r3581+42-6b6578feec) ) #0 Thu Aug 3 01:34:19 2017
[    0.000000] MyLoader: sysp=0e979506, boardp=24b4b694, parts=a382a52d
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] SoC: Qualcomm Atheros QCA956X ver 1 rev 0
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 04000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] No valid device tree found, continuing without
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] On node 0 totalpages: 16384
[    0.000000] free_area_init_node: node 0, pgdat 804304d0, node_mem_map 81000000
[    0.000000]   Normal zone: 128 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 16384 pages, LIFO batch:3
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
[    0.000000] Kernel command line:  board=ARCHER-C25-V1 mtdparts=spi0.0:128k(factory-uboot)ro,64k(u-boot)ro,1536k(kernel),6272k(rootfs),128k(config)ro,64k(art)ro,7808k@0x30000(firmware) console=ttyS0,115200 rootfstype=squashfs,jffs2 noinitrd
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 59968K/65536K available (3089K kernel code, 167K rwdata, 776K rodata, 312K init, 205K bss, 5568K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:51
[    0.000000] Clocks: CPU:775.000MHz, DDR:650.000MHz, AHB:258.333MHz, Ref:25.000MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4932285024 ns
[    0.000006] sched_clock: 32 bits at 387MHz, resolution 2ns, wraps every 5541893118ns
[    0.008208] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
[    0.071017] pid_max: default: 32768 minimum: 301
[    0.075978] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.082950] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.092469] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.102887] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.110341] NET: Registered protocol family 16
[    0.116279] MIPS: machine is TP-LINK Archer C25 v1
[    0.557636] registering PCI controller with io_map_base unset
[    0.571508] PCI host bridge to bus 0000:00
[    0.575875] pci_bus 0000:00: root bus resource [mem 0x12000000-0x13ffffff]
[    0.583112] pci_bus 0000:00: root bus resource [io  0x0001]
[    0.588989] pci_bus 0000:00: root bus resource [??? 0x00000000 flags 0x0]
[    0.596134] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
[    0.604520] pci 0000:00:00.0: [168c:0050] type 00 class 0x028000
[    0.604548] pci 0000:00:00.0: invalid calibration data
[    0.609976] pci 0000:00:00.0: reg 0x10: [mem 0x00000000-0x001fffff 64bit]
[    0.610029] pci 0000:00:00.0: reg 0x30: [mem 0x00000000-0x0000ffff pref]
[    0.610085] pci 0000:00:00.0: supports D1
[    0.610098] pci 0000:00:00.0: PME# supported from D0 D1 D3hot
[    0.610286] pci_bus 0000:00: busn_res: [bus 00-ff] end is updated to 00
[    0.610317] pci 0000:00:00.0: BAR 0: assigned [mem 0x12000000-0x121fffff 64bit]
[    0.618065] pci 0000:00:00.0: BAR 6: assigned [mem 0x12200000-0x1220ffff pref]
[    0.625677] pci 0000:00:00.0: using irq 40 for pin 1
[    0.631640] clocksource: Switched to clocksource MIPS
[    0.638012] NET: Registered protocol family 2
[... 271 lines cut ...]
[  238.539073] ath10k: [0104]: F00F0000 6AD40300 0128FC17 80881071 08000100 D4264400 10009600 FF000000
[  238.539087] ath10k: [0112]: 6AD40300 0128FC17 80881071 00000200 D4264400 10009600 00000000 6AD40300
[  238.539101] ath10k: [0120]: 0128FC17 80881071 00000300 D4264400 10009600 00000000 6AD40300 0128FC17
[  238.539114] ath10k: [0128]: 80881071 00000400 D4264400 10009600 00000000 6AD40300 0128FC17 80881071
[  238.539126] ath10k: [0136]: 00000500 D4264400 10009600 00000000
[  238.539137] ath10k_pci 0000:00:00.0: ATH10K_END
[  256.540204] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[  256.540235] ath10k: [0000]: F11C0400 0128FC17 070B1071 01000401 03900000 00000000 FF000000 F11C0400
[  256.540251] ath10k: [0008]: 0128FC17 80881071 08000000 64284400 10009600 F00F0000 F11C0400 0128FC17
[  256.540265] ath10k: [0016]: 80881071 08000100 64284400 10009600 FF000000 F11C0400 0128FC17 80881071
[  256.540279] ath10k: [0024]: 00000200 64284400 10009600 00000000 F11C0400 0128FC17 80881071 00000300
[  256.540292] ath10k: [0032]: 64284400 10009600 00000000 F11C0400 0128FC17 80881071 00000400 64284400
[  256.540306] ath10k: [0040]: 10009600 00000000 F11C0400 0128FC17 80881071 00000500 64284400 10009600
[  256.540316] ath10k: [0048]: 00000000
[  256.540326] ath10k_pci 0000:00:00.0: ATH10K_END
Tue Dec  5 20:46:33 2017 kern.debug kernel: [  601.563085] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
Tue Dec  5 20:46:33 2017 kern.debug kernel: [  601.563112] ath10k: [0000]: 7E800900 564C0014 C47B4300 7C059C00 06000000 0C000000 003D0000 7E800900
Tue Dec  5 20:46:33 2017 kern.debug kernel: [  601.563128] ath10k: [0008]: 244C0014 C47B4300 06000000 00000000 20000000 01000000 CB800900 244C0014
Tue Dec  5 20:46:33 2017 kern.debug kernel: [  601.563140] ath10k: [0016]: C47B4300 00000000 30000000 20000000 01000000
Tue Dec  5 20:46:33 2017 kern.debug kernel: [  601.563152] ath10k_pci 0000:00:00.0: ATH10K_END
```

If you need more infos please ask.
Thanks for your work!
norcon

rotanid, 2017-12-07:
@norcon this device is marked as BROKEN and only built if you explicitly say so.
therefore, this behaviour is a known bug and it's most probably not fixable for anyone in the Gluon team.
there's a solution you probably don't like: disable the 5ghz mesh and client network, the device is stable as long as it doesn't use the 5ghz chip

norcon, 2017-12-07:
@rotanid is there any possibility that this bug will be fixed by the driver manufacturer? And if it is so, will this fix get into Gluon?
Thanks!

rotanid, 2017-12-07:
@norcon i wouldn't count on it, there's no direct communication between the manufacturer and the open source community. but if they update their stuff, we'll integrate it of course!

please forgive, i'm closing this issue for now, as there's nothing we can do about it at the moment and you can comment again anyway.

closed 2017-12-07 by rotanid