freifunk-gluon/gluon #1282: gluon-respondd: no updates after first contact

blocktrron, 2017-12-21:
At FFDA, we are experiencing our nodes on current Gluon master not sending respondd messages to our map backend. Nodes will send one message and won't answer on any other request. 

They appear to be offline on map while the status-page is still reachable and mesh links are also drawn.

Last known working commit currently is f7f659c2541011fb406243abbaaa16f0baedba74.

rotanid, 2017-12-22:
7351fb5d4a58c685dafeda9545145177ffe75969 should be the last working, as the only firewall related changes were after that with https://github.com/freifunk-gluon/gluon/pull/1124 by @christf

rotanid, 2017-12-24:
so @mweinelt reported he didn't have issues anymore after a reboot? does this work for every affected node?

christf, 2017-12-24:
let's investigate if this indeed is the firewall. Can you please check the console log whether you can see rejected packets that correlate with the requests being sent?

mweinelt, 2017-12-24:
@rotanid it works for a while after boot and fails to work after that. When I started tcpdump on the interface it starts working again, presumably due `br-client` entering promiscuous mode.

@christf not sure what you mean … but my device uses the plain (legacy) mesh on the wan port, and the batman-adv neighbourship looks fine.

christf, 2017-12-25:
@mweinelt could you please use tcpdump -p to not start promisc mode?

I wondered if you see any droped packets in the dmesg-ring-buffer. you could increase verbosity by setting log = 1 for the respective zone.

mweinelt, 2017-12-26:
@christf I think we can rule out MOW/MOL, because *all* of our 20 experimental routers are affected.

mweinelt, 2017-12-26:
We're gonna inform our users when the new firmware is ready with the following snippet to update during the console, which is still reachable as a local client or from WAN.

```
ip6tables -I zone_mesh_input 1 -p ipv6-icmp -m icmp6 --icmpv6-type 130/0 -j ACCEPT
autoupdater
```

NeoRaider, 2017-12-27:
Should be working now.

mweinelt, 2017-12-28:
Confirmed working.

closed 2017-12-28 by NeoRaider