freifunk-gluon/gluon #1283: Target brcm2708-bcm2709 doesn't compile

MPW1412, 2017-12-23:
Hello,

the brcm2708-bcm2709 targets for v2017.1.3 (in this case) fails. But as I haven't seen any notifications in the git commit history, I assume that it doesn't compile with master or v2017.1.4 either.

Surprisingly, the brcm27-08-bcm2708 compiles just fine.

This has a really low priority. I just wanted to mention it, because it seems to be broken for quite some time now.

```
build/genattr /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/common.md /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/config/arm/arm.md \
  insn-conditions.md > tmp-attr.h
build/genattr-common /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/common.md /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/config/arm/arm.md \
  insn-conditions.md > tmp-attr-common.h
build/gencodes /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/common.md /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/config/arm/arm.md \
  insn-conditions.md > tmp-codes.h
/bin/bash /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/../move-if-change tmp-attr-common.h insn-attr-common.h
/bin/bash /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/../move-if-change tmp-flags.h insn-flags.h
echo timestamp > s-attr-common
build/genconfig /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/common.md /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/config/arm/arm.md \
  insn-conditions.md > tmp-config.h
echo timestamp > s-flags
build/genopinit /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/common.md /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/config/arm/arm.md \
  insn-conditions.md -htmp-opinit.h -ctmp-opinit.c
/bin/bash /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/../move-if-change tmp-codes.h insn-codes.h
echo timestamp > s-codes
build/genattrtab /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/common.md /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/config/arm/arm.md insn-conditions.md \
        -Atmp-attrtab.c -Dtmp-dfatab.c -Ltmp-latencytab.c
/bin/bash /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/../move-if-change tmp-attr.h insn-attr.h
echo timestamp > s-attr
build/genautomata /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/common.md /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/config/arm/arm.md \
  insn-conditions.md > tmp-automata.c
/bin/bash /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/../move-if-change tmp-config.h insn-config.h
echo timestamp > s-config
build/genemit /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/common.md /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/config/arm/arm.md \
  insn-conditions.md > tmp-emit.c
/bin/bash /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/../move-if-change tmp-opinit.h insn-opinit.h
/bin/bash /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/../move-if-change tmp-opinit.c insn-opinit.c
echo timestamp > s-opinit
build/genextract /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/common.md /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/config/arm/arm.md \
  insn-conditions.md > tmp-extract.c
/bin/bash /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/../move-if-change tmp-extract.c insn-extract.c
echo timestamp > s-extract
build/genoutput /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/common.md /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/config/arm/arm.md \
  insn-conditions.md > tmp-output.c
/bin/bash /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/../move-if-change tmp-emit.c insn-emit.c
echo timestamp > s-emit
build/genpeep /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/common.md /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/config/arm/arm.md \
  insn-conditions.md > tmp-peep.c
/bin/bash /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/../move-if-change tmp-peep.c insn-peep.c
echo timestamp > s-peep
build/genrecog /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/common.md /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/config/arm/arm.md \
  insn-conditions.md > tmp-recog.c
/bin/bash /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/../move-if-change tmp-recog.c insn-recog.c
echo timestamp > s-recog
x86_64-linux-gnu-g++ -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -DGENERATOR_FILE -I. -Ibuild -I/root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc -I/root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/build -I/root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/../include  -I/root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/../libcpp/include  \
        -o build/gcov-iov.o /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/gcov-iov.c
/bin/bash /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/../move-if-change tmp-output.c insn-output.c
echo timestamp > s-output
x86_64-linux-gnu-g++   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -DGENERATOR_FILE  \
        build/gcov-iov.o -o build/gcov-iov
build/gcov-iov '5.4.0' '' \
    > tmp-gcov-iov.h
/bin/bash /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/../move-if-change tmp-gcov-iov.h gcov-iov.h
echo timestamp > s-iov
x86_64-linux-gnu-g++ -c  -DIN_GCC_FRONTEND -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -Ic -I/root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc -I/root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/c -I/root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gc [... cut ...]
toolchain/Makefile:89: recipe for target '/root/gits/gluon/lede/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/stamp/.toolchain_install' failed
make[2]: *** [/root/gits/gluon/lede/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/stamp/.toolchain_install] Error 2
make[2]: Leaving directory '/root/gits/gluon/lede'
Build failed - please re-run with -j1 to see the real error message
/root/gits/gluon/lede/include/toplevel.mk:198: recipe for target 'world' failed
make[1]: *** [world] Error 1
make[1]: Leaving directory '/root/gits/gluon/lede'
```

Merry Christmas!
Matthias

NeoRaider, 2017-12-23:
Is this reproducible?

The line `/bin/bash: line 1: 32230 Killed                  build/genautomata /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/common.md /root/gits/gluon/lede/build_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-5.4.0_musl-1.1.16_eabi/gcc-5.4.0/gcc/config/arm/arm.md insn-conditions.md > tmp-automata.c` looks suspicious. Is it possible that the process was killed by something else (something unrelated to the build)?

rotanid, 2017-12-23:
i don't think there is a general problem, our build went fine a few weeks ago

MPW1412, 2017-12-23:
Oh yes, you're right. It seems that brcm2708-bcm2709 needs more RAM to build than all the other targets. At the time index, the build process crashed, I found those entries in the syslog. Seems like the VM ran out of RAM.

I just didn't think of it, because the previous targets compiled perfectly.

```
Dec 23 03:58:44 firmware kernel: [7325708.376009] genautomata invoked oom-killer: gfp_mask=0x10200da, order=0, oom_score_adj=0
Dec 23 03:58:44 firmware kernel: [7325708.609747] genautomata cpuset=/ mems_allowed=0
Dec 23 03:58:44 firmware kernel: [7325708.653636] CPU: 0 PID: 32230 Comm: genautomata Not tainted 3.16.0-4-amd64 #1 Debian 3.16.43-2+deb8u5
Dec 23 03:58:44 firmware kernel: [7325708.653639] Hardware name: QEMU Standard PC (i440FX + PIIX, 1996), BIOS 1.7.5-20140531_083030-gandalf 04/01/2014
Dec 23 03:58:44 firmware kernel: [7325708.653640]  0000000000000000 ffffffff81514291 ffff88007c97a210 0000000000000000
Dec 23 03:58:44 firmware kernel: [7325708.653643]  ffffffff81511e69 0000000000000096 0000000000000001 0000000000000000
Dec 23 03:58:44 firmware kernel: [7325708.653644]  ffffffff8151c4d8 ffff88007bddb8b0 ffff88003705caa0 0000000000000000
Dec 23 03:58:44 firmware kernel: [7325708.653646] Call Trace:
Dec 23 03:58:44 firmware kernel: [7325708.656191]  [<ffffffff81514291>] ? dump_stack+0x5d/0x78
Dec 23 03:58:44 firmware kernel: [7325708.656191]  [<ffffffff81511e69>] ? dump_header+0x76/0x1e8
Dec 23 03:58:44 firmware kernel: [7325708.656191]  [<ffffffff8151c4d8>] ? async_page_fault+0x28/0x30
Dec 23 03:58:44 firmware kernel: [7325708.656191]  [<ffffffff81142ddd>] ? oom_kill_process+0x21d/0x370
Dec 23 03:58:45 firmware kernel: [7325708.656191]  [<ffffffff8114299d>] ? find_lock_task_mm+0x3d/0x90
Dec 23 03:58:45 firmware kernel: [7325708.656191]  [<ffffffff81143543>] ? out_of_memory+0x473/0x4b0
Dec 23 03:58:45 firmware kernel: [7325708.656191]  [<ffffffff8114940f>] ? __alloc_pages_nodemask+0x9ef/0xb50
Dec 23 03:58:45 firmware kernel: [7325708.656191]  [<ffffffff8118894d>] ? alloc_pages_current+0x9d/0x150
Dec 23 03:58:45 firmware kernel: [7325708.656191]  [<ffffffff81140183>] ? pagecache_get_page_fixed+0x53/0x1b0
Dec 23 03:58:45 firmware kernel: [7325708.656191]  [<ffffffff81140315>] ? grab_cache_page_write_begin+0x25/0x40
Dec 23 03:58:45 firmware kernel: [7325708.656191]  [<ffffffffa020b6a9>] ? ext4_da_write_begin+0x99/0x310 [ext4]
Dec 23 03:58:45 firmware kernel: [7325708.656191]  [<ffffffffa0209a9a>] ? ext4_mark_inode_dirty+0x5a/0x200 [ext4]
Dec 23 03:58:45 firmware kernel: [7325708.656191]  [<ffffffff8113f526>] ? generic_perform_write+0xd6/0x1c0
Dec 23 03:58:45 firmware kernel: [7325708.656191]  [<ffffffff81141598>] ? __generic_file_write_iter+0x158/0x340
Dec 23 03:58:45 firmware kernel: [7325708.656191]  [<ffffffffa0200b40>] ? ext4_file_write_iter+0xf0/0x3a0 [ext4]
Dec 23 03:58:45 firmware kernel: [7325708.656191]  [<ffffffff811aa7f4>] ? new_sync_write+0x74/0xa0
Dec 23 03:58:45 firmware kernel: [7325708.656191]  [<ffffffff811aaf32>] ? vfs_write+0xb2/0x1f0
Dec 23 03:58:45 firmware kernel: [7325708.656191]  [<ffffffff811aba72>] ? SyS_write+0x42/0xa0
Dec 23 03:58:45 firmware kernel: [7325708.656191]  [<ffffffff8151a48d>] ? system_call_fast_compare_end+0x10/0x15
Dec 23 03:58:45 firmware kernel: [7325708.839858] Mem-Info:
Dec 23 03:58:45 firmware kernel: [7325708.839863] Node 0 DMA per-cpu:
Dec 23 03:58:45 firmware kernel: [7325708.839866] CPU    0: hi:    0, btch:   1 usd:   0
Dec 23 03:58:45 firmware kernel: [7325708.839867] CPU    1: hi:    0, btch:   1 usd:   0
Dec 23 03:58:45 firmware kernel: [7325708.839868] CPU    2: hi:    0, btch:   1 usd:   0
Dec 23 03:58:45 firmware kernel: [7325708.839869] CPU    3: hi:    0, btch:   1 usd:   0
Dec 23 03:58:45 firmware kernel: [7325708.839870] Node 0 DMA32 per-cpu:
Dec 23 03:58:45 firmware kernel: [7325708.839871] CPU    0: hi:  186, btch:  31 usd:   0
Dec 23 03:58:45 firmware kernel: [7325708.839872] CPU    1: hi:  186, btch:  31 usd:   0
Dec 23 03:58:45 firmware kernel: [7325708.839873] CPU    2: hi:  186, btch:  31 usd: 156
Dec 23 03:58:45 firmware kernel: [7325708.839873] CPU    3: hi:  186, btch:  31 usd: 173
Dec 23 03:58:45 firmware kernel: [7325708.839876] active_anon:456995 inactive_anon:11754 isolated_anon:0
Dec 23 03:58:45 firmware kernel: [7325708.839876]  active_file:324 inactive_file:333 isolated_file:0
Dec 23 03:58:45 firmware kernel: [7325708.839876]  unevictable:0 dirty:0 writeback:0 unstable:0
Dec 23 03:58:45 firmware kernel: [7325708.839876]  free:13233 slab_reclaimable:7238 slab_unreclaimable:19250
Dec 23 03:58:45 firmware kernel: [7325708.839876]  mapped:2039 shmem:11807 pagetables:1867 bounce:0
Dec 23 03:58:45 firmware kernel: [7325708.839876]  free_cma:0
Dec 23 03:58:45 firmware kernel: [7325708.839878] Node 0 DMA free:8308kB min:348kB low:432kB high:520kB active_anon:6104kB inactive_anon:56kB active_file:4kB inactive_file:24kB unevictable:0kB isolated(anon):0kB isolated(file):0kB present:15992kB managed:15908kB mlocked:0kB dirty:0kB writeback:0kB mapped:36kB shmem:56kB slab_reclaimable:124kB slab_unreclaimable:968kB kernel_stack:208kB pagetables:16kB unstable:0kB bounce:0kB free_cma:0kB writeback_tmp:0kB pages_scanned:2480 all_unreclaimable? yes
Dec 23 03:58:45 firmware kernel: [7325708.839882] lowmem_reserve[]: 0 1992 1992 1992
Dec 23 03:58:45 firmware kernel: [7325708.839884] Node 0 DMA32 free:44624kB min:44704kB low:55880kB high:67056kB active_anon:1821876kB inactive_anon:46960kB active_file:1292kB inactive_file:1308kB unevictab [... cut ...]
Dec 23 03:58:45 firmware kernel: [7325708.845077] [32342]     0 32342     2049       27       9        0             0 x86_64-linux-gn
Dec 23 03:58:45 firmware kernel: [7325708.845078] [32343]     0 32343    12753     1982      27        0             0 cc1plus
Dec 23 03:58:45 firmware kernel: [7325708.845080] Out of memory: Kill process 32230 (genautomata) score 434 or sacrifice child
Dec 23 03:58:45 firmware kernel: [7325708.914957] Killed process 32230 (genautomata) total-vm:933740kB, anon-rss:919128kB, file-rss:20kB
```

closed 2017-12-23 by MPW1412