freifunk-gluon/gluon #1284: route lost on hostapd disassociate

christf, 2017-12-25:
when a client is disconnected (see 13:31:18 hostapd message) and re-connects afterwards (see 13:31:27 hostapd message) the client is losing its default route.

> Mon Dec 25 13:31:15 2017 kern.warn kernel: [45020.393820] REJECT(src wan)IN=br-wan OUT= MAC=ff:ff:ff:ff:ff:ff:de:42:86:62:89:92:08:00 SRC=192.168.13.117 DST=192.168.13.255 LEN=97 TOS=0x00 PREC=0x00 TTL=64 ID=28746 DF PROTO=UDP SPT=39311 DPT=21027 LEN=77 
> Mon Dec 25 13:31:15 2017 daemon.info fastd[1863]: sending handshake to <mesh_vpn_backbone_peer_babelgw2>[185.206.209.154:10001]...
> Mon Dec 25 13:31:15 2017 daemon.info fastd[1863]: resolving host `gw02.babel.ffffm.net' for peer <mesh_vpn_backbone_peer_babelgw2>...
> Mon Dec 25 13:31:15 2017 daemon.info fastd[1863]: resolved host `gw02.babel.ffffm.net' successfully
> Mon Dec 25 13:31:18 2017 daemon.info fastd[1863]: resolving host `gw04.babel.ffffm.net' for peer <mesh_vpn_backbone_peer_babelgw4>...
> Mon Dec 25 13:31:18 2017 daemon.info fastd[1863]: resolved host `gw04.babel.ffffm.net' successfully
> Mon Dec 25 13:31:18 2017 daemon.notice hostapd: client0: AP-STA-DISCONNECTED 60:57:18:88:d7:f9
> Mon Dec 25 13:31:21 2017 daemon.info fastd[1863]: sending handshake to <mesh_vpn_backbone_peer_babelgw3>[[2a06:8187:fb14:1::3:1]:10001]...
> Mon Dec 25 13:31:22 2017 daemon.notice netifd: wan (994): udhcpc: sending renew
> Mon Dec 25 13:31:22 2017 daemon.notice netifd: wan (994): udhcpc: lease of 192.168.13.161 obtained, lease time 3600
> Mon Dec 25 13:31:27 2017 daemon.info hostapd: client0: STA 60:57:18:88:d7:f9 IEEE 802.11: authenticated
> Mon Dec 25 13:31:27 2017 daemon.info hostapd: client0: STA 60:57:18:88:d7:f9 IEEE 802.11: associated (aid 2)
> Mon Dec 25 13:31:27 2017 daemon.notice hostapd: client0: AP-STA-CONNECTED 60:57:18:88:d7:f9
> Mon Dec 25 13:31:27 2017 daemon.info hostapd: client0: STA 60:57:18:88:d7:f9 IEEE 802.11: associated (aid 2)
> Mon Dec 25 13:31:27 2017 daemon.info hostapd: client0: STA 60:57:18:88:d7:f9 IEEE 802.11: associated (aid 2)
> Mon Dec 25 13:31:36 2017 daemon.info fastd[1863]: sending handshake to <mesh_vpn_backbone_peer_babelgw2>[[2a06:8187:fb14:1::2:1]:10001]...
> Mon Dec 25 13:31:37 2017 kern.warn kernel: [45042.333568] REJECT(src wan)IN=br-wan OUT= MAC=ff:ff:ff:ff:ff:ff:d0:50:99:47:70:66:08:00 SRC=192.168.13.9 DST=192.168.13.255 LEN=76 TOS=0x18 PREC=0xA0 TTL=64 ID=52536 DF PROTO=UDP SPT=123 DPT=123 LEN=56 
> Mon Dec 25 13:31:38 2017 authpriv.info dropbear[31548]: Child connection from 2a06:8187:fbab:2:2457:abd:3f3f:ec28:45368
> Mon Dec 25 13:31:39 2017 daemon.info fastd[1863]: sending handshake to <mesh_vpn_backbone_peer_babelgw4>[[2a06:8187:fb14:1::4:1]:10001]...
> Mon Dec 25 13:31:42 2017 authpriv.info dropbear[30211]: Exit (root): Error writing: Connection reset by peer

christf, 2017-12-25:
... I just saw a re-connect and my client obtained a route shortly therafter just fine. That should be fun to debug....

NeoRaider, 2017-12-27:
Is the *client* losing its default route? In this case, it is expected behaviour and has nothing to do with Gluon.

christf, 2017-12-28:
the client does not have default route after this happened.

closed 2018-01-22 by christf