freifunk-gluon/gluon #1299: respondd :unable to read providers

sap-dev, 2018-01-08:
Hello,

we have many problems with Respondd at many hardware:

TP-Link WR1043
TP-Link WR841

in log we see this:
Mon Jan  8 12:00:59 2018 daemon.info td-client: Performing broker selection...
Mon Jan  8 12:00:59 2018 daemon.notice netifd: Interface 'mesh_vpn' is disabled
Mon Jan  8 12:01:00 2018 daemon.info respondd[20128]: unable to read providers from '/usr/lib/respondd', ignoring

After a reboot, respondd work.
We use batman and batctl 2017.3 at our network

NeoRaider, 2018-01-08:
There are no unusual messages in the log snippet you posted (except for the tunneldigger connection being disabled - which will obviously also break respondd if there is no other connection between the node and the respondd querier).

Which Gluon version are you using? (content of `/lib/gluon/gluon-version`)

sap-dev, 2018-01-08:
Hi,

we use v2017.1.4-4-g2ae74fe7+
After a reboot the Router send respondd data

NeoRaider, 2018-01-08:
As I mentioned, this does not look like a respondd problem.

Do other features of the node, e.g. internet access, work correctly when respondd does not?

rotanid, 2018-01-08:
@sap-dev are you sure respondd doesn't work or do you just assume that based on the log message?
also, just to be safe, your build is based in the v2017.1.x branch with no custom changes?

lephisto, 2018-01-19:
I have noticed this behaviour while testing the current master too, but stat tracking worked.

NeoRaider, 2018-01-19:
@lephisto What behaviour?

lephisto, 2018-01-19:
That respondd write "unable to read providers from '/usr/lib/respondd', ignoring" to the log..

NeoRaider, 2018-01-19:
As mentioned, this is a completely normal message. It just informs you that you don't have any providers installed in `/usr/lib/respondd`, which is always the case (unless you have the airtime stat package).

rotanid, 2018-01-19:
closing this issue now as there is no report about a real problem so far.
feel free to open a new issue with exact information on your problem.

closed 2018-01-19 by rotanid