freifunk-gluon/gluon #1301: ar71xx:  UBNT UniFi AP-AC Mesh broken WiFi on current Gluon Master

blocktrron, 2018-01-08:
Flashing the current Gluon master on a UBNT UniFi AP AC Mesh results in a broken WiFi interface.

`iw dev` results in `nl80211 not found.`

## logread
```
BusyBox v1.25.1 () built-in shell (ash)

     _________
    /        /\      _    ___ ___  ___
   /  LE    /  \    | |  | __|   \| __|
  /    DE  /    \   | |__| _|| |) | _|
 /________/  LE  \  |____|___|___/|___|                      lede-project.org
 \        \   DE /
  \    LE  \    /  -----------------------------------------------------------
   \  DE    \  /    Reboot (17.01-SNAPSHOT, r3597+52-7f3dab2fc3)
    \________\/    -----------------------------------------------------------

root@64xxx-f09fc2dcc4c5:~# logread
Tue Jan  2 01:51:54 2018 kern.notice kernel: [    0.000000] Linux version 4.4.93 (hexa@build.ffda.io) (gcc version 5.4.0 (LEDE GCC 5.4.0 r3581+44-6b6578feec) ) #0 Wed Nov 22 00:24:53 2017
Tue Jan  2 01:51:54 2018 kern.debug kernel: [    0.000000] MyLoader: sysp=1bb19371, boardp=75533157, parts=5317f955
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.000000] bootconsole [early0] enabled
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.000000] SoC: Qualcomm Atheros QCA956X ver 1 rev 0
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.000000] Determined physical RAM map:
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.000000]  memory: 08000000 @ 00000000 (usable)
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Tue Jan  2 01:51:54 2018 kern.warn kernel: [    0.000000] No valid device tree found, continuing without
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.000000] Zone ranges:
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.000000] Movable zone start for each node
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.000000] Early memory node ranges
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
Tue Jan  2 01:51:54 2018 kern.debug kernel: [    0.000000] On node 0 totalpages: 32768
Tue Jan  2 01:51:54 2018 kern.debug kernel: [    0.000000] free_area_init_node: node 0, pgdat 804344d0, node_mem_map 81000000
Tue Jan  2 01:51:54 2018 kern.debug kernel: [    0.000000]   Normal zone: 256 pages used for memmap
Tue Jan  2 01:51:54 2018 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Tue Jan  2 01:51:54 2018 kern.debug kernel: [    0.000000]   Normal zone: 32768 pages, LIFO batch:7
Tue Jan  2 01:51:54 2018 kern.warn kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Tue Jan  2 01:51:54 2018 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
Tue Jan  2 01:51:54 2018 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Tue Jan  2 01:51:54 2018 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 32512
Tue Jan  2 01:51:54 2018 kern.notice kernel: [    0.000000] Kernel command line:  board=UBNT-UF-AC-LITE mtdparts=spi0.0:384k(u-boot)ro,64k(u-boot-env)ro,7744k(firmware),7744k(ubnt-airos)ro,128k(bs)ro,256k(cfg)ro,64k(EEPROM)ro console=ttyS0,115200 rootfstype=squashfs,jffs2 noinitrd
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.000000] Writing ErrCtl register=00000000
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.000000] Readback ErrCtl register=00000000
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.000000] Memory: 124944K/131072K available (3103K kernel code, 165K rwdata, 780K rodata, 296K init, 205K bss, 6128K reserved, 0K cma-reserved)
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.000000] NR_IRQS:51
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.000000] Clocks: CPU:775.000MHz, DDR:650.000MHz, AHB:258.333MHz, Ref:25.000MHz
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4932285024 ns
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.000007] sched_clock: 32 bits at 387MHz, resolution 2ns, wraps every 5541893118ns
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.008206] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.071016] pid_max: default: 32768 minimum: 301
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.076382] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.083367] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.096359] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.106781] futex hash table entries: 256 (order: -1, 3072 bytes)
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.115299] NET: Registered protocol family 16
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.123484] MIPS: machine is Ubiquiti UniFi-AC-LITE
Tue Jan  2 01:51:54 2018 kern.warn kernel: [    0.345124] registering PCI controller with io_map_base unset
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.375381] PCI host bridge to bus 0000:00
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.379708] pci_bus 0000:00: root bus resource [mem 0x12000000-0x13ffffff]
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.386996] pci_bus 0000:00: root bus resource [io  0x0001]
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.392898] pci_bus 0000:00: root bus resource [??? 0x00000000 flags 0x0]
Tue Jan  2 01:51:54 2018 kern.info kernel: [    0.400061] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
[... 240 lines cut ...]
[    6.697351] random: ubusd: uninitialized urandom read (4 bytes read, 10 bits of entropy available)
[    6.707162] random: ubusd: uninitialized urandom read (4 bytes read, 10 bits of entropy available)
[    6.716462] random: ubusd: uninitialized urandom read (4 bytes read, 10 bits of entropy available)
[    6.726052] random: ubusd: uninitialized urandom read (4 bytes read, 10 bits of entropy available)
[    6.735374] random: ubusd: uninitialized urandom read (4 bytes read, 10 bits of entropy available)
[    6.744864] random: ubusd: uninitialized urandom read (4 bytes read, 10 bits of entropy available)
[    6.754372] random: ubusd: uninitialized urandom read (4 bytes read, 10 bits of entropy available)
[    6.764153] procd: - init -
[    8.348804] random: jshn: uninitialized urandom read (4 bytes read, 13 bits of entropy available)
[    8.418030] random: jshn: uninitialized urandom read (4 bytes read, 13 bits of entropy available)
[   15.483559] jffs2_scan_eraseblock(): End of filesystem marker found at 0x0
[   15.527187] jffs2_build_filesystem(): unlocking the mtd device... done.
[   15.534028] jffs2_build_filesystem(): erasing all blocks after the end marker... 
[   19.586719] device eth0 entered promiscuous mode
[   19.622504] IPv6: ADDRCONF(NETDEV_UP): br-setup: link is not ready
[   19.669421] eth0: link up (1000Mbps/Full duplex)
[   19.674250] br-setup: port 1(eth0) entered forwarding state
[   19.680083] br-setup: port 1(eth0) entered forwarding state
[   19.763107] IPv6: ADDRCONF(NETDEV_CHANGE): br-setup: link becomes ready
[   21.677098] br-setup: port 1(eth0) entered forwarding state
[   34.141455] done.
[   34.143479] jffs2: notice: (836) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
[   77.477138] random: nonblocking pool is initialized

```

NeoRaider, 2018-01-08:
What was GLUON_ATH10K_MESH defined to during the build?

mweinelt, 2018-01-08:
Yes. We are seeing old kernels in dmesg after sysupgrading to newer ones. Seems like the sysupgrades are incomplete.

codedust, 2018-01-08:
Corresponding dmesg (AC Mesh sysupgraded from openwrt to gluon):

```
root@64xxx-788a20266914:~# dmesg 
[    0.000000] Linux version 4.4.14 (buildbot@owrtbuild01) (gcc version 5.3.0 (OpenWrt GCC 5.3.0 12009) ) #1 Wed Dec 6 23:56:36 UTC 2017
[    0.000000] MyLoader: sysp=a3f3ab2f, boardp=a19ab3fb, parts=0b7ffbaa
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] SoC: Qualcomm Atheros QCA956X ver 1 rev 0
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 08000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] No valid device tree found, continuing without
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000] free_area_init_node: node 0, pgdat 8043c5d0, node_mem_map 81000000
[    0.000000]   Normal zone: 256 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 32512
[    0.000000] Kernel command line:  board=UBNT-UF-AC-LITE mtdparts=spi0.0:384k(u-boot)ro,64k(u-boot-env)ro,7744k(firmware),7744k(ubnt-airos)ro,128k(bs)ro,256k(cfg)ro,64k(EEPROM)ro console=ttyS0,115200 rootfstype=squashfs,jffs2 noinitrd
[    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 124880K/131072K available (3134K kernel code, 174K rwdata, 784K rodata, 316K init, 206K bss, 6192K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:51
[    0.000000] Clocks: CPU:775.000MHz, DDR:650.000MHz, AHB:258.333MHz, Ref:25.000MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4932285024 ns
[    0.000007] sched_clock: 32 bits at 387MHz, resolution 2ns, wraps every 5541893118ns
[    0.008206] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
[    0.071016] pid_max: default: 32768 minimum: 301
[    0.075977] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.082957] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.092517] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.103910] NET: Registered protocol family 16
[    0.109806] MIPS: machine is Ubiquiti UniFi-AC-LITE
[    0.339615] registering PCI controller with io_map_base unset
[    0.353870] PCI host bridge to bus 0000:00
[    0.358237] pci_bus 0000:00: root bus resource [mem 0x12000000-0x13ffffff]
[    0.365479] pci_bus 0000:00: root bus resource [io  0x0001]
[    0.371352] pci_bus 0000:00: root bus resource [??? 0x00000000 flags 0x0]
[    0.378499] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
[    0.386890] pci 0000:00:00.0: [168c:003c] type 00 class 0x028000
[    0.386917] pci 0000:00:00.0: invalid calibration data
[    0.392349] pci 0000:00:00.0: reg 0x10: [mem 0x00000000-0x001fffff 64bit]
[    0.392401] pci 0000:00:00.0: reg 0x30: [mem 0x00000000-0x0000ffff pref]
[    0.392457] pci 0000:00:00.0: supports D1
[    0.392469] pci 0000:00:00.0: PME# supported from D0 D1 D3hot
[    0.392644] pci_bus 0000:00: busn_res: [bus 00-ff] end is updated to 00
[    0.392673] pci 0000:00:00.0: BAR 0: assigned [mem 0x12000000-0x121fffff 64bit]
[    0.400418] pci 0000:00:00.0: BAR 6: assigned [mem 0x12200000-0x1220ffff pref]
[    0.408035] pci 0000:00:00.0: using irq 40 for pin 1
[    0.413993] clocksource: Switched to clocksource MIPS
[    0.420374] NET: Registered protocol family 2
[    0.425738] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
[    0.433094] TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
[    0.439839] TCP: Hash tables configured (established 1024 bind 1024)
[    0.446624] UDP hash table entries: 256 (order: 0, 4096 bytes)
[... 43 lines cut ...]
[    3.290698] random: procd urandom read with 5 bits of entropy available
[    5.452426] eth0: link up (100Mbps/Full duplex)
[    5.457167] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[    5.528943] mount_root: jffs2 not ready yet, using temporary tmpfs overlay
[    5.539159] urandom-seed: Seed file not found (/etc/urandom.seed)
[    5.695521] eth0: link down
[    5.707686] procd: - early -
[    5.710740] procd: - watchdog -
[    6.281985] procd: - watchdog -
[    6.285509] procd: - ubus -
[    6.356139] procd: - init -
[   14.312560] jffs2_scan_eraseblock(): End of filesystem marker found at 0x0
[   14.337693] jffs2_build_filesystem(): unlocking the mtd device... done.
[... cut ...]
[   20.561028] br-setup: port 1(eth0) entered forwarding state
[   20.566841] br-setup: port 1(eth0) entered forwarding state
[   20.596204] IPv6: ADDRCONF(NETDEV_CHANGE): br-setup: link becomes ready
[   22.564029] br-setup: port 1(eth0) entered forwarding state
[   33.634085] done.
[   33.636107] jffs2: notice: (801) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
[   70.957387] random: nonblocking pool is initialized
```

mweinelt, 2018-01-08:
```
m25p80 spi0.0: mx25l12805d (16384 Kbytes)
7 cmdlinepart partitions found on MTD device spi0.0
Creating 7 MTD partitions on "spi0.0":
0x000000000000-0x000000060000 : "u-boot"
0x000000060000-0x000000070000 : "u-boot-env"
0x000000070000-0x000000800000 : "firmware"
2 uimage-fw partitions found on MTD device firmware
0x000000070000-0x0000001c0000 : "kernel"
0x0000001c0000-0x000000800000 : "rootfs"
mtd: device 4 (rootfs) set to be root filesystem
1 squashfs-split partitions found on MTD device rootfs
0x000000480000-0x000000800000 : "rootfs_data"
0x000000800000-0x000000f90000 : "ubnt-airos"
0x000000f90000-0x000000fb0000 : "bs"
0x000000fb0000-0x000000ff0000 : "cfg"
0x000000ff0000-0x000001000000 : "EEPROM"
```

Just an educated guess:
As we are writing our image to two partitions with mtd in the beginning, shouldn't we also do that on sysupgrade? The assumption is, that the image should be written to both `firmware` and `ubnt-airos`, but is only written to `firmware`.

blocktrron, 2018-01-09:
We did remove the ro flag from the ubnt-airos partition and tried to sysupgrade from that image. Still no luck, sysupgrade seems only to write to the firmware partition. Writing the image again to ubnt-airos partition via mtd, the router boots the new kernel. (https://github.com/lede-project/source/blob/master/target/linux/ar71xx/image/ubnt.mk#L116)

It seems like the router loads the kernel image from ubnt-airos while mounting the rootfs from firmware-partition. There is some kind of bootflag according to this lede-forum article (https://github.com/lede-project/source/blob/master/target/linux/ar71xx/image/ubnt.mk#L116) which decides the default boot partition.

mweinelt, 2018-01-09:
This forum article: https://forum.lede-project.org/t/solved-ubnt-unifi-ac-pro-partition-problems-kernel0-vs-kernel1/1944

>  the Ubnt boot loader image selection magic must be stored in the “bs” partition.

Goes on an describes what needs to be written to change the boot partition.

mweinelt, 2018-01-09:
Upstream issue: https://bugs.lede-project.org/index.php?do=details&task_id=662

rotanid, 2018-01-23:
today i tested one of my UAP AC Pro in depth:
1) flashed stock via TFTP, flashed Gluon v2016.2.4 via mtd, flashed v2016.2.7 via sysupgrade, flashed v2017.1.x via sysupgrade --> still working
2) flashed stock via TFTP, updated stock via "syswrapper.sh upgrade2 &", flashed Gluon v2016.2.4 via mtd, flashed v2016.2.7 via sysupgrade, working, flashed v2017.1.x via sysupgrade, soft-brick!
3) flashed stock via TFTP, updated stock via "syswrapper.sh upgrade2 &", flashed Gluon v2016.2.4 via mtd, flashed v2016.2.7 via sysupgrade, flashed v2017.1.x via sysupgrade --> working again!

so there is a workaround, but a solution would be better of course.

blocktrron, 2018-02-05:
It is possible to write the indicator on BS partition from Stock UniFi Firmware. (Verify mtdpart of bs partition first!) 

`dd if=/dev/mtd4 of=/tmp/bs.bin && echo -ne \\x00 | dd conv=notrunc bs=1 count=1 of=/tmp/bs.bin && mtd write /tmp/bs.bin bs`

Maybe we integrate this into the sysupgrade procedure, alternatively we document this need somewhere. Because there is already a lot of documentation which lacks this step, this is probably not the desired solution (The breakage only occurs on kernel updates).

biggesee, 2018-02-06:
> today i tested one of my UAP AC Pro in depth:
> 
> flashed stock via TFTP, flashed Gluon v2016.2.4 via mtd, flashed v2016.2.7 via sysupgrade, flashed v2017.1.x via sysupgrade --> still working

Was this just a lucky draw, or is a sysupgrade from 2016.2.7 to 2017.1.5 basically harmless?

mweinelt, 2018-02-06:
Pure luck depending on the state of your `bs` partition.

NeoRaider, 2018-02-06:
@blocktrron I think fixing the documentation is a good idea, and is IMO good enough to close this issue (and the upsteam one). Making bs writable from OpenWrt/LEDE would also be nice, so it will be possible to fix this issue after flashing as well in the futute.

I propose to use /dev/zero instead of echo though to get a zero byte. Does the bs partition contain anything interesting after the first byte? Otherwise we could just overwrite it completely. (mtd directly from /dev/zero?)

biggesee, 2018-02-06:
sysupgrade with “mtd write /tmp/firmware.bin kernel0” following“mtd **-r** write /tmp/firmware.bin kernel1”sould also solve the problem, or am I going wrong?

blocktrron, 2018-02-06:
@NeoRaider There is something on the bs partition but i am not able to relate it to something.

@biggesee In case your new Kernel is booting from kernel1 partition, the rootfs is still mounted from kernel0, which leads to problems.

rotanid, 2018-02-07:
> Was this just a lucky draw, or is a sysupgrade from 2016.2.7 to 2017.1.5 basically harmless?

my test proofs, that there is a flag being switched with each stock upgrade. so yes, the outcome of these tests are different depending on where your start.

rotanid, 2018-02-11:
@blocktrron your upstream change was merged, maybe you could document your fix on upstream's wiki and link to that in upstream bugreport and here?

mweinelt, 2018-02-11:
> >>>>> Wiki-merge in process >>>>> wiki.openwrt.org + openwrt.org are going to be merged soon, this wiki therefore being read only. Once pages are transfered to openwrt.org, you can edit them again.

pbriesch, 2018-02-24:
Not sure if I'm at the right place here.
Just tried to flash gluon v2017.1.5 to an Unifi AP AC Pro which runs stock firmware BZ.v3.9.19.
There seems to be no _mtd_ command any more:
```
BZ.v3.9.19# mtd
-sh: mtd: not found
BZ.v3.9.19#
```

How do I flash?

MarcProe, 2018-02-28:
Downgrade to 3.7 stock with this firmware: http://dl.ubnt.com/unifi/firmware/U7PG2/3.7.21.5389/BZ.qca956x.v3.7.21.5389.161017.0852.bin

mweinelt, 2018-02-28:
I just updated the documentation in the openwrt wiki: 

https://openwrt.org/toh/ubiquiti/unifiac#installing_openwrt

MarcProe, 2018-02-28:
I was able to install gluon on an AP-AC-PRO-Gen2 by first downgrading it to 3.4.7 (just "before march 2017" is not enough) via TFTP ([using the firmware.bin image mentioned here](https://gist.github.com/lg/998d3e908d547bd9972a6bb604df377b)), then flashing the &lt;gluon&gt;-ubiquiti-unifi-**_ac-pro-sysupgrade_**.bin to kernel0 with `mtd -r kernel0 write /tmp/<gluon>-ubiquiti-unifi-ac-pro-sysupgrade.bin kernel0`

mweinelt, 2018-02-28:
Firmware 3.4.70 is the latest one before march 2017, and it has `mtd` at `/bin/mtd`. So "just before march 2017" seems quite alright. I downloaded that firmware from https://www.ubnt.com/download/unifi/unifi-ap-ac-pro/uap-ac-pro.

Can you please post the output of 
```
# cat /proc/mtd
```
Ultimately the issue is not that you cannot flash or boot the device, it is the issue that booting may break when a kernel ABI bump happens.

MarcProe, 2018-02-28:
On the site you are referencing, the firmware just before march is 3.**7.40**, not 3.**4.70** ([Screenshot](https://i.imgur.com/d9jh1d5.png)).
However, using this firmware I was not able to flash gluon. I needed 3.**4.7** (without the zero). 
See also [here ](https://discuss.ct-networks.io/t/openmesh-1750-ac-unifi-ac-lite/258/43) and [here](http://ffhgw.de/howto-unifi-ac-lite-ap-flashen/).

I'll post the `/proc/mtd` output tomorrow, as I don't have access to the device right now.

mweinelt, 2018-02-28:
Okay, mixup in the version number. What does not work for you on 3.7.40?

Looking at the first log It is more likely that the device booted from `kernel1` and you flipped the bootselect to `kernel0` when you flashed the older stock firmware, thus making it work. But that does not invalidate the way bootselect works.

MarcProe, 2018-02-28:
Can't tell what exactly caused the failure on 3.7.40, the device kept rebooting into stock. I'll be getting 11 of the devices in about 4 weeks for use with gluon, I can do some more tests then. I'm not an expert on the bootselect or mtd, so I can't really give a guess about the reasons.

mweinelt, 2018-02-28:
Ok, walking you through the situation:

1. Unifi AC devices feature a flip-flop flash layout, that means:
1.1 There are two partitions (`kernel0`, `kernel1`) containing the firmware
1.2 The bootselect (located on partition `bs`) devices which partition to boot
1.3. With every stock upgrade the upgrade is written to the inactive partition and bootselect gets flipped  to that partition, always leaving a known-working state behind
1.4. If a partition doesn't boot it will reboot into the other one

2. There is a chance that your `bs` partition contains the correct value already - but it also might not.

3. OpenWrt's sysupgrade will currently only upgrade the `kernel0` partition,
3.1 You still need to flash `kernel0` and `kernel1` because the bootloader prefers a signed image (Ubiquiti Stock) over an unsigned one (OpenWrt)

4. OpenWrt will load the root filesystem from `kernel0`, even if it booted the kernel from `kernel1`
4.1 If the kernel modules from partition `kernel0`, due to an upgrade, become incompatible with the kernel on partition `kernel1` and `kernel1` is booted … it will result in a kernel panic.

5. `mtd` was removed from the stock firmware sometime in march/april 2017
5.1 `mtd` is needed to write the sysupgrade image, `dd` is needed to write the bootselect nullbyte
5.2. You can get `mtd` from an earlier stock image or from an old OpenWrt that used uClibc

MarcProe, 2018-02-28:
Thanks for giving those details, this helps a lot. 
I can totally not rule out that I just became confused with the kernel(0|1) partitions. I'll try and flash both with gluon on 3.7.40 and report back.

rotanid, 2018-02-28:
closing this issue as there is a workaround and a perfect solution is unlikely to appear in the near future.
discussion is still possible here, though.

closed 2018-02-28 by rotanid

MarcProe, 2018-03-01:
Just in case it helps:
```

# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00060000 00010000 "u-boot"
mtd1: 00010000 00010000 "u-boot-env"
mtd2: 00790000 00010000 "firmware"
mtd3: 00140000 00010000 "kernel"
mtd4: 00650000 00010000 "rootfs"
mtd5: 003e0000 00010000 "rootfs_data"
mtd6: 00790000 00010000 "ubnt-airos"
mtd7: 00020000 00010000 "bs"
mtd8: 00040000 00010000 "cfg"
mtd9: 00010000 00010000 "EEPROM"

```

mweinelt, 2018-03-01:
Ok, looks like the firmware partitions on the UAP-AC-Pro (that is your device, right?) are `firmware` and `ubnt-airos`. Also it looks like you're already running OpenWrt on that device, correct?

MarcProe, 2018-03-01:
Both correct.
Today I reverted to stock firmware 3.4.16 and flashed gluon again on kernel0 and kernel1  and it worked wihthout any problems.

yunhai20082008, 2022-10-23:
> Flashing the current Gluon master on a UBNT UniFi AP AC Mesh results in a broken WiFi interface.
>
> `iw dev` results in `nl80211 not found.`
>
> ## logread
> ```
> BusyBox v1.25.1 () built-in shell [...]

could you do me a favor about this?https://forum.openwrt.org/t/need-a-copy-content-of-full-flash-about-uap-ac-m/140438