freifunk-gluon/gluon #1302: segfault while running autoupdater. logread output saved

lrnzo, 2018-01-10:
output of autoupdater:
``Connecting to autoupdate.ffnw ([2a01:4f8:192:832b:8::1]:80)``
``-                    100% |*...***| 55216   0:00:00 ETA``
``New version available.``
``Stopping cron...``
``Stopping haveged...``
``Stopping micrond...``
``Stopping sysntpd...``
``Stopping gluon-radvd...``
``Stopping uhttpd...``
``Stopping sse-multiplexd...``
``Stopping gluon-respondd...``
``vm.drop_caches = 3``
``Connecting to autoupdate.ffnw ([2a01:4f8:150:50a2::107]:80)``
``lua_E66Czp           100% |*...**|  3840k  0:00:00 ETA``
``Stopping network...``
``Segmentation fault``
``vm.drop_caches = 3``
``Upgrading firmware...``
``Saving config files...``
``Commencing upgrade. All shell sessions will be closed now.``

``logread -f`` was running in another terminal and the output can be found at https://pastebin.com/EywZcQLu

rotanid, 2018-01-10:
what is your gluon release version?
does this happen with the same firmware and same community on other devices or does it only happen on this device?

lrnzo, 2018-01-10:
it occured an a CPE210 v1.0 with gluon-release 2016.2.6. I never realized it before but today happend to me twice. but i have only this one logread.

rotanid, 2018-01-10:
sorry i have bad news, i was told that your issue is a duplicate of other issues:
#680 #910 #1258
also, we probably cant/wont fix this in this old release branch anymore.

closed 2018-01-10 by rotanid