freifunk-gluon/gluon #1304: odhcp6c spawns processes frequently, creating noticeable load

T-X, 2018-01-13:
Every time the odhcp6c receives an ICMPv6 Router Advertisement, it spawns a shell script: /lib/netifd/dhcpv6.script.

With the default intervals for a radvd or bird sending RAs on a gateway, a gateway will issue a broadcasted RA to ff02::1 every 3 seconds due to the MinDelayBetweenRAs and the many Router Solicitation messages it receives.

With for instance six gateways this accumulates to two RAs per second received on a node and therefore leading to two script spawns per second. This can create a significant amount of load, especially if the CPU is already quite busy anyways.

Not sure what the best solution to tackle this would be. Feel free to discuss :-).

christf, 2018-01-22:
The node-IP is static and based on the mac addresses, routes are transported via the mesh-protocol.
Why do we need to have this script spawned when receiving RA on the mesh and wouldn't it be enough it this was executed for RA from the WAN-side?

rotanid, 2018-01-23:
although your point seems valid, last weekend we had someone whose local WAN router was sending to many RAs which lead to problems on the Gluon node, too

NeoRaider, 2018-01-23:
At the moment, a shell script is used to transfer RA information to netifd. This is necessary to update prefix and route lifetimes.

Implementing direct netifd signalling via ubus in odhcp6c would solve this. This should be a configuration option (odhcp6c must still be buildable withous ubus), so it can be done upstream-first (i.e. in the odhcp6c master).

T-X, 2018-06-05:
According to @NeoRaider on IRC the limit for dhcpv6.script spawns should be 30s on 17.01 and 3s in 18.06 (for RFC compliance).

The former should probably not cause any high load. I'm not sure anymore why I thought I saw rates higher than 3s of dhcpv6.script spawns. Will need a reevaluation.

mweinelt, 2018-12-04:
And apparently 3s are a problem. For more details see https://forum.darmstadt.freifunk.net/t/hohe-load-mit-gluon-testing-1-3-20181118-mehrfachaufrufe-von-dhcpv6-script-br-client-ra-updated/707.

Can apparently be remedied via

```sh
# uci set network.client.ra_holdoff=30
# uci commit network
# reboot
```

but as that is not rfc conform...

rotanid, 2018-12-07:
workaround/fix merged in 76f591995e359f10cc7b06f6b03428a57b0886a7

closed 2018-12-07 by rotanid