freifunk-gluon/gluon #1305: Remove/unload unused modules?

T-X, 2018-01-13:
There are several modules, especially for netfilter and packet scheduling, being loaded into RAM on startup which we do actually not use.

I did some tests with this script and the module lists in here: https://gist.github.com/T-X/a6cf0f1e33122904bc4ecf8cb2555707

Unloading those network related modules seems to save about 400-440KB of RAM.

Furthermore, some communities have added USB related packages to their site.mk. Unloading USB related kernel modules as listed in the above gist saves about ~305KB.

Unloading both at the same time gave me slightly more freed space than the sums above, about 760-800KB (due to pages shared between modules?).

This is not as much as I had hoped for. However,  800KB are still 2.5% of a 32MB RAM device.

Therefore the proposal here: Remove the unused netfilter and scheduler modules from the images and try to unload USB related modules before fetching a new image for the autoupdate (/usr/lib/autoupdater/download.d/).

christf, 2018-01-21:
actually... can you please elaborate - why keep the modules in place and only unload for autoupdate?

T-X, 2018-01-22:
@christf: Good question. I think someone from a community who is building images with USB modules should explain a bit how they are using them.

Not sure whether we should accumulate a modules blacklist for all modules a community might add via their site.mk without knowing their use-case. Adding such things like USB modules at least for a try to unload to before updating I'd feel safe to do without stepping on anyone's toes.

A-Kasper, 2018-01-22:
We are using them in x86 images where we have enough memory. They are used for USB network interfaces in Futro-Based Offloader. There we also added some default USB Stuff for keyboard support etc. I think this is a problem that should be addressed via documentation: add usb modules only where is enough memory and not stupid to all targets. 

in our site.mk this reads:

#####

ifeq ($(GLUON_TARGET),$(filter $(GLUON_TARGET),x86-generic x86-kvm_guest  x86-xen_domu x86-64))
# support the usb stack on x86 devices
# and add a few common USB NICs
GLUON_SITE_PACKAGES += \
        kmod-usb-core \
        kmod-usb2 \
        kmod-usb-hid \
        kmod-usb-net \
        kmod-usb-net-asix \
        kmod-usb-net-dm9601-ether\
        kmod-carl9170\
        kmod-r8169 \
        kmod-rtl8192cu\
        kmod-usb-net-rtl8150\
        kmod-usb-net-rtl8152\
        kmod-ath9k-htc
endif
#####

(Sorry, I didn't manage to stop github from formating this)

I would recommend to add an block like this in the site.mk example to show how it is done. Than we don't have to worry about unloading the modules.

rotanid, 2018-01-23:
we shouldn't focus too much on only a small part of modules from @T-X 's list, please.
regarding that small part, you shouldn't unload usb modules that are e.g. used for usb-networking. some people also use usb-sticks for storage. so the usb modules are more tricky than other modules may be.
also, i don't think we want to talk about x86 devices here @A-Kasper , the problem are devices with only 32MB of memory - and i don't think someone uses x86 devices with only 32MB and Gluon.

A-Kasper, 2018-01-23:
My point was, that we shouldn't care about community loaded modules. They need to handle them on their own. There are many ways to get memproblems by adding stuff to gluon. Whoever adds something has to take care that it works. You can't handle all usecases.
The named usb drivers are often added to images where they aren't needed. I suggested to show how to add them to just the images where they are needed.

For other cases we could think about an additional unload modules section in site.conf, where communities can add their custom modules on their own.

NeoRaider, 2018-01-23:
A fully booted, idle node still has about 10MB of free memory. We can start microoptimizing memory usage... or... we can work on the root issue, that seems to be networking bufferbloat. I strongly suggest we don't care about things like loaded modules for now (and in any case, I would like to avoid Gluon-specific hacks that aren't upstreamable to OpenWrt...)

rotanid, 2018-04-20:
@T-X i guess this issue is solved with the commits from about a week ago from @NeoRaider ?
feel free to close this issue yourself, if you agree.

T-X, 2018-06-05:
I only looked at @NeoRaider's changes briefly but yes, seems like it.

Thanks for pointing that out, @rotanid! Closing this issue.

closed 2018-06-05 by T-X