freifunk-gluon/gluon #1307: TP-Link TL-WR1043n V5 don't find autoupdate image

Byggvir, 2018-01-14:
The TP-Link TL-WR1043n V5 reports as model **tp-link-tl-wr1043n-v5** in the manifest it is listed as **tp-link-tl-wr1043n-nd-v5**.

Autoupdater doesn't work for the backport CodeFetch/2017.1.x and the gluon/master.

```
root@su-rhb-mittelerde-02:~# autoupdater
Downloading 'http://[fda0:747e:ab29:2241:16cc:20ff:fecd:15bc]/fffw/bp/images/sysupgrade/stable.manifest'
Connecting to fda0:747e:ab29:2241:16cc:20ff:fecd:15bc:80
Writing to stdout

                   100% |*******************************| 45160   0:00:00 ETA

Download completed (45160 bytes)
No matching firmware found (model tp-link-tl-wr1043n-v5)
No usable mirror found.
```

```
root@su-rhb-vpk-01:~# autoupdater 
Downloading 'http://[fda0:747e:ab29:2241:a62b:b0ff:fedf:2901]/fffw/master/images/sysupgrade/master.manifest'
Connecting to fda0:747e:ab29:2241:a62b:b0ff:fedf:2901:80
Writing to stdout
-                    100% |*******************************| 36154   0:00:00 ETA
Download completed (36154 bytes)
No matching firmware found (model tp-link-tl-wr1043n-v5)
Downloading 'http://[fda0:747e:ab29:2241:16cc:20ff:fecd:15bc]/fffw/master/images/sysupgrade/master.manifest'
Connecting to fda0:747e:ab29:2241:16cc:20ff:fecd:15bc:80
Writing to stdout
-                    100% |*******************************| 36154   0:00:00 ETA
Download completed (36154 bytes)
No matching firmware found (model tp-link-tl-wr1043n-v5)
No usable mirror found.
```
master.manifest
```
...
tp-link-tl-wr1043n-nd-v5 master-20180105 4a94bb84b7abe7486e28f7a49000cb37252cf0c9457be4ed9b3edab21425853a gluon-su-rhb-master-20180105-tp-link-tl-wr1043n-nd-v5-sysupgrade.bin
tp-link-tl-wr1043n-nd-v5 master-20180105 b79e63c427df68b203829493b365ff5c7b35d8fcdf8a7e4464e6348062ad5b7ab16f49d05275dcd73e1ba97ba971ee7b00363c73e32361b323a20c1b4c8ed0eb gluon-su-rhb-master
...
```

Byggvir, 2018-01-14:
I changed the model in the master.manifest manually from **tp-link-tl-wr1043n-nd-v5** to 
**tp-link-tl-wr1043n-nd-v5** and copied it to the update server. After this change the update worked fine.

```
....
tp-link-tl-wr1043n-v5 master-20180114 6b3648964d23d0d2f20cb92ab1c389a565f1598a396984878cec68c5f2088b90 3866809 gluon-su-rhb-master-20180114-tp-link-tl-wr1043n-nd-v5-sysupgrade.bin
tp-link-tl-wr1043n-v5 master-20180114 6b3648964d23d0d2f20cb92ab1c389a565f1598a396984878cec68c5f2088b90 gluon-su-rhb-master-20180114-tp-link-tl-wr1043n-nd-v5-sysupgrade.bin
tp-link-tl-wr1043n-v5 master-20180114 dfc62c3298b8361762ddbd3fd3d6fe14687a6b862ca4ee0f2dfa8f391760626dade42463ddb47852f2757e1eae5bbea01010fe21a004082f4c55d2b70248012c gluon-su-rhb-master-20180114-tp-link-tl-wr1043n-nd-v5-sysupgrade.bin
...
```
Autoupdater with changed model name

```
root@su-rhb-mittelerde-02:~# autoupdater 
Downloading 'http://[fda0:747e:ab29:2241:a62b:b0ff:fedf:2901]/fffw/master/images/sysupgrade/master.manifest'
Connecting to fda0:747e:ab29:2241:a62b:b0ff:fedf:2901:80
Writing to stdout
-                    100% |*******************************| 59379   0:00:00 ETA
Download completed (59379 bytes)
New version available.
No autoupdate this time. Use -f to override.
root@su-rhb-mittelerde-02:~# autoupdater -f
Downloading 'http://[fda0:747e:ab29:2241:a62b:b0ff:fedf:2901]/fffw/master/images/sysupgrade/master.manifest'
Connecting to fda0:747e:ab29:2241:a62b:b0ff:fedf:2901:80
Writing to stdout
-                    100% |*******************************| 59379   0:00:00 ETA
Download completed (59379 bytes)
New version available.
Stopping cron...
Stopping haveged...
Stopping micrond...
Stopping sysntpd...
Stopping gluon-radvd...
Stopping uhttpd...
Stopping sse-multiplexd...
Stopping alfred...
sed: /etc/crontabs/root: No such file or directory
Command failed: Not found
Stopping gluon-respondd...
vm.drop_caches = 3
Downloading 'http://[fda0:747e:ab29:2241:a62b:b0ff:fedf:2901]/fffw/master/images/sysupgrade//gluon-su-rhb-master-20180114-tp-link-tl-wr1043n-nd-v5-sysupgrade.bin'
Connecting to fda0:747e:ab29:2241:a62b:b0ff:fedf:2901:80
Writing to '/tmp/lua_FmHjEc'
/tmp/lua_FmHjEc      100% |*******************************|  3776k  0:00:00 ETA
Download completed (3866809 bytes)
Stopping network...

```

blocktrron, 2018-01-14:
This seems to be an oversight by myself while trying to keep the naming scheme in line with prior models.
I've added a manifest-alias to this device here: https://github.com/blocktrron/gluon/commit/3e11973457b5076d648cb436c479336a2eb1ec26

Manifests generated with this commit will look like this: 
```
tp-link-tl-wr1043n-nd-v5 1.1~20180114 ce7f427c21cf9ef03a29fc3d865920cbdfc112d0b834c6eb8cc8a396a03f5573 4718774 gluon-ffda-1.1~20180114-tp-link-tl-wr1043n-nd-v5-sysupgrade.bin
tp-link-tl-wr1043n-nd-v5 1.1~20180114 ce7f427c21cf9ef03a29fc3d865920cbdfc112d0b834c6eb8cc8a396a03f5573 gluon-ffda-1.1~20180114-tp-link-tl-wr1043n-nd-v5-sysupgrade.bin
tp-link-tl-wr1043n-nd-v5 1.1~20180114 1b5ad7bd283dadea9a5a67159de04f942bde650ed7fb9fde9d238e3e0e9037ea375574db637937d14795f83b4403d2ad5336a68655aba53e5abd678a48d02ec1 gluon-ffda-1.1~20180114-tp-link-tl-wr1043n-nd-v5-sysupgrade.bin
tp-link-tl-wr1043n-v5 1.1~20180114 ce7f427c21cf9ef03a29fc3d865920cbdfc112d0b834c6eb8cc8a396a03f5573 4718774 gluon-ffda-1.1~20180114-tp-link-tl-wr1043n-nd-v5-sysupgrade.bin
tp-link-tl-wr1043n-v5 1.1~20180114 ce7f427c21cf9ef03a29fc3d865920cbdfc112d0b834c6eb8cc8a396a03f5573 gluon-ffda-1.1~20180114-tp-link-tl-wr1043n-nd-v5-sysupgrade.bin
tp-link-tl-wr1043n-v5 1.1~20180114 1b5ad7bd283dadea9a5a67159de04f942bde650ed7fb9fde9d238e3e0e9037ea375574db637937d14795f83b4403d2ad5336a68655aba53e5abd678a48d02ec1 gluon-ffda-1.1~20180114-tp-link-tl-wr1043n-nd-v5-sysupgrade.bin
```

I will confirm this working next time i have access to a TL-WR1043v5. If you like, you can test it for yourself if it solves this problem.

rotanid, 2018-01-15:
@Byggvir thank you for your report.
in your second comment you wrote "from tp-link-tl-wr1043n-nd-v5 to
tp-link-tl-wr1043n-nd-v5"
this is the same string twice, maybe use the edit button to correct this part of the text.

@blocktrron i am unsure whether we support such oversights in the master branch, meaning if we should really add a manifest_alias which will stay there forever or if we just change to the correct string?
@NeoRaider will have to decide in this case i guess

Brother-Lal, 2018-01-16:
`MIPS_MACHINE(ATH79_MACH_TL_WR1043N_V5, "TL-WR1043N-v5", "TP-LINK TL-WR1043N v5",`

I would just change to the correct string - neither the openwrt/lede nor tplink have the nd in the description - we had the same discussion for the support, keep the naming consitent or use the correct name, and the later was choosen (to my dismay, but its correct...)

Here was the error:
https://github.com/freifunk-gluon/gluon/commit/70784cb3b24fea9da2b40c62af467d8b1cb6abfa#diff-9782a3f3966b5ff8070d88b538ff4337R171

NeoRaider, 2018-01-16:
The value from MIPS_MACHINE is not relevant for the autoupdater. The model name is defined in /lib/ar71xx.sh for TP-Link devices.

rotanid, 2018-01-16:
fixed

closed 2018-01-16 by rotanid

Byggvir, 2018-01-18:
Tested with me nightly build of master. Autoupdater works fine now, but the release in /lib/gluon/release is not updated on the first run.

Byggvir, 2018-01-21:
My new build from freifunk/gluon v2017.1.4 doesn't autoupdate.

```
tp-link-tl-wr1043n-v5 v2017.1.4-20180120-stable-usb 450d10ee85b4cb5a9709c0b67a17913a92dbaccf89cac4f02add12944bb5b8f2 gluon-su-rhb-v2017.1.4-20180120-stable-usb-tp-link-tl-wr1043n-v5-sysupgrade.bin
tp-link-tl-wr1043n-v5 v2017.1.4-20180120-stable-usb ef21259fed6da0256dc6c0aa2d5af98e6f304c90c1eef229b190b442e2bb2fa0d7db896c9d8f3c9c209e6dff77a2f1b959e2ec23b74add7cf112782c45af7c36 gluon-su-rhb-v2017.1.4-20180120-stable-usb-tp-link-tl-wr1043n-v5-sysupgrade.bin
```

Report: 
```
root@su-rhb-mittelerde-02:~# autoupdater 
Retrieving manifest from http://[fda0:747e:ab29:2241:a62b:b0ff:fedf:2901]/fffw/stable/images/sysupgrade//stable.manifest ...
autoupdater: warning: no matching firmware found (model tp-link-tl-wr1043n-v5)
Retrieving manifest from http://[fda0:747e:ab29:2241:16cc:20ff:fecd:15bc]/fffw/stable/images/sysupgrade//stable.manifest ...
autoupdater: warning: no matching firmware found (model tp-link-tl-wr1043n-v5)
autoupdater: error: no usable mirror found
```

commit fb93c3b2 (2018-01-16): ar71xx: correct model name for TL-WR1043N v5, fixes #1307
commit 9295abce (2018-01-17): ar71xx: correct model name for TL-WR1043N v5, fixes #1307