freifunk-gluon/gluon #1310: upgrade scripts fail due to replaced autoupdater

lephisto, 2018-01-16:
The Upgrade Script of the autoupdater fails due to replacement of the package in gluon-packages:

`/lib/gluon/upgrade/500-autoupdater:/usr/bin/lua: /lib/gluon/upgrade/500-autoupdater:29: module 'autoupdater.util' not found:
        no field package.preload['autoupdater.util']
        no file './autoupdater/util.lua'
        no file '/usr/share/lua/autoupdater/util.lua'
        no file '/usr/share/lua/autoupdater/util/init.lua'
        no file '/usr/lib/lua/autoupdater/util.lua'
        no file '/usr/lib/lua/autoupdater/util/init.lua'
        no file './autoupdater/util.so'
        no file '/usr/lib/lua/autoupdater/util.so'
        no file '/usr/lib/lua/loadall.so'
        no file './autoupdater.so'
        no file '/usr/lib/lua/autoupdater.so'
        no file '/usr/lib/lua/loadall.so'
stack traceback:
        [C]: in function 'require'
        /lib/gluon/upgrade/500-autoupdater:29: in main chunk
        [C]: ?
 error
`
This happens because of [https://github.com/freifunk-gluon/packages/commit/49cb4b3fdb8b9db0abb81783b9b3b386a8ed6a92](url).

It breaks some stuff, because several other script are kept from being executed..

NeoRaider, 2018-01-17:
For the record: This was breaking the creation of the autoupdater cron job, nothing else.

closed 2018-01-17 by NeoRaider

lephisto, 2018-01-17:
.. als the jobs afterwards when for instance running domain_changed.sh

> 500-mesh-vpn
> 500-node-info-system
> 500-opkg
> 500-radvd-remove-user
> 500-status-page-api
> 510-node-info-role
> 520-node-info-whitespace-fix
> 800-migrate-batadv
> 820-dns-config
> 981-firewall-status-page
> 998-commit
> 999-version

I noticed that while testing domain-select. Thanks for fixing.

NeoRaider, 2018-01-17:
This is an issue in domain_changed.sh then, failing scripts should print only print an error message and not break anything else.

lephisto, 2018-01-17:
@kb-light should maybe have a look at this.

rotanid, 2018-01-17:
no, @lemoer is the author of domain_changed.sh ;)

commit 18feb29b (2018-01-17): gluon-autoupdater: don't reference old autoupdater util library

Fixes #1310