freifunk-gluon/gluon #1317: ath10k, TP-Link Archer C7 v2

mm-chen, 2018-01-24:
logread https://pastebin.com/1VKuVMh6
LEDE Version: LEDE Reboot 17.01-SNAPSHOT r3581+39-6b6578feec (r3581+39-6b6578feec)
Gluon Version: v2017.1.4-4-g2ae74fe7+

mm-chen, 2018-01-24:
https://pastebin.com/ZyhXNaCv

Casey1979, 2018-01-25:
These incrorrect idle disconnects happened in pure Lede Trunk too, that probably was just a bad snapshot as it didn't happened to me again after i tried a later snapshot from 22.10.2018 on an Archer C7 V2

rotanid, 2018-01-26:
@mm-chen github allows to edit your comments, i did it for you this time.
could you please also edit your text adding some information on what your problem is? also your problem should be in the Title of your issue.
you can't expect everyone to open your issue, open your link and only then knowing the problem you want to report.

regarding your issue, did you have problems all the time or did it work on an older build, if so, which?

@Casey1979 are you talking about "real" snapshot builds? because Gluon is using the lede-17.01 stable branch only at the moment.

Casey1979, 2018-01-26:
Yes i meant pure Lede Trunk Snapshots not what Gluon is using as my Archer C7 V2 is working not for Freifunk but at the moment as a Homenet Accesspoint with WPA2 and 802.11w.
Lede stable at least for me had no idle disconnect issues but was prone to ath10k Firmware crashes related to this bug https://dev.openwrt.org/ticket/20854 when Qualcomm Firmwares were in use.
ath10k_pci 0000:01:00.0: failed to delete peer X for vdev 0: -145
ath10k_pci 0000:01:00.0: found sta peer entry X on vdev 0 after it was supposedly removed
Candelatech ath10k Firmware was ok but with only 40% of the QC Performance in this very bad RF-Environment.
The latest Lede Trunk Snapshots with QC Firmware finally worked well without ath10k Firmwarecrashes or incorrect idle disconnect and good Performance which may help mm-chen on gluon too at some time.

rotanid, 2018-02-13:
no response, closing

closed 2018-02-13 by rotanid