freifunk-gluon/gluon #1320: Next-node IPv6 ULA not working if node has an uplink

tackin, 2018-01-30:
Using Gluon 2017.1.x

Our site.conf uses
prefix6 = 'fd6a:2c55:3c00:9d0c::/64',
next_node = {
		ip6 = 'fd6a:2c55:3c00:9d0c::1',
		mac = '16:41:95:40:f7:dc',
},

As Client I can connect to the next-node-address fd6a:2c55:3c00:9d0c::1 as long as the node has no Uplink.
After the node has an Uplink to our GWs the fd6a:2c55:3c00:9d0c::1 is suddenly unreachable but the node is still reachable on its individual fd6a:2c55: ... -address.
As checked with neoraider, no probably odd config for RA on our GW seams to be responsible for this behaviour.
We actually suspect an issue in gluon and the node itself.

rotanid, 2018-01-30:
@tackin which v2017.1.x commit is your firmware based on? please edit your report to include the exact commit message, thanks :)

NeoRaider, 2018-01-30:
Content of `/lib/gluon/gluon-version`? Please also link your site configuration.

rotanid, 2018-02-03:
for the record: he added the requested information to his initial report.

rotanid, 2018-02-16:
@tackin i could not reproduce this issue using a WR841N v11 and trying many combinations...

could you please add the following information:
1) how is your client connected? cable or wifi?
2) what is "node has an Uplink"? is it via VPN or via ibss-mesh or via 11s-mesh or via mesh-on-lan or via mesh-on-wan or is the issue existing on all these types of uplink?
3) did you try more than one type of hardware?
4) could you try the latest v2017.1.5 , too ?

tackin, 2018-02-16:
@rotanid 
1. Wifi
2. fastd-uplink or ibss-mesh. No MoL or MoW tested
3. yes
4. did ...
Firmaware: https://github.com/freifunktrier/firmware_store/tree/master/firmware/stable/factory
Gluon 2017.1.5, pulled yesterday
Test-Router: TP-Link WR1043v1
did:  uci set wireless.ibss_radio0.disabled='1'; wifi
Node-WAN is connected to the DSL-Router
Node builds a vpn-Tunnel with fastd to one of our GWs
Connected an Android-client via WLAN to the node.
Android-Client has connection to the internet.
Android-Client connects via http:/router.fftr (DNS resolves fd6a:2c55:3c00:9d0c::1) to the Node
Android-Client connects via http://[fd6a:2c55:3c00:9d0c::1]/  to the node
did:  uci set wireless.ibss_radio0.disabled='0'; wifi
Node gets ibss-mesh with other Nodes
reconnect Android Client to Node via WLAN
Tried from Andorid -Client:  
http:/router.fftr      no response
http://[fd6a:2c55:3c00:9d0c::1]/  no response
now LanDroid tells me: fd6a:2c55:3c00:9d0c::1/128 via fe80::147b:d9ff:fe16:7bb0 dev wlan0
default route is via fe80::147b:d9ff:fe16:7bb0
did:  uci set wireless.ibss_radio0.disabled='1'; wifi
reconnect my Android-Client via WLAN
Tried from Andorid-Client:  
http:/router.fftr    no response
http://[fd6a:2c55:3c00:9d0c::1]/  no response
- shrug - 
Of course timing-issues for RA, default routes and IPs may also be responsable for the behavior. If you like, flash our firmware to your node, send me the fastd-key and test with our firmware environment.

tackin, 2018-02-16:
Next test:
http://[2001:bf7:fc0f:1:1ad6:c7ff:fe51:a3da]/
TP-Link WR1043v4 Gluon 2017.1.5 build yesterday
Node hast fastd-uplink and ibss-mesh
connect Android-Client via Wifi
Client has full Internet-access.
tried: http://[fd6a:2c55:3c00:9d0c::1]/ no response

tackin, 2018-02-16:
Next test:
TP-Link Archer C7v2
Gluon 2017.1.5 build yesterday
node has no fastd-uplink, no ibss-mesh, no MoL, no MoW
connect Android-Client via Wifi to the node
Client has no Internet access
tried: http://[fd6a:2c55:3c00:9d0c::1]/   ... works!

tackin, 2018-02-16:
Next test:
TP-Link Archer C7v2
Gluon 2017.1.5 build yesterday
node has no fastd-uplink, no ibss-mesh, no MoL, no MoW
connect Win7-Client via Cable to the node
Client has no Internet access
tried: http://[fd6a:2c55:3c00:9d0c::1]/ ... works!

tackin, 2018-02-16:
final test:
disconnect Win7-Client on Cable
let node get fastd-uplink via WAN
wait until node has established the fastd-uplink
reconnect Win7-Client via cable
tried: http://[fd6a:2c55:3c00:9d0c::1]/      no response

rotanid, 2018-02-16:
will test with your linked firmware.
i generated a key in advance for testing, here's the public key:
5c8f8afe2ef30b5ecdc3e0f09d723dfe7caec8aae50607d8216ed628745f9a10

tackin, 2018-02-16:
Your key is registered now to our segment 05.
You have your own L2-net-segment.

rotanid, 2018-02-16:
@tackin i tested it now with a WR841N v11 with your firmware.
although the next-node address takes a few seconds longer then the individual address, it becomes reachable from my Linux Laptop even when Mesh-VPN is connected.
so i think you did either not wait long enough or ... i don't know what you're doing differently.

rotanid, 2018-02-16:
the issue is reproducible in another segment of Freifunk Trier, but as it doesn't happen in every segment, it doesn't look like an issue with Gluon but maybe some other nodes or clients sending specific traffic patterns?

tackin, 2018-02-17:
Sorry, but negativ!
I could reproduce the problem with only 2 of my nodes (both on 2017.1.5 last commit) in an otherwise empty segment with no other nodes.
Szenario: [DSL] ---<fastd-uplink>---- [WAN-Port | WR1043v1 | mesh-WLAN]----[mesh-WLAN | WA801v1 | Client-WLAN]----[Client-PC]
Client-PC cannot use http://[fd6a:2c55:3c00:9d0c::1]/  no response from WA801 
If I  connect Client-PC via WLAN directly to WR1043v1  http://[fd6a:2c55:3c00:9d0c::1]/  works

rotanid, 2018-02-17:
ok, reopening until further tests.
@tackin please also test what happens if both devices in your empty segment have VPN, if the next-node address isn't reachable on both then.
after that, please disable wireless-mesh on both and test if that changes anything.

tackin, 2018-02-17:
After testing it for an hour or so in the morning with the upper discouraging result, I let the szenario run for more than 6 hours now to test it again and must confess: Next-Node-Address works now and also under all requested circumstances.
It's hard to find the reason for the obvious timing-issue and would'nt help us at all.
So I'm sorry to see that happen but glad ot tell, gluon has no issue and therefore please close this issue again.
The problem in in our other segments might be for other reasons. I will do some more research on that.

tackin, 2018-02-22:
Solved now!
There are two parts of the problem.
1. Some of the nodes with uplink need some more time to send RA with fd6a:2c55:3c00:9d0c:: Prefix. So a client at first has no idea where to search for the local-node fd6a:2c55:3c00:9d0c::1
My solution: patience
2. If it finally has a lease and build its fd6a:2c55:3c00:9d0c::.... address, it starts Neighbor Solicitation for fd6a:2c55:3c00:9d0c::1 ... after the http-request, but don't get an answer from the node.

This 2. problem could finally be solved by adding **gluon-ebtables-segment-mld** to the site.mk
The package should be a fixed part of gluon in next versions and not only an optional thing.

Thanks to rotanid and neoraider for your kind support!

NeoRaider, 2018-03-02:
Reopening as a reminder to move *gluon-ebtables-segment-mld* or parts of it into *gluon-core*.

commit df98bc7b (2018-03-04): gluon-ebtables: unconditionally segment IGMP/MLD

We must ensure that each node becomes IGMP/MLD querier for its local
clients; having only a single querier for the whole mesh is generally
unreliable, leading to frequent "IGMP/MLD querier appeared/disappeared"
messages from batman-adv and unreliable snooping.

In smaller meshes it might be interesting only segment querier domains, but
allow membership reports to pass through the mesh, in order to support
snooping switches outside the mesh without special configuration. A
site.conf switch is provided to control this behaviour.

Fixes #1320
commit 17370d84 (2018-03-07): gluon-ebtables: unconditionally segment IGMP/MLD

We must ensure that each node becomes IGMP/MLD querier for its local
clients; having only a single querier for the whole mesh is generally
unreliable, leading to frequent "IGMP/MLD querier appeared/disappeared"
messages from batman-adv and unreliable snooping.

In smaller meshes it might be interesting only segment querier domains, but
allow membership reports to pass through the mesh, in order to support
snooping switches outside the mesh without special configuration. A
site.conf switch is provided to control this behaviour.

Fixes #1320
PR #1339 (2018-03-07): gluon-ebtables: unconditionally segment IGMP/MLD
We must ensure that each node becomes IGMP/MLD querier for its local
clients; having only a single querier for the whole mesh is generally
unreliable, leading to frequent "IGMP/MLD querier appeared/disappeared"
messages from batman-adv and unreliable snooping.

In smaller meshes it might be interesting only segment querier domains, but
allow membership reports to pass through the mesh, in order to support
snooping switches outside the mesh without special configuration. A
site.conf switch is provided to control this behaviour.

Fixes #1320

@T-X
closed 2018-03-07 by NeoRaider