freifunk-gluon/gluon #1324: ddhcpd development tracking

sargon, 2018-02-05:
The distributed dhcp daemon [1] is still in development, this issue should help track the progress.
The implementation itself is feature complete, for now. So I am currently running a small test-bed for
debugging. My aim for a roadmaps is as following

- [x] small real-world test-bed
- [ ] site wide test-bed, probably in kiel
- [ ] integration into gluon

sargon, 2018-10-07:
We had a first release (r1) on 2018-10-01 and the freifunk kiel firmware team is currently preparing the site wide rollout. There will probably a second release before we hit that target, but things are progressing.

sargon, 2018-10-22:
Okay, this evening we rolled out a new firmware release in our community, which contains an active ddhcpd. Previous to this release we increased the block timeout to 5minutes, which decreased our background noise, as expected, by factor 5. According to wirerrd we have now 15kbit/s of background noice with 300 nodes, from 4 to 5 kbit/s with 36 nodes. A word on wirerrd, I personally don't belive those values, we where not able reproduce them by separate measurements. We have some more improvements in the queue to reduce block requests, which are costly, measured on the amount of needed multicast packages.

rubo77, 2019-03-16:
So what will be needed to make it possible to add a ddhcpd package to gluon?