freifunk-gluon/gluon #1331: web-based mailing list archive

rotanid, 2018-02-15:
mailing list software: mlmmj

we have an existing mailing list, but no web frontend, this is to be done

mweinelt, 2018-02-28:
http://mlmmj.org/docs/readme-archives/

Adorfer, 2018-03-02:
Note that here is big discussiion everywhere about protection of personal data. 
there is one position (not totally mine) which goes: 

- If you want to publish the content of a mailing lists "later" (e.g. via WEB) you need an Opt-In at subscription
- getting permission later "to publish old stuff" is difficult"
- if a ML wants to go public, you need the OK from every subscriber. Practical: Set up a new ML and let every user manually sign up. 
- You have to offer "delete content upon request" ("Recht auf Vergessen") on the archive. 

(I know that this is harsh, but it's a sad reality now.)

NeoRaider, 2018-03-02:
@Adorfer Even more reasons not to handle the archives ourselves.

> If you want to publish the content of a mailing lists (e.g. via WEB) you need an Opt-In at subscription point of time. (On the verification Mail)

The Gluon ML is an open list, you do not need to be on the list to write to the list. Subscribing to the list thus only changes that you now also receive mail from the list. The fact that the mail is archived should simply be stated together the the list's mail address.

> getting permission later "to publish old stuff" is difficult"

I would simply opt not to publish the old mail on the web. Note that the archive is already public: anyone can subscribe to the list and receive all old mails via the mail-based archive commands. This should be stated together with the mail address. At the moment this information can be obtained by sending a mail to gluon+help@luebeck.freifunk.net as described in the subscription notification, but it is somewhat hidden.

> if a ML wants to go public, you need the OK from every subscriber. Practical: Set up a new ML and let every user manually sign up.

See point 1; as it doesn't matter if you're subscribed at all for purposes of archiving, having everyone sign up again does not make sense in my opinion. We would obviously send a notification about the new archives to the mailing list.

> You have to offer "delete content upon request" ("Recht auf Vergessen") on the archive.

marc.info would allow senders to have their mails deleted upon request (see https://marc.info/?q=about ).

Adorfer, 2018-03-02:
(Just to make clear: i highly appreciate to make web archive publicly availabe. I just see a few "challenges" on the way. And how to respond to people stating that such thing should be "illegal" by what real or artificial reason.)

edeso, 2018-03-02:
@Adorfer, @NeoRaider 

not a lawyer myself, but the position @Adorfer states is just one positition, not the law of the land. eg. there is currently no " "delete content upon request" ("Recht auf Vergessen") " law existing in germany  wrt. emails to a public mailing list or forum posts or such..

being practical i would suggest to start archiving from a point in time and pointing that out wherever the list is officially documented. pretty sure that this should suffice the current state of law anyways.

btw. all this is pretty artificial anyways, as pretty much all issues are dealt w/ via github issues. also: i once posted a question on the ml and some people complained about this gluon specific question, wanting to move it to my local freifunk ml, because they only expected Neo to post release announcements and such!

tl;dr
everyone can set up an archive. it is an open list, one can register a listening  email address and arcive away. if the the archive is hosted in tuvalu, nobody can do **** about it!

just my 2 cents.. ede

rotanid, 2018-03-03:
@edeso @Adorfer you both don't get the point, that there is an archive already, only that it is not accessible via HTTP but via SMTP.
i don't get why changing that makes any difference in regards to privacy?

edeso, 2018-03-03:
On 3/3/2018 20:32, Andreas Ziegler wrote:
> @edeso @Adorfer you both don't get the point, that there is an archive already, only that it is not accessible via HTTP but via SMTP.

if you don't mind, how would i access/search the archive via SMTP then?

> i don't get why changing that makes any difference in regards to privacy?
> 

if non subscribers could read messages, that would make a difference in theory. although i am not aware of (software devel)-lists that take that into account. ..ede

Adorfer, 2018-03-03:
If it would be "in the web by accident by some venezolanian hosting: perfect, as long as it's available. 

(just if you have it somewhere over here in DE, sooner or later there will be people complaining. beeing reasonable or not, it will cause work. and if it's to explain why it's totally normal and will not be taken down... going to a next round where there will be stuff like "opt in procedure did not include valid disclaimers etc...". And in the end it will be taken down, just because some trolls are good in their hobby.)

christf, 2018-03-03:
well... even if this happens, we will not be worse off than now. Just my 2¢

rotanid, 2018-03-04:
the suggestion to use some 3rd party like marc.info does leave us without any problems @Adorfer might see...

edeso, 2018-03-04:
@rotanid actually no, as one might argue that whoever administers the ml (@NeoRaider  ?) is responsible for messages ending up in a 3rd party archive, because they usually do not support the double opt-in. they have to be allowed manually. 

but that's all nitpicking leading to no web archive at all. seeing the situation i'd suggest using one of the 3rd party archives and add info about it to gluon+help@... and wherever the ml is promoted, so new users should be aware (gluon+help@ is received on a new subscription i think). that should suffice at least german law, which is likely the one the ml admin want to abide by.

CodeFetch, 2020-02-08:
I don't understand the problem. Why don't we just switch to https://www.freelists.org/ with some anonymous being maintaining it or someone from another country where other laws apply?