freifunk-gluon/gluon #1338: gluon-radv-filterd: originators not updated

belzebub40k, 2018-02-26:
We are using a backport of the `gluon-radv-filterd` in our current stable firmware and so far it looked good. But I just discovered a small problem when the originator MAC of a gateway changes the daemon is no longer able to find a TQ. It looks like the list of known originators is never updated unless a unknown host sending RAs is discovered.

https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-radv-filterd/src/gluon-radv-filterd.c#L568

```
Mon Feb 26 15:04:48 2018 daemon.err gluon-radv-filterd[1695]: Unable to find TQ for originator 02:00:0a:25:01:17 (router 02:42:0a:25:00:17)
Mon Feb 26 15:04:48 2018 daemon.err gluon-radv-filterd[1695]: Unable to find TQ for originator 02:00:0a:25:01:07 (router 02:42:0a:25:00:07)
Mon Feb 26 15:04:48 2018 daemon.err gluon-radv-filterd[1695]: Unable to find TQ for originator 02:00:0a:25:01:e7 (router 02:42:0a:25:00:e7)
```

Translating the router MACs via `batctl` show different _originators_ then logged by daemon.
```
# batctl t 02:42:0a:25:00:17
02:00:0a:25:00:17

# batctl t 02:42:0a:25:00:07
02:00:0a:25:00:07

# batctl t 02:42:0a:25:00:e7
02:00:0a:25:00:e7
```

NeoRaider, 2018-02-26:
@jplitza

belzebub40k, 2018-02-27:
Forgot to mention that the TQs are resolved without a problem once gluon-radv-filterd is restarted.

And for the records these are the backported packages:
https://github.com/freifunk-mwu/packages-ffmwu/tree/v2017.1.x/gluon-radv-filterd
https://github.com/freifunk-mwu/packages-ffmwu/tree/v2017.1.x/libbatadv

belzebub40k, 2018-02-27:
Damn it! I just realized that it was my own idea to implement it like this [on 24 Nov 2016](https://github.com/freifunk-gluon/gluon/pull/838#pullrequestreview-10015169) (**show outdated**).

In general this assumption should be valid but now I think updating the originators from time to time would be a good idea. Normally nodes aren't restarted very often and by now would keep the the old originators until [_br-client_ changes it netdev ifindex and triggers a restart](https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-radv-filterd/files/etc/init.d/gluon-radv-filterd#L20).

jplitza, 2018-03-10:
@NeoRaider Thanks for highlighting me!

@belzebub40k I'm glad you rediscovered your comment yourself. :stuck_out_tongue_closed_eyes: 
What is you usecase? Did the gateway know that its originator changed? Because then it could expire its RA manually, causing gluon-radv-filterd to delete the whole router object and thus rediscover the originator on the next useful announcement.
OTOH, originators might change from time to time, for example if they are dynamically generated and thus non-persistent across reboots. So re-checking them from time to time can't hurt.