freifunk-gluon/gluon #1344: UBNT-UF-AC-PRO: WiFi interfaces brocken on IBSS + 11s

2tata, 2018-03-14:
In the Freifunk Nordwest community we recognize a problem with Ubiquity Unify AC Pro devices.
Firmwares with IBSS + 11s will brake the device and require a recovery over TFTP with the original firmware. This affected not all of this devices some devices even from the same order run without problems. The upgrade Firmware base on:

 * Gluon-Version: v2017.1.x
 * Commit ID: 64e0e28b24a78473d95ed7ea82ae72ed35265867
 * Download-fastd: https://firmware.ffnw.de/fastd/20180225
 * Download-l2tp: https://firmware.ffnw.de/l2tp/20170225

The previous Firmware base on:

 * Gluon-Version: v2017.1.x
 * Commit ID: 2ae74fe737a45e589f85ace098fa8466dfb7927e
 * Download-fastd: https://firmware.ffnw.de/fastd/20171220
 * Download-l2tp: https://firmware.ffnw.de/l2tp/20171220

Changes between the two images are:
https://github.com/freifunk-gluon/gluon/compare/2ae74fe737a45e589f85ace098fa8466dfb7927e...64e0e28b24a78473d95ed7ea82ae72ed35265867

logread print:
```
Wed Feb 21 04:56:04 2018 kern.notice kernel: [    0.000000] Linux version 4.4.93 (tata@runner02.ffnw.de) (gcc version 5.4.0 (LEDE GCC 5.4.0 r3581+39-6b6578feec) ) #0 Thu Aug 3 01:34:19 2017
Wed Feb 21 04:56:04 2018 kern.debug kernel: [    0.000000] MyLoader: sysp=3020e320, boardp=e3782264, parts=8b22bab2
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.000000] bootconsole [early0] enabled
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.000000] SoC: Qualcomm Atheros QCA956X ver 1 rev 0
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.000000] Determined physical RAM map:
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.000000]  memory: 08000000 @ 00000000 (usable)
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Wed Feb 21 04:56:04 2018 kern.warn kernel: [    0.000000] No valid device tree found, continuing without
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.000000] Zone ranges:
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.000000] Movable zone start for each node
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.000000] Early memory node ranges
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
Wed Feb 21 04:56:04 2018 kern.debug kernel: [    0.000000] On node 0 totalpages: 32768
Wed Feb 21 04:56:04 2018 kern.debug kernel: [    0.000000] free_area_init_node: node 0, pgdat 804304d0, node_mem_map 81000000
Wed Feb 21 04:56:04 2018 kern.debug kernel: [    0.000000]   Normal zone: 256 pages used for memmap
Wed Feb 21 04:56:04 2018 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Wed Feb 21 04:56:04 2018 kern.debug kernel: [    0.000000]   Normal zone: 32768 pages, LIFO batch:7
Wed Feb 21 04:56:04 2018 kern.warn kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Wed Feb 21 04:56:04 2018 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
Wed Feb 21 04:56:04 2018 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Wed Feb 21 04:56:04 2018 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 32512
Wed Feb 21 04:56:04 2018 kern.notice kernel: [    0.000000] Kernel command line:  board=UBNT-UF-AC-PRO mtdparts=spi0.0:384k(u-boot)ro,64k(u-boot-env)ro,7744k(firmware),7744k(ubnt-airos)ro,128k(bs)ro,256k(cfg)ro,64k(EEPROM)ro console=ttyS0,115200 rootfstype=squashfs,jffs2 noinitrd
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.000000] Writing ErrCtl register=00000000
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.000000] Readback ErrCtl register=00000000
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.000000] Memory: 124944K/131072K available (3089K kernel code, 167K rwdata, 776K rodata, 312K init, 205K bss, 6128K reserved, 0K cma-reserved)
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.000000] NR_IRQS:51
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.000000] Clocks: CPU:775.000MHz, DDR:650.000MHz, AHB:258.333MHz, Ref:25.000MHz
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4932285024 ns
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.000006] sched_clock: 32 bits at 387MHz, resolution 2ns, wraps every 5541893118ns
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.008208] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.071017] pid_max: default: 32768 minimum: 301
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.075978] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.082950] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.092465] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.102878] futex hash table entries: 256 (order: -1, 3072 bytes)
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.110331] NET: Registered protocol family 16
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.116278] MIPS: machine is Ubiquiti UniFi-AC-PRO
Wed Feb 21 04:56:04 2018 kern.warn kernel: [    0.339815] registering PCI controller with io_map_base unset
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.354422] PCI host bridge to bus 0000:00
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.358789] pci_bus 0000:00: root bus resource [mem 0x12000000-0x13ffffff]
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.366026] pci_bus 0000:00: root bus resource [io  0x0001]
Wed Feb 21 04:56:04 2018 kern.info kernel: [    0.371911] pci_bus 0000:00: root bus resource [??? 0x00000000 flags 0x0]
[... 468 lines cut ...]
ip a print:
```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN qlen 1
link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
inet 127.0.0.1/8 scope host lo
valid_lft forever preferred_lft forever
inet6 ::1/128 scope host
valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP qlen 1000
link/ether 80:2a:a8:d3:cf:99 brd ff:ff:ff:ff:ff:ff
inet6 fe80::822a:a8ff:fed3:cf99/64 scope link
valid_lft forever preferred_lft forever
3: br-client: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
link/ether 80:2a:a8:d5:cf:99 brd ff:ff:ff:ff:ff:ff
inet6 fe80::822a:a8ff:fed5:cf99/64 scope link
valid_lft forever preferred_lft forever
4: eth0.2@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-client state UP qlen 1000
link/ether 80:2a:a8:d3:cf:99 brd ff:ff:ff:ff:ff:ff
5: br-wan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
link/ether 42:06:e2:ed:34:f0 brd ff:ff:ff:ff:ff:ff
inet6 fe80::4006:e2ff:feed:34f0/64 scope link
valid_lft forever preferred_lft forever
6: eth0.1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-wan state UP qlen 1000
link/ether 80:2a:a8:d3:cf:99 brd ff:ff:ff:ff:ff:ff
```

NeoRaider, 2018-03-15:
It is well-known that ath10k can't support 11s and IBSS using the same firmware. Don't create such setups.

NeoRaider, 2018-03-15:
Note: The issues in your log have nothing to do with 11s+IBSS. The issue in your log is #1301.

closed 2018-03-15 by NeoRaider

2tata, 2018-03-15:
Ah thanks, for the feadback.