freifunk-gluon/gluon #1345: enabling mesh on lan / wan

Byggvir, 2018-03-15:
I tried to enable mesh on lan / wan on some tp-link tl-wr841nd v9, a cpe 210 and a wr911nd v6 via ssh and uci as described in the docs (v217.1.5).

http://gluon.readthedocs.io/en/v2017.1.5/features/wired-mesh.html

The procedure works for a wr1043n v5 but not for the above devices.

For mesh on lan the interface br-mesh_lan is created but not added to as mesh to batman.

# /etc/init.d/network restart ; logread | grep 'bat'
```
Thu Mar 15 22:41:34 2018 daemon.notice netifd: Interface 'gluon_bat0' is now down
Thu Mar 15 22:41:34 2018 daemon.notice netifd: Interface 'bat0' is disabled
Thu Mar 15 22:41:35 2018 daemon.notice netifd: Network device 'bat0' link is down
Thu Mar 15 22:41:35 2018 daemon.notice netifd: Interface 'bat0' has link connectivity loss
Thu Mar 15 22:41:39 2018 daemon.notice netifd: Interface 'gluon_bat0' is setting up now
Thu Mar 15 22:41:40 2018 kern.info kernel: [362535.322444] batman_adv: bat0: Adding interface: primary0
Thu Mar 15 22:41:40 2018 kern.info kernel: [362535.328051] batman_adv: bat0: Interface activated: primary0
Thu Mar 15 22:41:40 2018 kern.info kernel: [362535.336075] 8021q: adding VLAN 0 to HW filter on device bat0
Thu Mar 15 22:41:40 2018 daemon.notice netifd: Interface 'bat0' is enabled
Thu Mar 15 22:41:40 2018 kern.info kernel: [362535.404931] device bat0 entered promiscuous mode
Thu Mar 15 22:41:40 2018 kern.info kernel: [362535.409862] br-client: port 2(bat0) entered forwarding state
Thu Mar 15 22:41:40 2018 kern.info kernel: [362535.415898] br-client: port 2(bat0) entered forwarding state
Thu Mar 15 22:41:40 2018 daemon.notice netifd: Network device 'bat0' link is up
Thu Mar 15 22:41:40 2018 daemon.notice netifd: Interface 'bat0' has link connectivity 
Thu Mar 15 22:41:40 2018 daemon.notice netifd: Interface 'bat0' is setting up now
Thu Mar 15 22:41:40 2018 daemon.notice netifd: Interface 'bat0' is now up
Thu Mar 15 22:41:40 2018 kern.info kernel: [362535.614534] batman_adv: bat0: Interface deactivated: primary0
Thu Mar 15 22:41:40 2018 kern.info kernel: [362535.716703] batman_adv: bat0: Interface activated: primary0
Thu Mar 15 22:41:40 2018 daemon.notice netifd: Interface 'gluon_bat0' is now up
Thu Mar 15 22:41:40 2018 kern.info kernel: [362536.024434] batman_adv: bat0: Adding interface: mesh-vpn
Thu Mar 15 22:41:40 2018 kern.info kernel: [362536.030030] batman_adv: bat0: The MTU of interface mesh-vpn is too small (1364) to handle the transport of batman-adv packets. Packets going over this interface will be fragmented on layer2 which could impact the performance. Setting the MTU to 1532 would solve the problem.
Thu Mar 15 22:41:40 2018 kern.info kernel: [362536.055436] batman_adv: bat0: Interface activated: mesh-vpn
Thu Mar 15 22:41:42 2018 kern.info kernel: [362537.415691] br-client: port 2(bat0) entered forwarding state
Thu Mar 15 22:41:42 2018 kern.info kernel: [362537.560126] batman_adv: bat0: Changing gw mode from: off to: client
Thu Mar 15 22:41:42 2018 kern.info kernel: [362537.622691] batman_adv: bat0: hop_penalty: Changing from: 30 to: 15
Thu Mar 15 22:41:42 2018 kern.info kernel: [362537.629922] batman_adv: bat0: orig_interval: Changing from: 1000 to: 5000
Thu Mar 15 22:41:44 2018 kern.info kernel: [362539.304465] batman_adv: bat0: Adding interface: mesh0
Thu Mar 15 22:41:44 2018 kern.info kernel: [362539.310093] batman_adv: bat0: Interface activated: mesh0

```
On the 843 for the lan you see following options.

uci show network.mesh_lan
```
network.mesh_lan=interface
network.mesh_lan.ifname='eth1'
network.mesh_lan.mesh_no_rebroadcast='1'
network.mesh_lan.mesh='bat0'
network.mesh_lan.igmp_snooping='0'
network.mesh_lan.transitive='1'
network.mesh_lan.fixed_mtu='1'
network.mesh_lan.proto='gluon_mesh'
network.mesh_lan.type='bridge'
network.mesh_lan.macaddr='c2:c8:02:a4:ed:f4'
network.mesh_lan.auto='1'
```
On the working 1043n v5 I get following options:

```
network.mesh_lan=interface
network.mesh_lan.igmp_snooping='0'
network.mesh_lan.ifname='eth0.1'
network.mesh_lan.index='4'
network.mesh_lan.legacy='0'
network.mesh_lan.type='bridge'
network.mesh_lan.transitive='1'
network.mesh_lan.macaddr='46:71:b2:bc:e2:94'
network.mesh_lan.fixed_mtu='1'
network.mesh_lan.proto='gluon_mesh'
network.mesh_lan.auto='1'
```
Neither adding network.mesh_lan.legacy='0' or network.mesh_lan.legacy='1' does not help.  

I wonder why identical configurations lead to different behavior.

rotanid, 2018-03-16:
the legacy option isn't available in Gluon v2017.1.5 - so i think your WR1043N v5 is probably running a build based on the Gluon master branch perhaps?
please check if the WR1043N v5 also works with a build based on Gluon stable branch.
you could also test the firmware of another community, so you can rule out problems with your specific build.

Byggvir, 2018-03-16:
The 1043nd v5 may have this option from December, when I tested the backport.

Byggvir, 2018-03-16:
I have three CPE 211 and three 841 with mesh on lan / wan since v2016.x They have identical options and the identical firmware v2017.1.5 and are working.

rotanid, 2018-03-18:
i'm confused, now they're working fine?!?
please also re-flash the WR1043N v5 *without keeping the settings* to make sure no "old" stuff from testing builds makes the difference which results in different behaviour.

Byggvir, 2018-04-21:
Some work fine - some not.

Byggvir, 2018-04-21:
I have tried it with two routers in a controlled environment. Two routers on a separate switch, separate channel (9 instead 5) on own meshid, ...

Switching on mesh_on_lan worked as described. 

Issue closed.

Byggvir, 2018-04-21:
In a controlled environment it worked as designed.

closed 2018-04-21 by Byggvir