freifunk-gluon/gluon #1351: 2nd NetworkCard doesn't show up(x86)

HACKER-3000, 2018-03-24:
I've installed gluon(ff3l) on an old thin client, and then added an pci network Card, but i cant use it(mesh on lan, [](https://wiki.freifunk.net/Konsole#Meshing_on_LAN))

NeoRaider, 2018-03-24:
1. Was the card present during the initial installation of the Gluon image? We currently determine how to use network interfaces only once during the first boot.
2. Is the driver for the network card included in the image? Installing the driver afterwards using opkg cannot work for the same reason: the network interface will not show up during the first boot.

HACKER-3000, 2018-03-25:
I installed it later on but the driver is avaialble!
Some way to installit later?

rotanid, 2018-03-25:
yes, you have to edit /etc/config/network manually and add another section there

HACKER-3000, 2018-04-16:
and waht exactly

Adorfer, 2018-04-19:
network cards are detected during firstinstall (->"'firstboot' command"). 
If you have to add kernel modules first: No chance for "automatic process" with gluon. (sorry)

You may ask for "additional modules included", either in an existing gluon-target or to create a new target explicitly for the machine-type you have. 

But if it comes to "how to add a network card manuall:" That would not be an issue of gluon. but a question of understanding how a linux (and therefore an openwrt) system works. 
what you have to set there depends upon your requirements. 

Have a look (for example) into https://wiki.openwrt.org/doc/uci/network 
https://github.com/freifunk-gluon/gluon/wiki/Commandline-administration
https://wiki.freifunk.net/Konsole

closed 2018-04-19 by rotanid