freifunk-gluon/gluon #1352: Adding drivers for WiFi dongels(86x)

HACKER-3000, 2018-03-25:
Waht about adding drivers for some WiFi dongels to the 86x version of gluon?
Just for some popular ones?
Would be very cool to use thinclients as a normal router!

rotanid, 2018-03-25:
you can easily add drivers to Gluon by yourself by adding the driver to your "site.mk" file.

closed 2018-03-25 by rotanid

Adorfer, 2018-03-26:
and from past experience: at least on RPI or usb-equipped QCA platforms: ath9k-usb is not only rare and ovetpriced:

the wifi performance is even on the few (nominal) 2xMimo with external antennas poor, compared to a cheap 841.

The main point seems to be the total lack of performance when it comes to meshing. probably the driver generates too much io load (polled io? or inadequate buffers?). 
so even with perfect wifi link in a quiet environment, the mesh will be stuck at net effective less than 10MBit/s throughput. 

(if somebody finds the blocking issue and resolves that: Ver welcome. But for the time beeing, I have given up on USB-Wifi-mesh, since it is not possible to use in a serious case. )