freifunk-gluon/gluon #1353: dnsmasq is spamming logfiles

A-Kasper, 2018-03-26:
https://pastebin.com/Lj1UMruL

example: 
Mon Mar 26 19:05:12 2018 daemon.info dnsmasq[1879]: reading /tmp/resolv.conf.auto
Mon Mar 26 19:05:12 2018 daemon.info dnsmasq[1879]: using local addresses only for domain lan
Mon Mar 26 19:05:12 2018 daemon.info dnsmasq[1879]: using nameserver 2620:fe::fe#53

site.conf does not contain dns-cache parameter, so there is no dns configured

NeoRaider, 2018-03-26:
Is 2620:fe::fe the RDNSS advertised in router advertisements in your mesh? What RDNSS lifetime is configured in your radvd?

rotanid, 2018-03-26:
there you have it, from the docs:
AdvRDNSSLifetime "Default: 2*MaxRtrAdvInterval "
--> 8 seconds in your case
default would be 2*600s = 20 minutes

NeoRaider, 2018-03-26:
On LEDE 17.01.x, dhcpv6.script is run a most once every 30s. This means that all livetimes (both default route and DNS lifetimes) must be greater than 30s (for robustness against packet loss, I'd suggest using something like 70s or greater), otherwise default route and DNS servers will time out before being renewed.

In the OpenWrt master (gluon-next branch), this has been changed to once every 3s to be compliant with the very short RA intervals the RFCs allow (and that you use). Note that this is probably still a bad idea, as there are likely other client implementations that have problems with such short lifetimes, and running dhcpv6.script frequently leads to considerable load on nodes.

Considering that every RA causes load and short lifetimes only become relevant in case of unexpected gateway failures, my recommendation would be to use a MinRtrAdvInterval of 30-60s and lifetimes in the range of 1-5 minutes.

closed 2018-03-26 by NeoRaider