freifunk-gluon/gluon #1355: Error building gluon-next

mokkin, 2018-03-28:
I'm receiving an error while building the new OpenWRT-base gluon-next.
Site: https://github.com/freifunk-leipzig/freifunk-gluon-leipzig/blob/11s/site.mk
Build command: ```make -j1 V=s GLUON_TARGET=ar71xx-generic GLUON_RELEASE=20180325-11s BROKEN=1```
Error:
```
sl/gcc-7.3.0/gcc/../libcpp/include -I/srv2/gluon/20180325/openwrt/staging_dir/host/include -I/srv2/gluon/20180325/openwrt/staging_dir/host/include -I/srv2/gluon/20180325/openwrt/staging_dir/host/include  -I/srv2/gluon/20180325/openwrt/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../libdecnumber -I/srv2/gluon/20180325/openwrt/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../libdecnumber/dpd -I../libdecnumber -I/srv2/gluon/20180325/openwrt/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../libbacktrace   -o cfgbuild.o -MT cfgbuild.o -MMD -MP -MF ./.deps/cfgbuild.TPo /srv2/gluon/20180325/openwrt/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/cfgbuild.c
g++ -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I/srv2/gluon/20180325/openwrt/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc -I/srv2/gluon/20180325/openwrt/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/. -I/srv2/gluon/20180325/openwrt/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../include -I/srv2/gluon/20180325/openwrt/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../libcpp/include -I/srv2/gluon/20180325/openwrt/staging_dir/host/include -I/srv2/gluon/20180325/openwrt/staging_dir/host/include -I/srv2/gluon/20180325/openwrt/staging_dir/host/include  -I/srv2/gluon/20180325/openwrt/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../libdecnumber -I/srv2/gluon/20180325/openwrt/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../libdecnumber/dpd -I../libdecnumber -I/srv2/gluon/20180325/openwrt/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../libbacktrace   -o cfgcleanup.o -MT cfgcleanup.o -MMD -MP -MF ./.deps/cfgcleanup.TPo /srv2/gluon/20180325/openwrt/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/cfgcleanup.c
g++ -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I/srv2/gluon/20180325/openwrt/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc -I/srv2/gluon/20180325/openwrt/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/. -I/srv2/gluon/20180325/openwrt/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../include -I/srv2/gluon/20180325/openwrt/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../libcpp/include -I/srv2/gluon/20180325/openwrt/staging_dir/host/include -I/srv2/gluon/20180325/openwrt/staging_dir/host/include -I/srv2/gluon/20180325/openwrt/staging_dir/host/include  -I/srv2/gluon/20180325/openwrt/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../libdecnumber -I/srv2/gluon/20180325/openwrt/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../libdecnumber/dpd -I../libdecnumber -I/srv2/gluon/20180325/openwrt/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/../libbacktrace   -o cfgexpand.o -MT cfgexpand.o -MMD -MP -MF ./.deps/cfgexpand.TPo /srv2/gluon/20180325/openwrt/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0/gcc/cfgexpand.c
Makefile:1099: recipe for target 'cfgexpand.o' failed
make[6]: *** [cfgexpand.o] Terminated
make[6]: Leaving directory '/srv2/gluon/20180325/openwrt/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0-initial/gcc'
Makefile:4210: recipe for target 'all-gcc' failed
make[5]: *** [all-gcc] Error 2
make[5]: Leaving directory '/srv2/gluon/20180325/openwrt/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0-initial'
Makefile:36: recipe for target '/srv2/gluon/20180325/openwrt/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0-initial/.built' failed
make[4]: *** [/srv2/gluon/20180325/openwrt/build_dir/toolchain-mips_24kc_gcc-7.3.0_musl/gcc-7.3.0-initial/.built] Error 2
make[4]: Leaving directory '/srv2/gluon/20180325/openwrt/toolchain/gcc/initial'
toolchain/Makefile:96: recipe for target 'toolchain/gcc/initial/compile' failed
make[3]: *** [toolchain/gcc/initial/compile] Error 2
make[3]: Leaving directory '/srv2/gluon/20180325/openwrt'
toolchain/Makefile:94: recipe for target '/srv2/gluon/20180325/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/stamp/.toolchain_compile' failed
make[2]: *** [/srv2/gluon/20180325/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/stamp/.toolchain_compile] Error 2
make[2]: Leaving directory '/srv2/gluon/20180325/openwrt'
/srv2/gluon/20180325/openwrt/include/toplevel.mk:216: recipe for target 'world' failed
make[1]: *** [world] Error 2
make[1]: Leaving directory '/srv2/gluon/20180325/openwrt'
Makefile:148: recipe for target 'all' failed
make: *** [all] Error 2

```

mweinelt, 2018-03-28:
HI,

could you post a full build log?

I've successfully built next with all targets (except mvebu and ar71xx-mikrotik) today and yesterday.

NeoRaider, 2018-03-28:
Is this both stdout and stderr? The actual error message seems to be missing in the log.

`make[6]: *** [cfgexpand.o] Terminated` looks a bit like there might be an external cause, although I don't know what the cause could be.

mokkin, 2018-03-29:
Here is the complete build log:
http://firmware.leipzig.freifunk.net:8004/srv2/gluon/20180325/build.log

rotanid, 2018-03-30:
@NeoRaider high resource usage might also result in oom-killer running @mokkin

mokkin, 2018-03-30:
Thanks for your hint @NeoRaider ! The reason of the first error was a high blocking system load that I discovered later.
Now I'm running into a new error that can be found here:
http://firmware.leipzig.freifunk.net:8004/srv2/gluon/20180325/newlog

NeoRaider, 2018-03-31:
Does your build system have any unusual limitations regarding the use of /tmp? I can't reproduce the error, but the ldbus package is somewhat unclean as it uses /tmp outside the OpenWrt build dir...

NeoRaider, 2018-03-31:
I have pushed an update for gluon-next, ldbus is not built anymore with the new version.

mokkin, 2018-04-01:
I have tried with your update on the old system and another one, but both with the same error I think:
http://firmware.leipzig.freifunk.net:8004/srv2/gluon/20180331/log2

NeoRaider, 2018-04-01:
You did not properly update your tree. ldbus should not be mentioned at all in the log.

Note that gluon-next is regularly rebased and force-pushed, so you need to reset your tree on each update (a simple pull will not work), and make sure you `make update` after updating the git tree.